realsenseai/librealsense #11493: Coordinate system

IndiGleb, 2023-02-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

How i can get coordinate from camera (D455) ? Maybe there is a function that defines this ? As a result, I want to get a matrix containing coordinates in each pixel

MartyG-RealSense, 2023-02-27:
Hi @IndiGleb  It sounds as though exporting the data to a **.csv** file may fit your needs.  A .csv file is a textual file of all the coordinates that can be imported into a database or spreadsheet.  A csv could be exported directly with a program script, or you could record a **bag** file of the camera streams and then convert the bag to csv format with the RealSense SDK's **rs-convert** tool.  This subject is discussed at https://github.com/IntelRealSense/librealsense/issues/11090

IndiGleb, 2023-02-27:
I saw that in the view you can find out the values of coordinates at a point. My task is to get a three-channel matrix containing coordinates (x,y,z)
[image]

MartyG-RealSense, 2023-02-27:
You could generate a pointcloud with **pc.calculate** and then extract and print the XYZ values of each coordinate with the instruction **points.get_vertices**.  https://github.com/IntelRealSense/librealsense/issues/4612#issuecomment-566864616 has a Python example of points.get_vertices whilst https://github.com/IntelRealSense/librealsense/issues/5728 is a C++ example.

IndiGleb, 2023-02-27:
I can measured coordinate without pointcloud ?

MartyG-RealSense, 2023-02-27:
Obtaining the 3D world-space coordinate of a single specific pixel coordinate on an image without using a pointcloud or alignment can be done with the instruction **rs2_project_color_pixel_to_depth_pixel** and information about this instruction is at https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

IndiGleb, 2023-02-27:
where can I find this method ?

MartyG-RealSense, 2023-02-27:
Are you using Python or C++ please?

IndiGleb, 2023-02-27:
С++. I want to know where the measured coordinate from pointcloud and rs2_project_color_pixel_to_depth_pixel

MartyG-RealSense, 2023-02-27:
A C++ example of using rs2_project_color_pixel_to_depth_pixel  is at https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-615249398

For the pointcloud, using points.get_vertices like in https://github.com/IntelRealSense/librealsense/issues/5728 to obtain the coordinates will likely be the best method to use.

IndiGleb, 2023-02-28:
In which cpp source file can i find what is contained in these methods ?

IndiGleb, 2023-02-28:
I'm trying to figure out exactly how the coordinates are calculated

MartyG-RealSense, 2023-02-28:
These two sections of the SDK's source code may be especially relevant:

**rs.cpp*
https://github.com/IntelRealSense/librealsense/blob/3371f4d6c80f37ccae1491d16f41b039f2049396/src/rs.cpp#L3762-L3769

**unit-tests-live.cpp**
https://github.com/IntelRealSense/librealsense/blob/1b9a40da35f20a46b2113cabd920ce37f757743b/unit-tests/unit-tests-live.cpp#L5220-L5233

IndiGleb, 2023-03-01:
Post-processing work when camera enable on ethernet ?

MartyG-RealSense, 2023-03-01:
My understanding from a RealSense team member's advice at https://github.com/IntelRealSense/librealsense/issues/6376 is that if a network camera running on the **rs-server** ethernet networking system has a serial number defined in a script then it should behave in that script like a normal non-networked camera.  I cannot recall a previous case of post-processing being applied to a network camera though.

The rs-server ethernet system is being removed in the next RealSense SDK version however, so it would not be a suitable networking solution to choose if you plan to update the SDK in future.

The **EtherSense** Python-based ethernet system should continue to work.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server
https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

IndiGleb, 2023-03-01:
how i can disable post-processing in code &

MartyG-RealSense, 2023-03-01:
You do not need to disable post-processing filters in code when creating your own program as they are not active by default and have to be deliberately programmed into a script.  They are only enabled by default in the RealSense Viewer tool.

IndiGleb, 2023-03-01:
How i can check this ?

MartyG-RealSense, 2023-03-01:
You do not need to check if the filters are disabled if you have not programmed them into your script as the filters will not be applied.  This is because the program does not know how to use filters unless you tell it how to by adding post-processing filter code.

MartyG-RealSense, 2023-03-06:
They share the general principle of being able to send data from a camera attached to a computer (the remote server) to another computer with a display and no camera attached (the host) but otherwise they are completely different in how they work.

IndiGleb, 2023-03-06:
How the image will be sent over the network?

IndiGleb, 2023-03-06:
BGR will remain jpeg ?

IndiGleb, 2023-03-06:
Or maybe H264?

IndiGleb, 2023-03-06:
Where can I read more about EtherSense ?

IndiGleb, 2023-03-06:
I mean how the image will compress both RGB and depth ?

MartyG-RealSense, 2023-03-06:
It is sent by the TCP protocol in EtherSense, whilst rs-server uses the RTSP protocol.

There are not further information resources about EtherSense, unfortunately.  The paper linked to earlier is the best guide available.

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

It appears from the section of the paper linked to below that the streams are uncompressed, but advice is offered on different ways to reduce bandwidth.

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper#network-bandwidth

closed 2023-03-23 by MartyG-RealSense