realsenseai/librealsense #11496: OpenCV

sar21455, 2023-02-28:
[image]

Hi all, I am trying to use OpenCV Samples for Intel® RealSense™ cameras, I followed the instructions in this link  
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/readme.md#windows

Everything went well and every projects runs as it should except with OpenCV projects, I received the error shown in the picture above for every project from OpenCV. Can anyone help please? Thank you in advance 

BTW I am using Windows 10 Home 
Visual Studio 2019
Intel D455

sar21455, 2023-02-28:
I found out that when I use opencv-4.x found here https://github.com/opencv/opencv 
everything started working except OpenCV dnn
here is the  results when I run it

[ INFO:0@1.678] global registry.impl.hpp:114 cv::highgui_backend::UIBackendRegistry::UIBackendRegistry UI: Enabled backends(4, sorted by priority): GTK(1000); GTK3(990); GTK2(980); WIN32(970) + BUILTIN(WIN32UI)
[ INFO:0@1.678] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load C:\git\librealsense\build\Debug\opencv_highgui_gtk470_64.dll => FAILED
[ INFO:0@1.680] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_highgui_gtk470_64.dll => FAILED
[ INFO:0@1.680] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load C:\git\librealsense\build\Debug\opencv_highgui_gtk3470_64.dll => FAILED
[ INFO:0@1.681] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_highgui_gtk3470_64.dll => FAILED
[ INFO:0@1.681] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load C:\git\librealsense\build\Debug\opencv_highgui_gtk2470_64.dll => FAILED
[ INFO:0@1.681] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_highgui_gtk2470_64.dll => FAILED
[ INFO:0@1.682] global backend.cpp:90 cv::highgui_backend::createUIBackend UI: using backend: WIN32 (priority=970)
[ INFO:0@1.682] global window_w32.cpp:3008 cv::impl::Win32BackendUI::createWindow OpenCV/UI: Creating Win32UI window: Display Image (1)
[ INFO:0@2.693] global registry_parallel.impl.hpp:96 cv::parallel::ParallelBackendRegistry::ParallelBackendRegistry core(parallel): Enabled backends(3, sorted by priority): ONETBB(1000); TBB(990); OPENMP(980)
[ INFO:0@2.694] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load C:\git\librealsense\build\Debug\opencv_core_parallel_onetbb470_64d.dll => FAILED
[ INFO:0@2.696] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_core_parallel_onetbb470_64d.dll => FAILED
[ INFO:0@2.696] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load C:\git\librealsense\build\Debug\opencv_core_parallel_tbb470_64d.dll => FAILED
[ INFO:0@2.697] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_core_parallel_tbb470_64d.dll => FAILED
[ INFO:0@2.697] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load C:\git\librealsense\build\Debug\opencv_core_parallel_openmp470_64d.dll => FAILED
[ INFO:0@2.699] global plugin_loader.impl.hpp:67 cv::plugin::impl::DynamicLib::libraryLoad load opencv_core_parallel_openmp470_64d.dll => FAILED
OpenCV(4.7.0-dev) Error: Assertion failed (!blobs.empty() || inputs.size() > 1) in cv::dnn::ConvolutionLayerImpl::getMemoryShapes, file C:\git\opencv-4.x\modules\dnn\src\layers\convolution_layer.cpp, line 392
[ERROR:0@2.797] global net_impl.cpp:1164 cv::dnn::dnn4_v20221220::Net::Impl::getLayerShapesRecursively OPENCV/DNN: [Convolution]:(conv0): getMemoryShapes() throws exception. inputs=1 outputs=0/1 blobs=0
[ERROR:0@2.797] global net_impl.cpp:1167 cv::dnn::dnn4_v20221220::Net::Impl::getLayerShapesRecursively     input[0] = [ 1 3 300 300 ]
[ERROR:0@2.798] global net_impl.cpp:1177 cv::dnn::dnn4_v20221220::Net::Impl::getLayerShapesRecursively Exception message: OpenCV(4.7.0-dev) C:\git\opencv-4.x\modules\dnn\src\layers\convolution_layer.cpp:392: error: (-215:Assertion failed) !blobs.empty() || inputs.size() > 1 in function 'cv::dnn::ConvolutionLayerImpl::getMemoryShapes'

OpenCV(4.7.0-dev) C:\git\opencv-4.x\modules\dnn\src\layers\convolution_layer.cpp:392: error: (-215:Assertion failed) !blobs.empty() || inputs.size() > 1 in function 'cv::dnn::ConvolutionLayerImpl::getMemoryShapes'

C:\git\librealsense\build\Debug\OFFrs-dnn.exe (process 11256) exited with code 1.
To automatically close the console when debugging stops, enable Tools->Options->Debugging->Automatically close the console when debugging stops.
Press any key to close this window . . .

MartyG-RealSense, 2023-02-28:
Hi @sar21455  Which OpenCV were you originally using when you tried the samples, please?  They were designed for use with 3.4, with minor code changes required for use with OpenCV 4.

There are some additional OpenCV at the link below that you could try.

https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md

ehtisham-Fazal, 2023-03-04:
i think this error message suggests that there is an assertion failure in the convolution layer implementation of OpenCV's deep neural network (DNN) module. This could be due to a number of reasons, such as incorrect input shapes or missing blob data.
It is possible that the issue is related to the way you installed or compiled OpenCV. The failed library loads in the log you provided could indicate that some dependencies are missing or incorrectly linked.
You may want to try recompiling OpenCV from source, ensuring that all required dependencies are installed and properly linked. You may also want to check that your input data is correctly formatted and that the model you are trying to use is compatible with your input data.

MartyG-RealSense, 2023-03-04:
Thanks so much @ehtisham-Fazal for your advice to @sar21455

MartyG-RealSense, 2023-03-09:
Hi @sar21455  Do you require further assistance with this case, please?  Thanks!

sar21455, 2023-03-13:
@ehtisham-Fazal Thank you so much for your help

sar21455, 2023-03-13:
@MartyG-RealSense no thank you so much to all of you

closed 2023-03-13 by sar21455