realsenseai/librealsense #11504: rs2_context error in rs_context.hpp while loading Unreal Engine 5.1

ffreality, 2023-03-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version | Windows 11 22H2 |
| Platform                                 | PC  |
| SDK Version                           |  v2.53.1 | 
| Language                               | C++ and Unreal Engine 5.1.1| 
| Segment			        |  VR (Oculus Quest Pro) | 

### Issue Description
I am working on Unreal Engine 5 Android plugin for Realsense for attaching my D455 to Quest Pro and getting RGB texture from it.
My Windows SDK is v2.53.1 from release section.

My build.cs file is like this.
```
        if (UnrealTargetPlatform.Win64 == Target.Platform)
        {
            string Location_Realsense = "../Source/FF_RealSense/ThirdParty/Windows/include";
            PrivateIncludePaths.Add(Location_Realsense);

            PublicAdditionalLibraries.Add(Path.Combine(ModuleDirectory, "ThirdParty", "Windows", "libs", "realsense2.lib"));
            RuntimeDependencies.Add(Path.Combine(ModuleDirectory, "ThirdParty", "Windows", "libs", "realsense2.dll"));
            PublicDelayLoadDLLs.Add("realsense2.dll");
        }

        if (UnrealTargetPlatform.Android == Target.Platform)
        {
            PrivateDependencyModuleNames.Add("Launch");

            string Location_Realsense = "../Source/FF_RealSense/ThirdParty/Android/include";
            PrivateIncludePaths.Add(Location_Realsense);

            PublicAdditionalLibraries.Add(Path.Combine(ModuleDirectory, "ThirdParty", "Android", "libs", "libnative-lib.so"));
            PublicAdditionalLibraries.Add(Path.Combine(ModuleDirectory, "ThirdParty", "Android", "libs", "librealsense2.so"));

            AdditionalPropertiesForReceipt.Add("AndroidPlugin", Path.Combine(ModuleDirectory, "FF_RealSense_UPL_Android.xml"));
        }
```
My DLL delayed load module is like this
```
void FFF_RealSenseModule::StartupModule()
{
#ifdef _WIN64
	const FString BasePluginDir = IPluginManager::Get().FindPlugin("FF_REALSENSE")->GetBaseDir();
	const FString DLL_Path = FPaths::Combine(*BasePluginDir, TEXT("Source/FF_RealSense/ThirdParty/Windows/libs/realsense2.dll"));
	RealSenseHandle = FPlatformProcess::GetDllHandle(*DLL_Path);

	if (RealSenseHandle != nullptr)
	{
		UE_LOG(LogTemp, Log, TEXT("realsense2.dll loaded successfully!"));
	}

	else
	{
		UE_LOG(LogTemp, Fatal, TEXT("realsense2.dll failed to load!"));
	}
#endif
}

void FFF_RealSenseModule::ShutdownModule()
{
#ifdef _WIN64
	FPlatformProcess::FreeDllHandle(RealSenseHandle);
	RealSenseHandle = nullptr;
#endif
}
```
I am using #include "librealsense2/rs.hpp" and it pass visual but when Unreal Engine loading hits %45 (delayed load section)
it gives crash like this and Visual Studio's output ils like this.
```
Exception thrown at 0x00007FFA06A9FE7C (KernelBase.dll) in UnrealEditor.exe: 0xC06D007E: Module not found (parameters: 0x0000003E84979550).
Unhandled exception at 0x00007FFA06A9FE7C (KernelBase.dll) in UnrealEditor.exe: 0xC06D007E: Module not found (parameters: 0x0000003E84979550).
```
[image]

Is there any suggestion for that ?

MartyG-RealSense, 2023-03-01:
Hi @ffreality  A project at the link below for connecting a RealSense D435i camera to the original Oculus Quest headset may be a helpful reference.  The RealSense Android wrapper's **AAR** should enable the camera to connect to an Android device if included in a project as the AAR provides device access permissions.

https://github.com/GeorgeAdamon/quest-realsense

ffreality, 2023-03-01:
> Hi @ffreality A project at the link below for connecting a RealSense D435i camera to the original Oculus Quest headset may be a helpful reference. The RealSense Android wrapper's **AAR** should enable the camera to connect to an Android device (including the Android-based Quest headsets) if included in a project as the AAR provides device access permissions.
> 
> https://github.com/GeorgeAdamon/quest-realsense

But my problem is not about Android or Android package. First, I need to integrate Windows libraries to open Unreal Engine. This also allows me to test my functions. Problem is about Windows side because I can't even start editor on Windows.
If problem was about my Android integration, It would crash at Android package or on Quest's runtime.

So, that repo can't solve my problem.

MartyG-RealSense, 2023-03-01:
Looking at your GitHub repository for your project, I did not see the RealSense x64 DLL files **Intel.RealSense.dll** and **realsense2.dll** included.  Usually when creating your own wrapper, these two dll files need to be included in the project in order for librealsense functions to be used.

On Windows, pre-made versions of these DLL files can be found by installing the full RealSense SDK on your Windows computer with the installer program **Intel.RealSense.SDK-WIN10** and then going to the following folder location:

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin > x64**
 
You can then copy the two DLL files from this folder and paste them into your project folder.

If you have not installed the full RealSense SDK on your computer, you can do so by visiting the link below and downloading and running **Intel.RealSense.SDK-WIN10**, which is in the 'Assets' file list at the bottom of the page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

[image]

ffreality, 2023-03-01:
I saw Intel.RealSense.dll on Unity repo but latest Windows releases don't have that. There is only ```realsense2.dll in bin folder``` and ```realsense2.lib in lib``` folder.

I included both of them. You can find these in here.
https://github.com/FF-Projects-UE/FF_RealSense/tree/main/Source/FF_RealSense/ThirdParty/Windows
and there integrations are in build.cs.

These are screenshot of my SDK folder.
[image]
[image]

Also I rewrote my functions with
```
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_frame.h>
```
and Unreal opened without problem.
If released sdk.exe should contain Intel.RealSense.dll, it is broken.

> Looking at your GitHub repository for your project, I did not see the RealSense x64 DLL files **Intel.RealSense.dll** and **realsense2.dll** included. Usually when creating your own wrapper, these [...]

I saw Intel.RealSense.dll on Unity repo but latest Windows releases don't have that. There is only ```realsense2.dll in bin folder``` and ```realsense2.lib in lib``` folder.

I included both of them. You can find these in here.
https://github.com/FF-Projects-UE/FF_RealSense/tree/main/Source/FF_RealSense/ThirdParty/Windows
and there integrations are in build.cs.

These are screenshot of my SDK folder.
[image]
[image]

Also I rewrote my functions with
```
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_frame.h>
```
and Unreal opened without problem.
If released sdk.exe should contain Intel.RealSense.dll, it is broken.

ffreality, 2023-03-01:
> Looking at your GitHub repository for your project, I did not see the RealSense x64 DLL files **Intel.RealSense.dll** and **realsense2.dll** included. Usually when creating your own wrapper, these [...]

I saw Intel.RealSense.dll on Unity repo but latest Windows releases don't have that. There is only ```realsense2.dll in bin folder``` and ```realsense2.lib in lib``` folder.

I included both of them. You can find these in here.
https://github.com/FF-Projects-UE/FF_RealSense/tree/main/Source/FF_RealSense/ThirdParty/Windows
and there integrations are in build.cs.

These are screenshot of my SDK folder.
[image]
[image]

Also I rewrote my functions with
```
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_frame.h>
```
and Unreal opened without problem.
If released sdk.exe should contain Intel.RealSense.dll, it is broken.

ffreality, 2023-03-01:
On the other hand android wrappers give this error when I try to use gradlew.
```
Error: Could not find or load main class org.gradle.wrapper.GradleWrapperMain
Caused by: java.lang.ClassNotFoundException: org.gradle.wrapper.GradleWrapperMain
```

but I can build this repo with gradle without problem and using it as android library.
https://github.com/mihir-chauhan/IntelRealsenseNative

MartyG-RealSense, 2023-03-01:
It is unusual that both of the DLL files are not appearing in your SDK folder as I did a test install on Windows (though Windows 10 Home rather than Windows 11 Enterprise) for SDKs 2.53.1 and 2.51.1 and the files are there on my computer.

[image]

Usually I recommend not changing the default component installation settings on the installer program and just clicking 'Next' a few times until SDK installation is complete.  I also tried your method of installing just the C / C++ SDK, debug tools and Viewer and the DLL files were still present in the SDK bin folder.

It's great to hear that you made very significant progress!

There was a past case involving this **Could not find or load main class org.gradle.wrapper.GradleWrapperMain** error and a RealSense user posted their solution at https://github.com/IntelRealSense/librealsense/issues/6748#issuecomment-747250814 which was to change gradle version in Android Studio.  I do not know if you are using Android Studio yourself though.

ffreality, 2023-03-01:
using rs.h instead of rs.hpp solved my problem.
Also I built android wrapper from Android studio as you mentioned.

So, only problem is initating realsense for android.

closed 2023-03-01 by ffreality

MartyG-RealSense, 2023-03-02:
It's great news that you achieved a solution.  Thanks very much for the update!