realsenseai/librealsense #11506: Installing realsense on Raspberry Compute Module 4 ARM Ubuntu Server 20.04

pfcouto, 2023-03-01:
Hi @MartyG-RealSense  is there a way for me to install the realsense SDK, I need to be able to develop using pyrealsense, in my raspberry pi compute module 4, also, I need to have Ubuntu Server 20.04.

I found this post `https://support.intelrealsense.com/hc/en-us/community/posts/1500000077181-Realsense-Install-Ubuntu-20-04-Raspberry-PI-4` where you mention this comment `https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049`. I removed the `CMake flag  -DBUILD_WITH_CUDA:bool=true` as you mention, however I am still getting some error (shown in the picture), referencing to cuda. Looking for a fix I found this issue `https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13`. However, I do't believe that their solution is going to work for me as I don't have CUDA. DO you know how to fix this error?

I also found this page `https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md`  where is presented a solution to build from source, will I be able to use python to get the color and depth streams using this method? As per this method I do have some questions. Which flags should I in the CMAKE command and which command show I do to build? Should I do the first one which doesn't have the `make -j4` command or the other that does? Thanks!

[image]

---
---

[image]

[image]

MartyG-RealSense, 2023-03-02:
Hi @pfcouto  There have only been a couple of past cases about the Compute Module form of Raspbery Pi.  In general though, if a device is capable of having an operating system installed on it then a RealSense 400 Series camera should be able to be used with it.

It is also not a problem to have Ubuntu Server 20.04 installed on a Raspberry Pi device.

If you cannot use CUDA because the Pi does not have an Nvidia graphics GPU then you can simply leave **-DBUILD_WITH_CUDA:bool=true** out of the CMake build instruction.  I note that you have already done this though.

You would also need to have Python 3.6 installed on your computer in order for the CMake build instruction that you used to work.  If you do not need to use Python then you could remove the Python related commands.

`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

You will need to use **make** in order to build librealsense.  It is usually fine to use **make -j4** but if that does not work then you can just use **make**, though the build process will take longer to complete (the -j command accelerates build time by making use of multiple cores in the CPU chip).

**********

There is an alternative installation guide for Pi 4 and Ubuntu 20.04 that you could try.  This installs librealsense and the RealSense ROS wrapper.

https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

pfcouto, 2023-03-02:
Hello @MartyG-RealSense 

To either solution I believe I will have to use cmake. Can you tell me which Flags should I use? As mentioned I am using Ubuntu Server, so I have no desktop, only terminal and I need to use python.

MartyG-RealSense, 2023-03-02:
Installing librealsense on Raspberry Pi tends to have more problems than nstalling on other computing devices, especially with Ubuntu instead of Raspberry Pi OS.

If you have Python 3.6 installed on your Pi then the flags to use would be:

`cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python3.6 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true`

DBUILD_GRAPHICAL_EXAMPLES is left out of the build instruction because you only have a terminal, not a graphics interface, and installing graphics support and graphical examples / tools on a terminal-only computer can cause problems.

My research of your problem indicates that the red error may imply that a C++ compiler is not installed on your Pi.  One can be installed in Ubuntu using the command **sudo apt-get install gcc-5 g++-5**

MartyG-RealSense, 2023-03-08:
Hi @pfcouto  Do you require further assistance with this case, please?  Thanks!

pfcouto, 2023-03-08:
I don't. I am closing it. Thanks for your help!

closed 2023-03-08 by pfcouto

MartyG-RealSense, 2023-03-08:
You are very welcome. :)

ruthvik92, 2023-06-19:
> Hi @pfcouto There have only been a couple of past cases about the Compute Module form of Raspbery Pi. In general though, if a device is capable of having an operating system installed on it then a RealSense 400 Series camera should be able to be used with it.
> 
> It is also not a problem to have Ubuntu Server 20.04 installed on a Raspberry Pi device.
> 
> If you cannot use CUDA because the Pi does not have an Nvidia graphics GPU then you can simply leave **-DBUILD_WITH_CUDA:bool=true** out of the CMake build instruction. I note that you have already done this though.
> 
> You would also need to have Python 3.6 installed on your computer in order for the CMake build instruction that you used to work. If you do not need to use Python then you could remove the Python related commands.
> 
> `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`
> 
> You will need to use **make** in order to build librealsense with CMake. It is usually fine to use **make -j4** but if that does not work then you can just use **make**, though the build process will take longer to complete (the -j command accelerates build time by making use of multiple cores in the CPU chip).
> 
> There is an alternative installation guide for Pi 4 and Ubuntu 20.04 that you could try. This installs librealsense and the RealSense ROS wrapper.
> 
> https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

@MartyG-RealSense : the author in the [ROS forum](https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/) marked it as deprecated because Intel has provided the SDK bianries for AMD64 through the package manager. However, RPi Ubuntu is AMD64 so I'm gessing that one should follow the instructions in the ROS forum to build realsense SDK for Ubuntu on RPi ?

MartyG-RealSense, 2023-06-20:
I believe the **deprecated** statement is referring to how librealsense officially supported Ubuntu 20.04 (Focal) from version 2.40.0 onwards (released in November 2020) because of their comment "Ubuntu 20.04 LTS is now natively supported by the RealSense SDK 2.0".  The author originally published their question in October 2020 and updated it the next February, hence the '[Deprecated] February 2021' edit because their previous advice was overriden by the new Ubuntu 20.04 librealsense support.

Intel do not provide Arm64 packages for building librealsense on Ubuntu 20.04 for Arm devices such as Raspberry Pi and Nvidia Jetson (they do have Arm64 packages for Ubuntu 18.04), so it is recommended to build it from source code on 20.04 instead, preferably using the **RSUSB backend** installation method in order to bypass the Linux kernel.

Regarding using the ROS installation instructions, the ROS1 wrapper (Kinetic, Melodic, Noetic) had installation packages that are based on RSUSB, which is why it is worth trying it for Arm64 devices such as Pi.

ruthvik92, 2023-06-20:
@MartyG-RealSense :
I was able to succesfully use the [script](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py) to display the frames. 

I had to add extra delay in the line:
`frames = pipeline.wait_for_frames(10000)`

Here's short note for someone trying to build `librealsense` on `Ubuntu 22.04.2` on `Raspberry Pi 4` (8GB RAM).  
1. Install dependencies:
````
sudo apt-get update
sudo apt-get upgrade
sudo apt install cmake
sudo apt-get install build-essential
sudo apt install plocate
sudo apt-get install python3-dev
sudo apt install python3-pip
pip3 install numpy
pip3 install opencv-python
````
2. Follow the instructions in the [ROS Forum](https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/). Instead of just doing `make install` do a `make -j 4` followed by `sudo make install`. This will speed up the process by approximately 4x. 
3. In my case, I forgot to build the `pyrealsense` so I had to issue the `make/install` again by the following commands. See [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md). 

````
cd ~/projects/librealsense/build
cmake .. -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3)
make -j4
sudo make install
````
4. I had to delete extra .so files generated usually with an appended number in (`/opt/realsense/bin/` and `/opt/realsense/lib/`).
5.  After this, i tried to run the above python script but I got the following error:
````
ine 8, in <module>
import pyrealsense2 as rs
ModuleNotFoundError: No module named 'pyrealsense2'
````
6. I checked the usual `/usr/local/lib/python3.10/dist-packages`, `~/.local/lib/python3.10/site-packages/` and didn't find anything about `pyrealsense2`. I also checked `/opt/realsense/lib` and I found `.so` files for `librealsense` but not for `pyrealsense`.  I moved the `pyrealsense2` `.so` files from `~/projects/librealsense/build/wrappers/python` to `$HOME` and renamed them as shown below. 
````
cp ~/projects/librealsense/build/wrappers/python/pyrealsense2.cpython-310-aarch64-linux-gnu.so.2.54 .
cp ~/projects/librealsense/build/wrappers/python/pybackend2.cpython-310-aarch64-linux-gnu.so .
cp ~/projects/librealsense/build/wrappers/python/pyrealsense2.cpython-310-aarch64-linux-gnu.so .
mv pyrealsense2.cpython-310-aarch64-linux-gnu.so pyrealsense2.so
mv pybackend2.cpython-310-aarch64-linux-gnu.so pybackend2.so
mv pyrealsense2.cpython-310-aarch64-linux-gnu.so.2.54 pyrealsense2.so.2.54
````
Of course, one can move these into appropriate Python library directories like `site-packages` in `~/.local` or elsewhere but this at least gets the job done for now.

MartyG-RealSense, 2023-06-20:
It's excellent news that you were successful, @ruthvik92 - thanks so much for sharing the details of your solution with the RealSense community!