realsenseai/librealsense #11507: /camera/imu is black

shivangi-futuristicbots, 2023-03-02:
I followed the GitHub link for the experience the real-sense camera d435i : https://github.com/IntelRealSense/realsense-ros [ros-development]

I am able to read the linear acceleration and angular velocity form /camera/accel/sample and /camera/gyro/sample but i want that to publish in one topic 

i already use the unite_imu_method: linear-acceleration and copy parameter too but it is not publishing any thing.

Please reply soon..

MartyG-RealSense, 2023-03-02:
Hi @shivangi-futuristicbots  If you are using the ROS2 **ros2-development** wrapper then the unite_imu_method instruction is defined in a different way.  Instead of using **unite_imu_method:=linear_interpolation**, the linear_interpolation setting is simply defined with the value '2'.

`unite_imu_method:=2`

shivangi-futuristicbots, 2023-03-02:
Thank for your support @MartyG-RealSense  
I have one queries is real-sense camera would be preferred for the AVG and IMU reading of the Real-sense camera and imu reading of other imu sensors [lord microstraine], which is better?

MartyG-RealSense, 2023-03-02:
If you are asking whether an external high quality, high cost IMU sensor will provide better results than the IMU that is built into the RealSense camera, the answer is likely yes.

MartyG-RealSense, 2023-03-10:
Hi @shivangi-futuristicbots  Do you require further assistance with this case, please?  Thanks!

shivangi-futuristicbots, 2023-03-13:
Not Yet , may be in future Thanks @MartyG-RealSense  for the help

MartyG-RealSense, 2023-03-13:
Okay, thanks very much @shivangi-futuristicbots for the update.  I will close this issue as you do not require further assistance.  Plese feel free to re-open it at a future date or create a new issue if you need to do so.  Thanks again!

closed 2023-03-13 by MartyG-RealSense