realsenseai/librealsense #11509: C++ Equelevent of RsContext init

ffreality, 2023-03-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                        | D455 | 
| Firmware Version                    | 05.14.00.00 | 
| Operating System & Version  | Windows 11 | 
| Platform                                  | Android 10 |
| SDK Version                            | v2.53.1 | 
| Language                                |  C++ / Unreal Engine | 
| Segment			          | VR | 

### Issue Description
I created a plugin to access RGB feed of realsense from Unreal Engine. It is working on Windows platform and I believe it will also work on Android (if ```rs.h``` works on Android, it will work.)

But application's itself don't see Realsense. I checked some old issues and this repository and found out that I need to initialize RsContext.
https://github.com/GeorgeAdamon/quest-realsense

**Current sample is for C# and Unity. Is there any sample how can I do it with C++ ?**
we can get game activity with ```FAndroidGameActivity()``` function. So we need these.

``` 
using (var rsContext = new AndroidJavaClass("com.intel.realsense.librealsense.RsContext"))
        {
            Debug.Log(rsContext);
            rsContext.CallStatic("init", currentActivity);
        }
```

MartyG-RealSense, 2023-03-02:
Hi @ffreat ality  An Intel RealSense developer who worked on the Android wrapper provides advice at https://github.com/IntelRealSense/librealsense/issues/3385#issuecomment-481723287 about RsContext and permissions.

ffreality, 2023-03-03:
It's done. I did initalization on Java and called its function with JNI. Currently I can get RGB stream.
Tomorrow, I will add some other functions like distance calculation and other stuffs.

closed 2023-03-03 by ffreality

immensivevr, 2023-05-24:
> It's done. I did initalization on Java and called its function with JNI. Currently I can get RGB stream. Tomorrow, I will add some other functions like distance calculation and other stuffs.

i can ask you where ? i have a problem with my project unreal on android run  Context->QueryDevices(); this don't found the realsense device.

ffreality, 2023-05-24:
> > It's done. I did initalization on Java and called its function with JNI. Currently I can get RGB stream. Tomorrow, I will add some other functions like distance calculation and other stuffs.
> 
> i can ask you where ? i have a problem with my project unreal on android when inspecto run Context->QueryDevices(); this don't found the realsense device.

I used https://github.com/gtreshchev/AndroidNative to call JNI methods. I don't remember exact problem but there were some blocks about JNI.

I put every realsense related java files to its Private/Java folder.
include all of them and just initialized rsContext and call that method with static JNI call.
I didn't query anything on Java side. I did all device related thing with C.