realsenseai/librealsense #11510: How to open RAW16 RGB stream on Ubuntu 22

daniellehot, 2023-03-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i| 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  5.19.0-32-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.53.1.4623  | 
| Language                            |  Python                    | 

### Issue Description
Hello. I would like to open a stream of RAW RGB data. This is my code:

```python
import pyrealsense2 as rs

img_width = 1920
img_height = 1080
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, img_width, img_height, rs.format.raw16, 30)
pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        color_frame = frames.get_color_frame()
        if not color_frame:
            continue
        else:
            print(color_frame)
            break
finally:
    pipeline.stop()
```
This code works for RGB8 format but results in a runtime error for raw16 and raw8.
```
RuntimeError: Couldn't resolve requests
```
My goal is to get uncompressed but rectified colour data from the camera. My current approach is to get the raw readings, and then convert and rectify them myself. The reason why I want to do it like this is that if I get data from the RGB8 stream directly, the data appears compressed (somewhat blurred, with finer details of my object lost). I would like to see if I can maybe get higher-quality images by processing raw readings myself. However, I am open to suggestions regarding achieving better RGB images.

daniellehot, 2023-03-02:
Thanks for the suggestions @MartyG-RealSense. FPS is not relevant for me, as my object of interest is static. I will investigate the effect of sharpness in my setup further, but I am afraid there might have been a small misunderstanding. I am aware that RAW data cannot be displayed by RealSense SDK. My issue is that I am unable to start a stream, i.e. the code fails at `pipeline.start(config)`

You suggest converting raw frames to a usable RGB format using an OpenCV function. However, I cannot start the RAW stream in the first place.

MartyG-RealSense, 2023-03-02:
Please try changing the resolution in your config instruction for RAW16 to 1280x800 instead of 1920x1080.  This may overcome the **RuntimeError: Couldn't resolve requests** error, though the stream could not be visually displayed.

[image]

MartyG-RealSense, 2023-03-10:
Hi @daniellehot  Do you require further assistance with this case, please?  Thanks!

daniellehot, 2023-03-13:
Hi @MartyG-RealSense. Sorry for the late response. We can close this issue for now, although I have not managed to start the raw stream. I have decided to use another RGB camera and RealSense as a depth sensor.

closed 2023-03-13 by daniellehot

MartyG-RealSense, 2023-03-13:
It's no problem at all.  Thanks very much for the update!