realsenseai/librealsense #11513: Two D435is return the same accelerometer data despite being at different orientations

sonali20, 2023-03-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.15.0-60-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0                      | 
| Language                            |  C++                         | 
| Segment			|  Robot                          | 

### Issue Description
I have two D435is connected to a PC and I'm printing 10 frames of the accelerometer values from each and I eventually compute the orientation. Here is the relevant snippet of code -

```
std::vector<rs2_vector> get_acceleration_frames(const rs2::device &device, int num_frames_to_average = 10)
{
  rs2::config config;
  config.enable_stream(RS2_STREAM_ACCEL, RS2_FORMAT_MOTION_XYZ32F);
  rs2::pipeline pipe;
  pipe.start(config);

  std::cout << "Capturing " << num_frames_to_average << " frames for camera serial " << device.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << std::endl;
  std::vector<rs2_vector> captured_frames;
  for (int frame_count = 0; frame_count < num_frames_to_average; ++frame_count)
  {
    rs2::frameset frames = pipe.wait_for_frames();
    rs2::frame frame = frames.first(RS2_STREAM_ACCEL);
    if (frame)
    {
      auto acceleration_frame = frame.as<rs2::motion_frame>().get_motion_data();
      std::cout << "z acceleration: " << acceleration_frame.z << std::endl;
      captured_frames.push_back(acceleration_frame);
    }
  }
  return captured_frames;
}
```
I call this function for the two devices one after the other and for some reason, they both print the same values (with some noise) despite being at completely different orientations. When I instead connect just one of the cameras at a time and call this function, they return distinct and reasonable values that correspond well to their orientation. It seems like when I have both connected, I'm getting readings from just one IMU. This is a bit baffling since that means I get the correct serial number but the wrong IMU data when I query the device. How can this be?

sonali20, 2023-03-02:
Also, I don't see this when I use the realsense-viewer. Both acceleration vectors look distinct in that case. 
I also tried upgrading the firmware to 5.14.0. 
Here is a snippet of what the program prints with the two cameras connected -
```
`Device #0: Intel RealSense D435I, firmware version: 05.14.00.00, serial_no: 152122071575
  This device has 3sensors:
    Sensor #0: Stereo Module
    Sensor #1: RGB Camera
    Sensor #2: Motion Module
Device #1: Intel RealSense D435I, firmware version: 05.14.00.00, serial_no: 948122073206
  This device has 3sensors:
    Sensor #0: Stereo Module
    Sensor #1: RGB Camera
    Sensor #2: Motion Module
Resetting device #0...
Resetting device #1...
Capturing 10 frames for camera serial 152122071575
z acceleration: 0.892405
z acceleration: 0.921825
z acceleration: 0.921825
z acceleration: 0.931632
z acceleration: 0.902212
z acceleration: 0.872792
z acceleration: 0.833565
z acceleration: 0.794339
z acceleration: 0.784532
z acceleration: 0.794339
Captured 10 frames
Mean x acceleration is 0.411879
Mean y acceleration is -9.66543
Mean z acceleration is 0.864947
Pitch of camera serial 152122071575 is 0.0882845
Capturing 10 frames for camera serial 948122073206
z acceleration: 0.843372
z acceleration: 0.872792
z acceleration: 0.902212
z acceleration: 0.912018
z acceleration: 0.912018
z acceleration: 0.902212
z acceleration: 0.872792
z acceleration: 0.853179
z acceleration: 0.843372
z acceleration: 0.833565
Captured 10 frames
Mean x acceleration is 0.420705
Mean y acceleration is -9.70662
Mean z acceleration is 0.874753
Pitch of camera serial 948122073206 is 0.0892881
```
With just the first camera connected -
```
Device #0: Intel RealSense D435I, firmware version: 05.14.00.00, serial_no: 948122073206
  This device has 3sensors:
    Sensor #0: Stereo Module
    Sensor #1: RGB Camera
    Sensor #2: Motion Module
Resetting device #0...
Capturing 10 frames for camera serial 948122073206
z acceleration: 9.74781
z acceleration: 9.70858
z acceleration: 9.70858
z acceleration: 9.68897
z acceleration: 9.7282
z acceleration: 9.74781
z acceleration: 9.70858
z acceleration: 9.70858
z acceleration: 9.7282
z acceleration: 9.64974
Captured 10 frames
Mean x acceleration is 0.237321
Mean y acceleration is 2.00252
Mean z acceleration is 9.7125
Pitch of camera serial 948122073206 is 1.42969
```
With just the second -
```
Device #0: Intel RealSense D435I, firmware version: 05.14.00.00, serial_no: 152122071575
  This device has 3sensors:
    Sensor #0: Stereo Module
    Sensor #1: RGB Camera
    Sensor #2: Motion Module
Resetting device #0...
Capturing 10 frames for camera serial 152122071575
z acceleration: 0.343233
z acceleration: 0.353039
z acceleration: 0.353039
z acceleration: 0.372653
z acceleration: 0.411879
z acceleration: 0.411879
z acceleration: 0.421686
z acceleration: 0.441299
z acceleration: 0.460913
z acceleration: 0.470719
Captured 10 frames
Mean x acceleration is 0.844353
Mean y acceleration is -9.75369
Mean z acceleration is 0.404034
Pitch of camera serial 152122071575 is 0.0411976
```
The cameras were kept in the same position throughout. Its clear that when both were connected, 948122073206's IMU values get reported for both cameras

MartyG-RealSense, 2023-03-03:
Hi @sonali20  In the print-out at https://github.com/IntelRealSense/librealsense/issues/11513#issuecomment-1452641009 of the accel values for both cameras at the same time, each camera has mostly unique accel values (with **0.902212** being the only value common to both cameras) and is not displaying the same values for both cameras.

It is not necessary to write your own routine to calculate orientation as this is provided for you by the **gyro** values of the IMU - **RS2_STREAM_GYRO**

sonali20, 2023-03-03:
@MartyG-RealSense  They're not the exact same values but they're all values close to 0 while I'm expecting values in the ballpark of 9.8 for 948122073206. I held one camera facing up and one camera facing sideways so there is roughly a 90 degree rotation between them which should mean a drastic difference between the accelerometer values from both. And you can see when I have just that one camera connected, the values are indeed in the ballpark of 9.8 for 948122073206 as I expect.

I'll try the gyro and get back to you.

MartyG-RealSense, 2023-03-03:
Thank you for the clarification.  I understand what you mean.

It seems from https://github.com/IntelRealSense/librealsense/issues/4297 that D435i multiple camera problems with IMU data have occurred in the past.

sonali20, 2023-03-09:
I no longer need this work correctly so closing the ticket. Thanks for your support @MartyG-RealSense

closed 2023-03-09 by sonali20