realsenseai/librealsense #11514: Extracting a rectified image for android device

Tikitaka02, 2023-03-03:
Hello,

I am using D435 depth camera. I could compile the sample application in the repo for an android device using android studio, but I would like to extract rectified frames. How can I do it? Or is there any detailed documentation for the realsense sdk for android?

I appreciate your answer in advance.

MartyG-RealSense, 2023-03-03:
Hi @Tikitaka02  Almost all RealSense stream formats except Y16 are already rectified by default so you do not need to write a special routine to rectify them.  The rectification is applied to the image inside the camera hardware before the image is sent along the USB cable.

There are example programs for the SDK's Android wrapper at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples

Tikitaka02, 2023-03-03:
Thank you for your message. I meant rectified images as not only the undistorted images, but also the combined images from 2 cameras.

MartyG-RealSense, 2023-03-03:
For an undistorted image the easiest method would be to use Y16 format (available as infrared and RGB), as it does not have a distortion model applied to it.  It is a monochrome image though.

There is not an unrectified format available for depth.

Unrectified RGB can also be obtained by accessing the camera's raw RGB frames and converting them to a usable RGB format with OpenCV's **cvtColor** instruction.

When you mention two cameras, do you mean two D435 cameras, or aligning two images (depth and colour) on the same camera, please

Tikitaka02, 2023-03-03:
I meant the 2 views in the same camera. I will explain the background. I noticed that there 4 sensors on the camera. 1 for infrared, 2 for depth, and 1 for normal rgb camera. I could place the normal rgb screen and depth screen side by side like the example named capture. However, the 2 screens have different perspective since I think the depth screen is the combination of 2 sensors. I would like to have the combined view as rgb image.

MartyG-RealSense, 2023-03-03:
The stereo depth image is constructed from images produced by the left and right infrared sensors.  These are **raw** infrared frames in the camera hardware.  Depth is not generated from the Infrared and Infrared 2 streams, which is why depth can be enabled even if the Infrared and Infrared 2 streams are inactive.

You can **align** the depth and RGB screeens into a single combined image.  There are few references though for how to perform depth-color alignment in Android.  One of the best available references is https://github.com/IntelRealSense/librealsense/issues/5520

As well as depth to color alignment (where the depth image is mapped onto the RGB image), you can also perform **color to depth** alignment, where the RGB image is mapped onto the depth image.

Tikitaka02, 2023-03-06:
Thank you so much. I could align the view. Would it be possible to render some points and lines on the aligned view?

Tikitaka02, 2023-03-08:
Thank you for the answer. Now I can draw on the display, but I would like to only display rgb stream and put the depth stream background so that I can still get depth information. How can I do it?

MartyG-RealSense, 2023-03-08:
You could perhaps create two separate pipelines (pipeline1 and pipeline2), each with their own configuration (config1 and config2).  Put the color stream definition in the config1 definition and the depth stream in the config2 definition.  Start both pipeline1 and pipeline2 so that both are streaming data, but only visually render pipeline1 (the color one).

Tikitaka02, 2023-03-08:
But I would like to align the color and depth also.

MartyG-RealSense, 2023-03-09:
Possibly you could put the depth code within a true / false **bool** condition so that when 'depth = true' the depth information is allowed to be produced, and if 'depth = false' then the depth information is not allowed to be produced.

Tikitaka02, 2023-03-09:
So what I have done is like this

`    private void configAndStart() throws Exception {
        try(Config config  = new Config())
        {
            // Set the format corresponding to bitmap
            config.enableStream(StreamType.COLOR, 640, 480, StreamFormat.RGBA8);
            config.enableStream(StreamType.DEPTH, 640, 480);
            // try statement needed here to release resources allocated by the Pipeline:start() method
            try(PipelineProfile pp = mPipeline.start(config)){

            }

        }
    }`

`try(FrameReleaser fr = new FrameReleaser()) {
                    try (FrameSet frames = mPipeline.waitForFrames(1000).releaseWith(fr)) {
                        try (VideoFrame videoFrame = frames.first(StreamType.COLOR).releaseWith(fr).as(Extension.VIDEO_FRAME);) {
                            try (Frame f = frames.first(StreamType.DEPTH)) {
                                f.releaseWith(fr);
                                DepthFrame depthFrame = f.as(Extension.DEPTH_FRAME);
                                extrinsic = videoFrame.getProfile().getExtrinsicTo(depthFrame.getProfile());
                                float[] rot = extrinsic.getRotation();
                                float[] trans = extrinsic.getTranslation();
                                for (int i =0; i<rot.length; i++) {
                                    Log.e("CamParams", "intRot "+String.valueOf(i)+" : "+String.valueOf(rot[i]));
                                }
                                for (int i =0; i<trans.length; i++) {
                                    Log.e("CamParams", "intTrans "+String.valueOf(i)+" : "+String.valueOf(trans[0]));
                                }

                                intrinsicDepth = depthFrame.getProfile().getIntrinsic();
                                float fxD = intrinsicDepth.getFx();
                                float fyD = intrinsicDepth.getFy();
                                float PpxD = intrinsicDepth.getPpx();
                                float PpyD = intrinsicDepth.getPpy();

                                Log.e("CamParamsDepth", "intFx: "+String.valueOf(fxD));
                                Log.e("CamParamsDepth", "intFy: "+String.valueOf(fyD));
                                Log.e("CamParamsDepth", "intPpx: "+String.valueOf(PpxD));
                                Log.e("CamParamsDepth", "intPpy: "+String.valueOf(PpyD));
                            }

                            intrinsic = videoFrame.getProfile().getIntrinsic();
                            float fx = intrinsic.getFx();
                            float fy = intrinsic.getFy();
                            float Ppx = intrinsic.getPpx();
                            float Ppy = intrinsic.getPpy();

                            Log.e("CamParamsColor", "intFx: "+String.valueOf(fx));
                            Log.e("CamParamsColor", "intFy: "+String.valueOf(fy));
                            Log.e("CamParamsColor", "intPpx: "+String.valueOf(Ppx));
                            Log.e("CamParamsColor", "intPpy: "+String.valueOf(Ppy));

                        }

                        mGLSurfaceView.upload(frames);
                    }

                }
                mHandler.post(mStreaming);
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }`

I could extract camera extrinsics and intrinsics from both depth an color views, but how can I find the pixcel correspondance to know the specific pixel in color view is corresponding to that pixel in depth view?

Tikitaka02, 2023-03-09:
So what I have done is like this

` private void configAndStart() throws Exception {
try(Config config = new Config())
{
// Set the format corresponding to bitmap
config.enableStream(StreamType.COLOR, 640, 480, StreamFormat.RGBA8);
config.enableStream(StreamType.DEPTH, 640, 480);
// try statement needed here to release resources allocated by the Pipeline:start() method
try(PipelineProfile pp = mPipeline.start(config)){

        }

    }
}`

and

`try(FrameReleaser fr = new FrameReleaser()) {
try (FrameSet frames = mPipeline.waitForFrames(1000).releaseWith(fr)) {
try (VideoFrame videoFrame = frames.first(StreamType.COLOR).releaseWith(fr).as(Extension.VIDEO_FRAME);) {
try (Frame f = frames.first(StreamType.DEPTH)) {
f.releaseWith(fr);
DepthFrame depthFrame = f.as(Extension.DEPTH_FRAME);
extrinsic = videoFrame.getProfile().getExtrinsicTo(depthFrame.getProfile());
float[] rot = extrinsic.getRotation();
float[] trans = extrinsic.getTranslation();
for (int i =0; i<rot.length; i++) {
Log.e("CamParams", "intRot "+String.valueOf(i)+" : "+String.valueOf(rot[i]));
}
for (int i =0; i<trans.length; i++) {
Log.e("CamParams", "intTrans "+String.valueOf(i)+" : "+String.valueOf(trans[0]));
}

                            intrinsicDepth = depthFrame.getProfile().getIntrinsic();
                            float fxD = intrinsicDepth.getFx();
                            float fyD = intrinsicDepth.getFy();
                            float PpxD = intrinsicDepth.getPpx();
                            float PpyD = intrinsicDepth.getPpy();

                            Log.e("CamParamsDepth", "intFx: "+String.valueOf(fxD));
                            Log.e("CamParamsDepth", "intFy: "+String.valueOf(fyD));
                            Log.e("CamParamsDepth", "intPpx: "+String.valueOf(PpxD));
                            Log.e("CamParamsDepth", "intPpy: "+String.valueOf(PpyD));
                        }

                        intrinsic = videoFrame.getProfile().getIntrinsic();
                        float fx = intrinsic.getFx();
                        float fy = intrinsic.getFy();
                        float Ppx = intrinsic.getPpx();
                        float Ppy = intrinsic.getPpy();

                        Log.e("CamParamsColor", "intFx: "+String.valueOf(fx));
                        Log.e("CamParamsColor", "intFy: "+String.valueOf(fy));
                        Log.e("CamParamsColor", "intPpx: "+String.valueOf(Ppx));
                        Log.e("CamParamsColor", "intPpy: "+String.valueOf(Ppy));

                    }

                    mGLSurfaceView.upload(frames);
                }

            }
            mHandler.post(mStreaming);
        }
        catch (Exception e) {
            Log.e(TAG, "streaming, error: " + e.getMessage());
        }`
I could extract camera extrinsics and intrinsics from both depth an color views, but how can I find the pixcel correspondance to know the specific pixel in color view is corresponding to that pixel in depth view?

Tikitaka02, 2023-03-09:
So what I have done is like this

`    private void configAndStart() throws Exception {
        try(Config config  = new Config())
        {
            // Set the format corresponding to bitmap
            config.enableStream(StreamType.COLOR, 640, 480, StreamFormat.RGBA8);
            config.enableStream(StreamType.DEPTH, 640, 480);
            // try statement needed here to release resources allocated by the Pipeline:start() method
            try(PipelineProfile pp = mPipeline.start(config)){

            }

        }
    }`

and

`Runnable mStreaming = new Runnable() {
        @Override
        public void run() {
            try {
                try(FrameReleaser fr = new FrameReleaser()) {
                    try (FrameSet frames = mPipeline.waitForFrames(1000).releaseWith(fr)) {
//                        FrameSet aligned_frames = align.process(frames).releaseWith(fr);

                        try (VideoFrame videoFrame = frames.first(StreamType.COLOR).releaseWith(fr).as(Extension.VIDEO_FRAME);) {

                            try (Frame f = frames.first(StreamType.DEPTH)) {
                                f.releaseWith(fr);
                                DepthFrame depthFrame = f.as(Extension.DEPTH_FRAME);

                                extrinsic = videoFrame.getProfile().getExtrinsicTo(depthFrame.getProfile());
                                float[] rot = extrinsic.getRotation();
                                float[] trans = extrinsic.getTranslation();
                                for (int i =0; i<rot.length; i++) {
                                    Log.e("CamParams", "intRot "+String.valueOf(i)+" : "+String.valueOf(rot[i]));
                                }
                                for (int i =0; i<trans.length; i++) {
                                    Log.e("CamParams", "intTrans "+String.valueOf(i)+" : "+String.valueOf(trans[0]));
                                }

                                intrinsicDepth = depthFrame.getProfile().getIntrinsic();
                                float fxD = intrinsicDepth.getFx();
                                float fyD = intrinsicDepth.getFy();
                                float PpxD = intrinsicDepth.getPpx();
                                float PpyD = intrinsicDepth.getPpy();

                                Log.e("CamParamsDepth", "intFx: "+String.valueOf(fxD));
                                Log.e("CamParamsDepth", "intFy: "+String.valueOf(fyD));
                                Log.e("CamParamsDepth", "intPpx: "+String.valueOf(PpxD));
                                Log.e("CamParamsDepth", "intPpy: "+String.valueOf(PpyD));
                            }

                            intrinsic = videoFrame.getProfile().getIntrinsic();
                            float fx = intrinsic.getFx();
                            float fy = intrinsic.getFy();
                            float Ppx = intrinsic.getPpx();
                            float Ppy = intrinsic.getPpy();

                            Log.e("CamParamsColor", "intFx: "+String.valueOf(fx));
                            Log.e("CamParamsColor", "intFy: "+String.valueOf(fy));
                            Log.e("CamParamsColor", "intPpx: "+String.valueOf(Ppx));
                            Log.e("CamParamsColor", "intPpy: "+String.valueOf(Ppy));

                            Log.e("bitmapSize","Height"+ String.valueOf(image.getHeight()));
                            Log.e("bitmapSize: ","Width" + String.valueOf(image.getWidth()));
                        }
                        mGLSurfaceView.upload(frames);
                    }
                }
                mHandler.post(mStreaming);
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }
        }
    };`
I could extract camera extrinsics and intrinsics from both depth an color views, but how can I find the pixcel correspondance to know the specific pixel in color view is corresponding to that pixel in depth view?

Tikitaka02, 2023-03-09:
So what I have done is like this

`    private void configAndStart() throws Exception {
        try(Config config  = new Config())
        {
            config.enableStream(StreamType.COLOR, 640, 480, StreamFormat.RGBA8);
            config.enableStream(StreamType.DEPTH, 640, 480);
            try(PipelineProfile pp = mPipeline.start(config)){
            }
        }
    }`

and

`Runnable mStreaming = new Runnable() {
        @Override
        public void run() {
            try {
                try(FrameReleaser fr = new FrameReleaser()) {
                    try (FrameSet frames = mPipeline.waitForFrames(1000).releaseWith(fr)) {

                        try (VideoFrame videoFrame = frames.first(StreamType.COLOR).releaseWith(fr).as(Extension.VIDEO_FRAME);) {

                            try (Frame f = frames.first(StreamType.DEPTH)) {
                                f.releaseWith(fr);
                                DepthFrame depthFrame = f.as(Extension.DEPTH_FRAME);

                                extrinsic = videoFrame.getProfile().getExtrinsicTo(depthFrame.getProfile());
                                float[] rot = extrinsic.getRotation();
                                float[] trans = extrinsic.getTranslation();
                                for (int i =0; i<rot.length; i++) {
                                    Log.e("CamParams", "intRot "+String.valueOf(i)+" : "+String.valueOf(rot[i]));
                                }
                                for (int i =0; i<trans.length; i++) {
                                    Log.e("CamParams", "intTrans "+String.valueOf(i)+" : "+String.valueOf(trans[0]));
                                }

                                intrinsicDepth = depthFrame.getProfile().getIntrinsic();
                                float fxD = intrinsicDepth.getFx();
                                float fyD = intrinsicDepth.getFy();
                                float PpxD = intrinsicDepth.getPpx();
                                float PpyD = intrinsicDepth.getPpy();

                                Log.e("CamParamsDepth", "intFx: "+String.valueOf(fxD));
                                Log.e("CamParamsDepth", "intFy: "+String.valueOf(fyD));
                                Log.e("CamParamsDepth", "intPpx: "+String.valueOf(PpxD));
                                Log.e("CamParamsDepth", "intPpy: "+String.valueOf(PpyD));
                            }

                            intrinsic = videoFrame.getProfile().getIntrinsic();
                            float fx = intrinsic.getFx();
                            float fy = intrinsic.getFy();
                            float Ppx = intrinsic.getPpx();
                            float Ppy = intrinsic.getPpy();

                            Log.e("CamParamsColor", "intFx: "+String.valueOf(fx));
                            Log.e("CamParamsColor", "intFy: "+String.valueOf(fy));
                            Log.e("CamParamsColor", "intPpx: "+String.valueOf(Ppx));
                            Log.e("CamParamsColor", "intPpy: "+String.valueOf(Ppy));

                            Log.e("bitmapSize","Height"+ String.valueOf(image.getHeight()));
                            Log.e("bitmapSize: ","Width" + String.valueOf(image.getWidth()));
                        }
                        mGLSurfaceView.upload(frames);
                    }
                }
                mHandler.post(mStreaming);
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }
        }
    };`
I could extract camera extrinsics and intrinsics from both depth an color views, but how can I find the pixcel correspondance to know the specific pixel in color view is corresponding to that pixel in depth view?

Tikitaka02, 2023-03-09:
So what I have done is like this

```
private void configAndStart() throws Exception {
    try(Config config  = new Config())
    {
        // Set the format corresponding to bitmap
        config.enableStream(StreamType.COLOR, 640, 480, StreamFormat.RGBA8);
        config.enableStream(StreamType.DEPTH, 640, 480);
        // try statement needed here to release resources allocated by the Pipeline:start() method
        try(PipelineProfile pp = mPipeline.start(config)){

        }

    }
}
```

and

```
Runnable mStreaming = new Runnable() {
        @Override
        public void run() {
            try {
                try(FrameReleaser fr = new FrameReleaser()) {
                    try (FrameSet frames = mPipeline.waitForFrames(1000).releaseWith(fr)) {

                        try (VideoFrame videoFrame = frames.first(StreamType.COLOR).releaseWith(fr).as(Extension.VIDEO_FRAME);) {

                            try (Frame f = frames.first(StreamType.DEPTH)) {
                                f.releaseWith(fr);
                                DepthFrame depthFrame = f.as(Extension.DEPTH_FRAME);

                                extrinsic = videoFrame.getProfile().getExtrinsicTo(depthFrame.getProfile());
                                float[] rot = extrinsic.getRotation();
                                float[] trans = extrinsic.getTranslation();
                                for (int i =0; i<rot.length; i++) {
                                    Log.e("CamParams", "intRot "+String.valueOf(i)+" : "+String.valueOf(rot[i]));
                                }
                                for (int i =0; i<trans.length; i++) {
                                    Log.e("CamParams", "intTrans "+String.valueOf(i)+" : "+String.valueOf(trans[0]));
                                }

                                intrinsicDepth = depthFrame.getProfile().getIntrinsic();
                                float fxD = intrinsicDepth.getFx();
                                float fyD = intrinsicDepth.getFy();
                                float PpxD = intrinsicDepth.getPpx();
                                float PpyD = intrinsicDepth.getPpy();

                                Log.e("CamParamsDepth", "intFx: "+String.valueOf(fxD));
                                Log.e("CamParamsDepth", "intFy: "+String.valueOf(fyD));
                                Log.e("CamParamsDepth", "intPpx: "+String.valueOf(PpxD));
                                Log.e("CamParamsDepth", "intPpy: "+String.valueOf(PpyD));
                            }

                            intrinsic = videoFrame.getProfile().getIntrinsic();
                            float fx = intrinsic.getFx();
                            float fy = intrinsic.getFy();
                            float Ppx = intrinsic.getPpx();
                            float Ppy = intrinsic.getPpy();

                            Log.e("CamParamsColor", "intFx: "+String.valueOf(fx));
                            Log.e("CamParamsColor", "intFy: "+String.valueOf(fy));
                            Log.e("CamParamsColor", "intPpx: "+String.valueOf(Ppx));
                            Log.e("CamParamsColor", "intPpy: "+String.valueOf(Ppy));

                            Log.e("bitmapSize","Height"+ String.valueOf(image.getHeight()));
                            Log.e("bitmapSize: ","Width" + String.valueOf(image.getWidth()));
                        }
                        mGLSurfaceView.upload(frames);
                    }
                }
                mHandler.post(mStreaming);
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }
        }
    };
```

I could extract camera extrinsics and intrinsics from both depth an color views, but how can I find the pixcel correspondance to know the specific pixel in color view is corresponding to that pixel in depth view?

MartyG-RealSense, 2023-03-09:
The SDK has an instruction called **rs2_project_color_pixel_to_depth_pixel** that can convert a single specific color pixel coordinate to a depth pixel coordinate in order to obtain 3D XYZ without using alignment or pointcloud.

Tikitaka02, 2023-03-09:
Would it be possible to paste the instruction page here? I cannot find it...

MartyG-RealSense, 2023-03-09:
There are not Android references about using rs2_project_color_pixel_to_depth_pixel with Android, unfortunately.  **kafan1986** at https://github.com/IntelRealSense/librealsense/issues/5766 where you just commented will likely have Android programming knowledge about using this instruction.

MartyG-RealSense, 2023-03-09:
For almost all 400 Series depth cameras except D405, the default depth scale value is **0.001** and does not change unless deliberately changed by the program or a user input.  So this value can be hard-coded into a script instead of retrieving the scale value from the camera in real-time.

Tikitaka02, 2023-03-09:
Got it thanks! Also, what do the depth max and min range  mean? When I try to obtain them without hard code, there are so many options to choose from.

MartyG-RealSense, 2023-03-09:
I have previously looked at depth_min and depth_max at https://github.com/IntelRealSense/librealsense/issues/10484#issuecomment-1137028484

Tikitaka02, 2023-03-14:
Thank you so much. I have one additional question. How can I change the orientation of view? Now I have 640px width and 480px height, but I would like to make it vertucal view like 360px width and 480px height without chaging the context meaning if a person is stands upright on a frame, the person should stand upright on a frame after changing the orientation.

MartyG-RealSense, 2023-03-14:
You could **crop** the image horizontally so that the width of the image is reduced but the height remains the same.  It is not a simple thing to do though.

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/8512#issuecomment-791742675 cropped a colorized depth image and shared the C++ code of their solution.

A different approach that another RealSense user described at https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157 was to define a **bounding box** on their depth image and exclude from the depth image all of the coordinates outside of the defined box.

Tikitaka02, 2023-03-20:
Sorry for this repetitive question. I was also wondering if there are any concerns about using the d435 camera under a glass screen.

MartyG-RealSense, 2023-03-20:
It's no trouble at all.  :)

It should be no problem to use the D435 behind a glass screen so long as both sides of the screen have an anti-reflective coating applied to them.  This is to avoid back-reflections from the camera's laser.  The image below is of a D435 in a custom housing for underwater use.

[image]

Tikitaka02, 2023-03-23:
Hello, Thank you for the information. Another question. Would it be possible to access only an RGB camera without using sdk in android?

MartyG-RealSense, 2023-03-23:
The two main examples of RealSense Android applications that are able to render RGB onscreen are **Capture** (which you mentioned earlier in this case) and the **RsCamera** demo app on the Google Play store.

https://play.google.com/store/apps/details?id=com.intel.realsense.camera

The project source code for the RsCamera app is available at the link below as the **Camera** tool.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/tools/camera

Tikitaka02, 2023-03-28:
Hello, I have 2 additional questions.

If I want to place the intel camera behind a glass (thickness 2.81mm), what is the exact type of anti-reflection treatment I have to place? There are a lot of types available in the market. Can I apply the treatment only in the area around the camera?

If I want to place the intel camera behind a glass (thickness 2.81mm), what is the acceptable distance between the camera lens and the glass surface to get a reliable measurement?

MartyG-RealSense, 2023-03-28:
There are not recommendations for a particular type of anti-reflective coating or for a minimum / maximum distance of the glass from the camera lens.  Yes, you could apply the coating only in the area around the camera.

If you would prefer not to coat the glass, you could alternatively try applying a **thin-film linear polarization filter** over the camera lenses on the outside of the camera.  This filter dampens glare from reflections in the scene such as window glass, like the 'without / with filter' image below.

[image]

Section **4.4  When to use polarizers and waveplates** of Intel's white-paper guide about use of optical filters with 400 Series cameras has more information about linear polarization filters.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#4-the-use-of-optical-filters

Any polarization filter that is linear will work, though the circular filters in 3D glasses will not.  This makes the filters inexpensive to purchase.  You can search stores such as Amazon for the term **linear polarizing filter sheet**

closed 2023-04-08 by MartyG-RealSense