realsenseai/librealsense #11517: The demo requires Depth camera with Color sensor

rongyaoxiaohuihui, 2023-03-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

rongyaoxiaohuihui, 2023-03-04:
The demo requires Depth camera with Color sensor

MartyG-RealSense, 2023-03-05:
Hi @rongyaoxiaohuihui  Which demo program are you referring to, please?

If it is the **align_depth2color.py** one that has an RGB sensor check, this program will not work with camera models that do not have an RGB sensor.  With the **D405** camera model, it has RGB but does not have a separate RGB sensor as RGB comes from the depth sensor.  align_depth2color.py can work with the D405 model if the RGB sensor check is removed from the script.  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/11329 has a version of the script that has this removal.

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense