realsenseai/librealsense #11521: librealsense installatino error with realtime Linux kernel

robertburk, 2023-03-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.15.86-rt56                                     | 
| Platform                                 | PC  |
| SDK Version                            |   }                          | 
| Language                            |                         | 
| Segment			|  Robot                    | 

### Issue Description
Hi,

I am trying to integrate the Intel RealSense camera into a ROS1 workspace that I have developed, however, it appears that the librealsense SDK is not compatible with my kernel version. I have tried to install via _https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md_, however during the _./libuvc_installation.sh_ command, I get the following error: 

Loading new librealsense2-dkms-1.3.19 DKMS files...
Building for 5.15.86-rt56
Building initial module for 5.15.86-rt56
ERROR (dkms apport): kernel package linux-headers-5.15.86-rt56 is not supported
Error! Bad return status for module build on kernel: 5.15.86-rt56 (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.19/build/make.log for more information.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

I would really appreciate some help in navigating around this problem. I cannot change kernel version as it is required in order to interface with hardware in the workspace.

Thanks for the help.

MartyG-RealSense, 2023-03-05:
Hi @robertburk  Using the **libuvc_installation.md** installation script with Ubuntu 20.04 fails as it is a very old installation method.

Support for use of kernel 5.15 with Ubuntu 20.04 was only first introduced in librealsense version **2.53.1**, the latest version at the time of writing this.  Also, using a realtime (RT) kernel with librealsense can add further complication.

The best approach for you to use is likely to be to install the SDK with the **RSUSB backend**, which is the successor to the libuvc installation method but is compatible with modern Ubuntu versions and works in a similar way to libuvc.  It involves building the SDK from source code with **CMake** instead of running an install script.   The old libuvc install script also uses CMake but automates the install process.

This installation method is not dependent on Linux versions or kernel versions and does not require patching, whilst allowing the non-RealSense components of your project to still interface with kernel 5.15.  Instructions for performing an RSUSB source code build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

**********

If you are using ROS1, a simpler approach to installation may be to use the RealSense ROS wrapper's **Method 1** install procedure, which is performed by inputting the following command into the Ubuntu terminal after having first installed ROS1 (Kinetic, Melodic or Noetic).

**sudo apt-get install ros-$ROS_DISTRO-realsense2-camera**  

This installs both librealsense and the ROS1 wrapper from **packages** at the same time instead of having to install librealsense and the wrapper separately.  As this package installation is based on the RSUSB backend, this should enable librealsense and the ROS wrapper to function without conflicting with the 5.15 kernel.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#installation-instructions

robertburk, 2023-03-05:
Thank you @MartyG-RealSense, I had gotten lost in the various issues raised on this and didn't know where to go exactly. In the end I installed the SDK using the RSUSB backend and the ROS node separately using the guides you linked. 

Now I am having issues connecting to the camera through the ROS node, similar to here: _https://github.com/IntelRealSense/realsense-ros/issues/2386_ with the original control_transfer error index of 768 followed by index 300. The full error message is below. I attempted your solution in the above thread of resetting the camera initially, however this becomes unresponsive, without completing the 'Resetting device...' message. Any suggestions?

[ INFO] [1678050087.057179315]: Initializing nodelet with 4 worker threads.
[ INFO] [1678050087.208928533]: RealSense ROS v2.3.2
[ INFO] [1678050087.208982883]: Built with LibRealSense v2.53.1
[ INFO] [1678050087.209019206]: Running with LibRealSense v2.53.1
[ INFO] [1678050087.247675663]:  
[ INFO] [1678050087.494258613]: Device with serial number 103422073065 was found.

[ INFO] [1678050087.494344624]: Device with physical ID 2-4-10 was found.
[ INFO] [1678050087.494446416]: Device with name Intel RealSense D435 was found.
[ INFO] [1678050087.496437812]: Device with port number 2-4 was found.
[ INFO] [1678050087.496516146]: Device USB type: 2.1
[ WARN] [1678050087.496545516]: Device 103422073065 is connected using a 2.1 port. Reduced performance is expected.
[ INFO] [1678050087.499892334]: getParameters...
[ INFO] [1678050087.564090674]: setupDevice...
[ INFO] [1678050087.564129755]: JSON file is not provided
[ INFO] [1678050087.564156901]: ROS Node Namespace: camera
[ INFO] [1678050087.564197832]: Device Name: Intel RealSense D435
[ INFO] [1678050087.564235680]: Device Serial No: 103422073065
[ INFO] [1678050087.564270754]: Device physical port: 2-4-10
[ INFO] [1678050087.564312491]: Device FW version: 05.13.00.50
[ INFO] [1678050087.564334905]: Device Product ID: 0x0B07
[ INFO] [1678050087.564356185]: Enable PointCloud: Off
[ INFO] [1678050087.564384528]: Align Depth: Off
[ INFO] [1678050087.564402601]: Sync Mode: Off
[ INFO] [1678050087.564462818]: Device Sensors: 
[ INFO] [1678050087.663866210]: Stereo Module was found.
[ INFO] [1678050087.682295590]: RGB Camera was found.
[ INFO] [1678050087.682410952]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1678050087.682460273]: num_filters: 0
[ INFO] [1678050087.682497279]: Setting Dynamic reconfig parameters.
[ INFO] [1678050090.533947953]: Done Setting Dynamic reconfig parameters.
[ INFO] [1678050090.535061899]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1678050090.536504993]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1678050090.536572840]: setupPublishers...
[ INFO] [1678050090.540909654]: Expected frequency for depth = 15.00000
[ INFO] [1678050090.548765432]: Expected frequency for color = 15.00000
[ INFO] [1678050090.554419857]: setupStreams...
 05/03 21:01:30,848 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1678050090.899674989]: SELECTED BASE:Depth, 0
[ INFO] [1678050090.916065351]: RealSense Node Is Up!
 05/03 21:01:31,017 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/03 21:01:31,798 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/03 21:01:31,853 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/03 21:01:31,904 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/03 21:01:32,066 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/03 21:01:32,121 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/03 21:01:32,278 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/03 21:01:32,329 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 05/03 21:01:32,486 WARNING [140055348979456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b

MartyG-RealSense, 2023-03-06:
Does your computer / computing device have a USB 3 port, as the log states that the camera connection type is being detected as the slower USB 2.

Please try launching RViz to see whether the depth and color streams are being published, as sometimes the streams will perform normally in RViz despite the control_transfer returned message continually generating in the log, and those messages can therefore be ignored.

robertburk, 2023-03-06:
Hi @MartyG-RealSense.

I was using the 3.0 port but not a 3.0 cable! Seems to have allowed RViz to view the messages + function correctly, albeit those warning messages are still being published. I am happy with this functionality. Thank you for your help, you can close the issue.

MartyG-RealSense, 2023-03-06:
You are very welcome, @robertburk - it's great to hear that you have achieved a satisfactory outcome!  As you suggested, I will close the issue.  Thanks again!

closed 2023-03-06 by MartyG-RealSense