realsenseai/librealsense #11523: V4L2 extension units to get values + converting raw depth stream to distance values

voxellated, 2023-03-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version | Ubuntu 17 | 
| Platform                                 | PC |
| Language                               | C++ |   

Hi, I am trying to find a way to use V4L2/UVC to directly interface with realsense cameras (to access the streams and configure it manually in my application without using the SDK as I am trying to minimize dependencies and only need it for viz).

Is there a tutorial on how to set/get intrinsic camera parameters such as fx, fy, saturation value, confidence threshold, etc. using UVC's extension units and how to use them to map the raw depth stream to actual distance values?

This use case is uncommon but I would be grateful for any response :)

MartyG-RealSense, 2023-03-06:
Hi @voxellated  If you build the SDK from source code with the CMake build flag **-DFORCE_LIBUVC=OFF** (to build it in V4L2 backend mode) then  you should be able to capture raw data from the camera with standard Linux tools instead of using SDK programming, though it would not be as cleanly colorized as an SDK-generated image).

It is possible that once you can access the raw data streams then you may be able to alter them with **OpenCV** programming instructions, such as its **calibrateMatrixFunctions** command for fx and fy, or splitting the **HSV** values to affect saturation.

https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga87955f4330d5c20e392b265b7f92f691

https://stackoverflow.com/questions/74526153/how-to-increase-the-saturation-values-of-an-image-using-hsv-in-opencv-using-c

Regarding depth values, a RealSense depth stream is in the 16-bit format.  Its raw pixel depth values are in the range 0-65535, and the real-world distance in meters is found by multiplying the raw depth value by the **depth scale** of a particular RealSense camera model, which is **0.000250** on the L515 model. 

When the 16-bit stream is converted for use in OpenCV as an 8-bit **cv::mat** image then the 0-65535 depth values are converted to a range between 0-255.

voxellated, 2023-03-06:
Hey @MartyG-RealSense thank you so much for the speedy response!

To use the **calibrateMatrixFunctions** we first need to have the intrinsic camera matrix, right? Is there a way to get it without using the SDK?

We also work with another realsense camera (ToF but I am not quite sure what model that is, it's not here currently) and I am hoping to be able to make it general - is there a way to get the values here https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20 (intrinsic/extrinsic parameters) and maybe also FPS without using the SDK? Since I think realsense uses the same stream formats for each sensor types. 

Can one also get these informations from reading a file? Maybe something like this https://github.com/IntelRealSense/realsense-ros/issues/709 but for linux and not for ROS? 

Thank you again for your help!

MartyG-RealSense, 2023-03-06:
In OpenCV you can find the intrinsics of a camera using a **calibration chessboard**.

https://docs.opencv.org/4.x/dc/dbb/tutorial_py_calibration.html

Whilst different RealSense cameras typically use similar stream formats, the resolution and frames per second (FPS) speeds that a particular model supports may vary.  In general, 640x480 and 30 FPS should be valid resolution / FPS settings for all models and depth / color stream types.

It sounds as though the ToF camera will be a RealSense L515 lidar depth camera.

Each individual RealSense camera unit will have its own unique intrinsic and extrinsic values.  The easist approach may be to do a one-off reading of each camera's calibration values with the SDK's 'rs-enumerate-devices' tool with the Ubuntu terminal command below, as the intrinsics and extrinsics of a particular camera will not change unless the camera is deliberately re-calibrated.

**rs-enumerate-devices -c**

voxellated, 2023-03-06:
I think I have information for now, thanks for the support :)

closed 2023-03-06 by voxellated