realsenseai/librealsense #11525: librealsense make -j5 error

tsaiboyu, 2023-03-06:
HI everyone , I have a problem compiling librealsense
I used jetson nano Xvaier nx

[  2%] Built target libcurl
[  3%] Built target fw
[  8%] Built target realsense-file
[ 41%] Built target realsense2
[ 46%] Built target pybackend2
[ 50%] Built target pyrealsense2
[ 55%] Built target glfw
[ 59%] Built target realsense2-gl
[ 59%] Built target rs-hello-realsense
[ 60%] Built target rs-software-device
[ 61%] Built target rs-capture
[ 61%] Built target rs-callback
[ 62%] Built target rs-save-to-disk
[ 62%] Built target rs-multicam
[ 63%] Built target rs-pointcloud
[ 64%] Built target rs-align
[ 65%] Built target rs-align-advanced
[ 65%] Built target rs-sensor-control
[ 66%] Built target rs-measure
[ 66%] Linking C executable rs-depth
../../../librealsense2.so.2.53.1: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
../../../librealsense2.so.2.53.1: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
../../../librealsense2.so.2.53.1: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
collect2: error：ld return 1
examples/C/depth/CMakeFiles/rs-depth.dir/build.make:101: recipe for target 'examples/C/depth/rs-depth' failed
make[2]: *** [examples/C/depth/rs-depth] Error 1
CMakeFiles/Makefile2:1221: recipe for target 'examples/C/depth/CMakeFiles/rs-depth.dir/all' failed
make[1]: *** [examples/C/depth/CMakeFiles/rs-depth.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

MartyG-RealSense, 2023-03-06:
Hi @tsaiboyu  If a build fails when using a '-j' value (which accelerates build time by enabling the use of multiple CPU threads) then you can simply use **make** - which can have a much longer build time but be more stable.

tsaiboyu, 2023-03-09:
@MartyG-RealSense HI i used "make -" get same result 
/catkin_ws/src/librealsense/build$ make -
[  2%] Built target libcurl
[  3%] Built target fw
[  8%] Built target realsense-file
[ 41%] Built target realsense2
[ 46%] Built target pybackend2
[ 50%] Built target pyrealsense2
[ 55%] Built target glfw
[ 59%] Built target realsense2-gl
[ 59%] Built target rs-hello-realsense
[ 60%] Built target rs-software-device
[ 61%] Built target rs-capture
[ 61%] Built target rs-callback
[ 62%] Built target rs-save-to-disk
[ 62%] Built target rs-multicam
[ 63%] Built target rs-pointcloud
[ 64%] Built target rs-align
[ 65%] Built target rs-align-advanced
[ 65%] Built target rs-sensor-control
[ 66%] Built target rs-measure
[ 66%] Linking C executable rs-depth
../../../librealsense2.so.2.53.1: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
../../../librealsense2.so.2.53.1: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
../../../librealsense2.so.2.53.1: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
collect2: 錯誤：ld 回傳 1
examples/C/depth/CMakeFiles/rs-depth.dir/build.make:101: recipe for target 'examples/C/depth/rs-depth' failed
make[2]: *** [examples/C/depth/rs-depth] Error 1
CMakeFiles/Makefile2:1221: recipe for target 'examples/C/depth/CMakeFiles/rs-depth.dir/all' failed
make[1]: *** [examples/C/depth/CMakeFiles/rs-depth.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

tsaiboyu, 2023-03-09:
Before, i have deleted build and mkdir build again

MartyG-RealSense, 2023-03-09:
Can you link me to the instructions that you are using to build librealsense from source code, please?  For example, is it the Jetson-specific instructions at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

MartyG-RealSense, 2023-03-15:
Hi @tsaiboyu  Do you require further assistance with this case, please?  Thanks!

tsaiboyu, 2023-03-15:
@MartyG-RealSense i'm Taiwanese,my english is not good 
i used this website command : https://blog.cavedu.com/2021/06/04/python1-nvidia-jetsonnano-realsense-d435/?fbclid=IwAR0ektOeF6jMRBAzxxELXsZBy-bQ75a8S8wKd-ug3Bar8SIsrznmibK42pE

MartyG-RealSense, 2023-03-15:
Your English is good, don't worry.  :)  If you would prefer to use your native language though then that is okay as I can translate.

When building with CMake, build problems may occur if your **gcc** version is older than 5.0.  you can check your current gcc version with the command **gcc -v**

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk

tsaiboyu, 2023-03-15:
Hi @MartyG-RealSense  
my gcc 
Use built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/aarch64-unknown-linux-gnu/11.2.0/lto-wrapper
Purpose: aarch64-unknown-linux-gnu
The configuration is: ../configure -enable-checking=release -enable-languages=c,c++ -disable-multilib
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 11.2.0 (GCC)

MartyG-RealSense, 2023-03-15:
At https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 there is an alternative set of instructions for building from source code on Jetson that works well on different Jetson models including NX.

tsaiboyu, 2023-03-15:
ok,I try it!

tsaiboyu, 2023-03-16:
HI @MartyG-RealSense 
I used "make -j" can run 100% ,but still error.
the problem:
[ 95%] Built target rs-pose-predict
[ 95%] Linking CXX executable rs-record
[ 95%] Built target rs-pose-and-image
[ 96%] Linking CXX executable rs-fw-logger
[ 96%] Built target rs-record
[ 97%] Linking CXX executable rs-gl
[ 97%] Built target rs-fw-logger
[ 97%] Linking CXX executable rs-fw-update
[ 97%] Linking CXX executable rs-ar-basic
[ 97%] Linking CXX executable rs-record-playback
[ 97%] Built target rs-gl
[ 98%] Linking CXX executable rs-trajectory
[ 99%] Linking CXX executable rs-motion
[ 99%] Linking CXX executable rs-tracking-and-depth
[ 99%] Linking CXX executable rs-post-processing
[ 99%] Built target rs-fw-update
[ 99%] Linking CXX executable rs-enumerate-devices
[100%] Linking CXX executable rs-ar-advanced
[100%] Built target rs-ar-basic
[100%] Linking CXX executable rs-hdr
[100%] Built target rs-record-playback
[100%] Built target rs-trajectory
[100%] Built target rs-motion
[100%] Built target rs-tracking-and-depth
[100%] Built target rs-enumerate-devices
[100%] Built target rs-post-processing
[100%] Built target rs-ar-advanced
[100%] Built target rs-hdr
[100%] Linking CXX executable rs-convert
[100%] Linking CXX executable rs-benchmark
[100%] Built target rs-convert
[100%] Built target rs-benchmark
[100%] Linking CXX executable rs-terminal
[100%] Built target rs-terminal
[100%] Linking CXX executable rs-data-collect
[100%] Built target rs-data-collect
[100%] Linking CXX executable rs-rosbag-inspector
[100%] Built target rs-rosbag-inspector
[100%] Linking CXX executable rs-depth-quality
[100%] Linking CXX executable realsense-viewer
[100%] Built target rs-depth-quality
[100%] Built target realsense-viewer
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

MartyG-RealSense, 2023-03-16:
It may be worth trying the installation script **installLibrealsense** provided by the website **JetsonHacks** instead to build the SDK from packages instead of compiling source code.

https://jetsonhacks.com/2019/12/22/install-realsense-camera-in-5-minutes-jetson-nano/

tsaiboyu, 2023-03-17:
@MartyG-RealSense I need "pyrealsense2 module" ,is there any other way to get it?

MartyG-RealSense, 2023-03-17:
Using the JetsonHacks **buildLibrealsense** script to build librealsense from source code instead of packages includes the instruction to build the pyrealsense2 wrapper.

https://github.com/jetsonhacks/installRealSenseSDK#buildlibrealsensesh

MartyG-RealSense, 2023-03-23:
There was a past case of that /usr/local/cuda/bin/nvcc error at https://github.com/IntelRealSense/librealsense/issues/7229

MartyG-RealSense, 2023-03-29:
Hi @tsaiboyu  Was the comment above useful to you, please?  Thanks!

closed 2023-04-10 by MartyG-RealSense