realsenseai/librealsense #11530: L515 Camera point_cloud and aligned_depth map frames drop (No stream match for pointcloud chosen texture Process - Color)

julienfal, 2023-03-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (1.01.05.08.01) | 
| Operating System & Version |   { Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.53.1}                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
I'm using the **L515** realsense camera for recording into a rosbog the RGB images (topic: /camera/color/image_raw) with the associated point cloud(/camera/depth/color/points) or aligned depth map (Camera/aligned_depth_to_color/image_raw). 
It works well but during a record of 1 min there is a kind of "Frame drop" and if the record is longer we got multiples frames drop approximately one per minutes. 

First problem: When we are recording the pointcloud at first everything goes well, but at a random moment we got the message: No stream match for pointcloud chosen texture Process - Color and that cause like a 10 frames drop for the point cloud that means that for 10 rgb frames we don't have any point cloud that are taken.

Second problem but maybe related to the first one: When we record using the arg align_depth:= true we got the same problem but the only things that change is that we don't have any warning message 

This is the parameters that we are using for the point cloud when we use **roslaunch realsense2_camera rs_camera.launch** filters:=pointcloud

PARAMETERS
* /camera/realsense2_camera/accel_fps: -1
* /camera/realsense2_camera/accel_frame_id: camera_accel_frame
* /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
* /camera/realsense2_camera/align_depth: False
* /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
* /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
* /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
* /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
* /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
* /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
* /camera/realsense2_camera/allow_no_texture_points: False
* /camera/realsense2_camera/base_frame_id: camera_link
* /camera/realsense2_camera/calib_odom_file: 
* /camera/realsense2_camera/clip_distance: -2.0
* /camera/realsense2_camera/color_fps: -1
* /camera/realsense2_camera/color_frame_id: camera_color_frame
* /camera/realsense2_camera/color_height: -1
* /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
* /camera/realsense2_camera/color_width: -1
* /camera/realsense2_camera/confidence_fps: -1
* /camera/realsense2_camera/confidence_height: -1
* /camera/realsense2_camera/confidence_width: -1
* /camera/realsense2_camera/depth_fps: -1
* /camera/realsense2_camera/depth_frame_id: camera_depth_frame
* /camera/realsense2_camera/depth_height: -1
* /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
* /camera/realsense2_camera/depth_width: -1
* /camera/realsense2_camera/device_type: 
* /camera/realsense2_camera/enable_accel: False
* /camera/realsense2_camera/enable_color: True
* /camera/realsense2_camera/enable_confidence: True
* /camera/realsense2_camera/enable_depth: True
* /camera/realsense2_camera/enable_fisheye1: False
* /camera/realsense2_camera/enable_fisheye2: False
* /camera/realsense2_camera/enable_fisheye: False
* /camera/realsense2_camera/enable_gyro: False
* /camera/realsense2_camera/enable_infra1: False
[... 82 lines cut ...]
[ INFO] [1678102327.360511192]: Enable PointCloud: On
[ INFO] [1678102327.360590164]: Align Depth: Off
[ INFO] [1678102327.360675361]: Sync Mode: On
[ INFO] [1678102327.360847291]: Device Sensors: 
[ INFO] [1678102327.361707807]: L500 Depth Sensor was found.
[ INFO] [1678102327.365257533]: RGB Camera was found.
[ INFO] [1678102327.365524515]: Motion Module was found.
[ INFO] [1678102327.365632274]: Add Filter: pointcloud
[ INFO] [1678102327.366626651]: num_filters: 1
[ INFO] [1678102327.366656389]: Setting Dynamic reconfig parameters.
[ INFO] [1678102327.494830300]: Done Setting Dynamic reconfig parameters.
[ INFO] [1678102327.498622661]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1678102327.501062923]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1678102327.501153952]: setupPublishers...
[ INFO] [1678102327.510460813]: Expected frequency for depth = 30.00000
[ INFO] [1678102327.604335297]: Expected frequency for color = 30.00000
[ INFO] [1678102327.663150045]: setupStreams...
[ INFO] [1678102327.727845113]: SELECTED BASE:Depth, 0
[ INFO] [1678102327.929416785]: RealSense Node Is Up!
[ WARN] [1678102327.931582246]: 
[ WARN] [1678102327.931725496]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
[ INFO] [1678102477.668086942]: publishing unordered pointcloud.
[ WARN] [1678102515.212162012]: No stream match for pointcloud chosen texture Process - Color

Is someone have any idea for resolving that problems?

MartyG-RealSense, 2023-03-06:
Hi @julienfal  You are correct that the message **No stream match for pointcloud chosen texture Process - Color**
coincides with frame drops, as advised at https://github.com/IntelRealSense/realsense-ros/issues/588#issuecomment-457890773 by the Intel RealSense team member who developed the ROS1 wrapper.

You could try defining a roslaunch instruction with a custom stream configuration that has a lower resolution and FPS to see whether this reduces the frequency of the warnings.  For example:

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=640 depth_height:=480 depth_fps:=15 color_width:=640 color_height:=480 color_fps:=15`

If you are using the pointcloud filter then you should not need to use align_depth:=true as the pointcloud should automatically map color onto the depth coordinates anyway.  Having both the pointcloud filter and align_depth enabled may cause performance issues.

julienfal, 2023-03-06:
Thank you for your quick response.
Yes of course we didn't enable the point cloud and align_depth at the same time. 
However, the depth fps cannot be changed to less than 30 for this L515 camera.
We tried with your parameters but with 30 fps for the depth frame but we still have the frame drop.
Is another way to change the fps of the depth sensor, maybe with a python script ?

MartyG-RealSense, 2023-03-06:
You are correct about the 30 FPS limit for the depth stream on the L515 camera.  My apologies, thanks for pointing that out.

The rate at which messages are published can be modified in ROS by setting a **rate**, though doing so may not be necessary for this particular problem.

https://roboticsbackend.com/ros-rate-roscpy-roscpp/#:~:text=ROS%20Rate%20explained,-Let's%20start%20with

Does performance after a minute of recording improve if you add **initial_reset:=true** to your roslaunch instruction to reset the camera at launch?

Does it improve if you add **global_time_enabled:=true** to enable Global Time?

julienfal, 2023-03-06:
It seems that after the initial_reset command now the point cloud is correctly publish during at least 53 sec and can go to 1 min without error but still have the error after that time ( I made the test approximately 10 times). 
After doing a launch with the initial_reset I launch without that but the point cloud still good until 53 sec, we got that because we ran with the reset before ? 
The global_time_enabled changes nothing to the problem.
if we can reach 1min 20 without any problem it would be perfect but I don't know if it's possible. I also tried that with a lower camera resolution but that changes nothing.

MartyG-RealSense, 2023-03-06:
Are you recording the rosbag in ROS using the **rosbag record** function please?  This function has a parameter called **--split** that can be used to set a maximum recording duration or file size.  When that defined limit is reached, the recording is automatically closed and a new bag recording automatically starts and continues onwards until the next time that the limit is reached.  

If having more than one bag file is acceptable to you then closing and re-opening the recording process may correct the loss of frames that you are experiencing at the one minute point and allow a longer period to be recorded across multiple bags.

Examples of --split commands for time and file-size are provided below.

```
--split

Split the bag when maximum size or duration is reached

$ rosbag record --split --size=1024 /chatter
$ rosbag record --split --duration=30 /chatter
$ rosbag record --split --duration=5m /chatter
$ rosbag record --split --duration=2h /chatter
```

julienfal, 2023-03-06:
Okay I'll try that, my command line for recording was : rosbag record -b 0 --duration=70 /topics

julienfal, 2023-03-06:
I tried and it stills dropping around 58 sec, do you think it could be an hardware problem in the camera? Like a setting to change?

MartyG-RealSense, 2023-03-06:
It sounds like it could be a **memory leak**, where a computer's available memory capacity is progressively consumed over time during the running of an application until that application becomes unstable and then may freeze / crash.  You can test for a memory leak in Ubuntu by using a system monitoring tool such as **htop** to monitor available memory over time during the recording process to see if the free memory is declining.

julienfal, 2023-03-06:
It seems that the memory stays ok during all the process at first before starting the launch and the recording the memory is at 3.5G/15.6G and at during it goes to 4.20G/15.6G.
I'm stopping for today but I'll be able tomorrow to try other things. Thanks for all your answers today !

MartyG-RealSense, 2023-03-06:
You are very welcome.  I look forward to your next update.  Good luck!

julienfal, 2023-03-07:
Hi!
A little Resume from yesterday : As you told me when I add initial_reset:=true the point cloud last at least 53 sec. 
But I still don't know why it stopped.
All the test that we made yesterday was with these lines adding in the launch file
- <rosparam> /camera/rgb_camera/auto_exposure_priority: false </rosparam>
- <rosparam> /camera/rgb_camera/auto_exposure: false </rosparam>
- <rosparam> /camera/rgb_camera/exposure: 200 </rosparam>

I made some test this morning when i try different launch file configuration here are my tests:

**1st test: I remove all the 3 lines**
- Consequences: I don't have any frame drop for the point cloud but i can't have a stable frame fps even if I launch the camera with **roslaunch realsense2_camera rs_camera.launch filters:=pointcloud color_width:=1280 color_height:=720 color_fps:=30 initial_reset:=true** it looks like at first it stays at 30 fps for 3 sec and after it just goes to 15 fps naturally 

**2nd test: I just add  <rosparam> /camera/rgb_camera/auto_exposure_priority: false </rosparam> in the launch file** 

- Consequences: the Frame rate is now consistent when I want 30 fps it stays at 30 but I have the frame drop much more quickly like around 20 sec and even sometimes in less than 10 sec 
- If i try using 15 fps everything works well, no frames drop **but** i think 15 fps is too low for my application so i really need 30 fps

**3th test: I just add <rosparam> /camera/rgb_camera/auto_exposure: false </rosparam> <rosparam> /camera/rgb_camera/exposure: 200 </rosparam> in the launch file** 

- Consequences: The frame rate is stable at 30 fps but point cloud frame drop around after 53 sec

**4th test: I just add <rosparam> /camera/rgb_camera/exposure: 200 </rosparam> in the launch file**

- Consequences : Frame rate stable at 30 fps but point cloud drop and still good for 15 fps

**5th test : I just add <rosparam> /camera/rgb_camera/auto_exposure_priority: false </rosparam>
<rosparam> /camera/rgb_camera/auto_exposure: false </rosparam> in the launch file** 

- Consequences: with 30 fps point cloud drop much more quickly around 25 sec, with 15 fps point cloud still good 

All of this test was launch with the initial_reset:=true each time.

I don't really get it because normally the Camera should be able to run the point cloud with 30 fps according to the datasheet and I don't think that the problems come from my PC. Do you have other idea that I can test to see if that improve something ? It looks like setting manually the exposure in the launch file help already to stabilize the point cloud.

MartyG-RealSense, 2023-03-07:
If depth and color streams are enabled and auto-exposure is enabled then it can result in a drop in FPS, such as a halving.  A fix for this is to set auto-exposure to manual and disable an RGB option called **auto_exposure_priority**.  This forces the SDK to try to maintain the FPS at a constant rate instead of being permitted to vary, as described at https://github.com/IntelRealSense/librealsense/issues/10051

julienfal, 2023-03-07:
When you say auto-exposure to manual it just putting that settings equal to False and after giving a manual exposure value? If yes I already tried that but still have the frame drop for the point cloud after 50 sec.

julienfal, 2023-03-07:
it seems that it can be a problem of exposure time because when I change the value of that this impact a lot the frame drop. 
**What is the best exposure time for 30 FPS ?** Actually it is set to 200 and when i try to go to 400 i loose the consistent fps speed but i have the consistent point cloud and when i try to go lower to 200 I have the consistent fps speed but i still have the point cloud frame drop

MartyG-RealSense, 2023-03-07:
Using a manual exposure can cause FPS to drop if the value is within a certain range, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

MartyG-RealSense, 2023-03-07:
Depth exposure is not customizable on L515 lidar depth cameras like it is on the 400 Series stereo cameras as the L515 is a different camera technology.  L515 uses a short exposure time of <100 ns per depth point.

RGB exposure is customizable though and is '156' by default, so 156 would be the optimal exposure time for RGB.  With the 200 and 400 values that you are using, it sounds as though it is RGB exposure that you are configuring.

closed 2023-03-21 by MartyG-RealSense