realsenseai/librealsense #11532: Syncing D435 Camera to Microcontroller

SResor, 2023-03-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435i) | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version | Windows 10 | 
| Kernel Version (Linux Only)    |  N/A | 
| Platform                                 | PC |
| SDK Version                           | N/A | 
| Language                               | Python | 
| Segment			        | Microcontroller | 

### Issue Description
I'm currently trying to build a device that utilizes a microcontroller and a D435i camera in a master/slave configuration. What I would like to do is to have the microcontroller send out a signal to the camera via the VSYNC pin (pin 5) on the external sensor sync connector and then have it capture and save a single image with a preset exposure time. Where I'm having trouble is how to code the camera to capture an image based on an input signal. All the code I've seen for multi-camera configurations sets the cameras as either master or slave using ds.set_option() and then they automatically synchronize so long as the FPS is correct.

My question is whether it's possible to treat that VSYNC pin almost like a GPIO input, so that I can use it as a trigger in my code?

MartyG-RealSense, 2023-03-07:
Hi @SResor  What you are seeking to achieve sounds like what the **External Synchronization** method of RealSense hardware sync provides: the ability to send a trigger signal from an external micro-controller to the sync pin of a RealSense slave camera.

This sync method is no longer recommended by Intel as it was an experimental proof-of-concept rather than an official, validated feature.  However, it can still be used and the details of it can be found in the PDF document at the link below.

[External Synchronization.pdf](https://github.com/IntelRealSense/librealsense/files/10908356/External.Synchronization.pdf)

SResor, 2023-03-10:
Unfortunately I'm running into a problem where the camera doesn't wait to acquire images. I have inter_cam_sync_mode set to 4, but the camera still acquires images even when there is no trigger signal present on pin 5 of the external SYNC connector. Sometimes it will wait, but whenever it's connected to a larger circuit it will acquire images regardless of whether the trigger signal occurs or not.

MartyG-RealSense, 2023-03-11:
When you say a 'larger' circuit, do you mean longer sync cabling, please?  With longer-distance cabling, there is an increased risk of an electrostatic discharge (ESD) building up in the cable and then discharging in a shock-pulse.  If there is ESD occurring then perhaps that is triggering the sync pin when a trigger pulse has not been transmitted.  Installing anti-ESD components into the cable (as described in the hardware sync white-paper guide document) can help to avoid such discharges.

SResor, 2023-03-16:
> Hi @SResor <https://github.com/SResor> Do you require further assistance
> with this case, please? Thanks!

Hello,

I managed to get it syncing correctly. You were correct, it was an ESD
issue. Thank you very much for your help.

All the best,
Sean

MartyG-RealSense, 2023-03-16:
You are very welcome.  That's excellent news - thanks very much for the update!  As you achieved a solution, I will close this case.  Thanks again!

closed 2023-03-16 by MartyG-RealSense