realsenseai/librealsense #11533: Intel real sense camera not working.

yadavashwani467, 2023-03-07:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|
|**Name**|Platform Camera|
|**Serial Number**|5f5dc48|
|**Usb Type Descriptor**|2.01|

Please provide a description of the problem
Intel Real sense Lidar Camera L515 is not working. When I am connecting to computer no video output is available on the intel real sense viewer window. Also Depth camera option is also missing.

MartyG-RealSense, 2023-03-07:
Hi @yadavashwani467  Could you first try unplugging the micro-sized end of the USB cable from the base of the camera, reversing the connector's orientation and plugging it back into the L515 camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

Also, which firmware driver version is installed on the camera please?  You can find this information by clicking on the 'Info' button at the top of the Viewer's options side-panel.  Thanks!

yadavashwani467, 2023-03-07:
Hi Marty,
    I have tried both the things but the problem is that the viewer is not
able to recognize the device as I have to add my device using Add source
option. Also in my device manager the camera have yellow flag over it. I
have attached the screenshots for your reference. Please reply with a
suitable solution to my issue.
                                          Thanks and Regards
__
Ashwani Kumar

MartyG-RealSense, 2023-03-07:
Images that are sent by email are not displayed on an issue's web-page.  Please add them on the page using the comment writing box.

I note that your listed USB type is 2.0.  The L515 model originally was USB 3 only, with USB 2 support added when the firmware driver is updated.  Does your computer have a USB 3 port that you can try the camera in?

yadavashwani467, 2023-03-07:
[image]
[image]
Almost all the USB ports in my computer are USB 3.0

yadavashwani467, 2023-03-07:
Also I am not able to figure out how can we update the firmware of our device. Since, the device is not showing in intel real sense viewer.

yadavashwani467, 2023-03-07:
Still the same situation is there. The drivers again installed in other devices section. Do I need to attach the USB cable that is provided with the device or any type C cable will work?

MartyG-RealSense, 2023-03-07:
I would recommend attaching the cable that is provided with the device if you have not done so already, as some Type-C cables can have quality issues that may affect the ability of a RealSense camera to function correctly.

MartyG-RealSense, 2023-03-15:
Hi @yadavashwani467  Do you require further assistance with this case, please?  Thanks!

yadavashwani467, 2023-03-15:
Yes, I will contact you once I get hold of the original cable that came
with the camera.
__
Ashwani

MartyG-RealSense, 2023-03-15:
Thanks very much for the update.  I look forward to your next report.  Good luck!

MartyG-RealSense, 2023-03-27:
Hi @yadavashwani467  Have you been able to obtain the original cable of your RealSense camera, please?  Thanks!

yadavashwani467, 2023-03-27:
Yes, I did. And it is working now. Thanks for your help.
___
Ashwani

MartyG-RealSense, 2023-03-27:
You are very welcome.  It's excellent news that it is working.  Thanks very much for the update!

closed 2023-03-27 by MartyG-RealSense