realsenseai/librealsense #11535: alignment depth values are not correct?

Ahmed-M-Naguib, 2023-03-07:
i have a question about the logic in https://github.com/IntelRealSense/librealsense/blame/e9f05c55f88f6876633bd59fd1cb3848da64b699/src/proc/align.cpp#L55

we are setting the depth value of the original depth image. that is the distance from the point to depth camera along depth camera Z axis.

but, the result should be depth map in `other` camera frame. shouldn't we set the depth value to be `other_point[2]` (possibly an interpolation between the 2 corners of the pixel)?

at the moment, deprojecting the generated depth image using color camera intrinsic is going to generate wrong point cloud and mapping back to depth camera will generate different depth image

of course, for realsense, the angle between depth and color cameras' Z axes is nearly 0 and the translation is nearly purely a baseline along X-axis. so, the error is very small

but the current alignment logic is not correct at the moment if my understanding is correct. or maybe i misunderstand the meaning/purpose of alignment?

MartyG-RealSense, 2023-03-07:
Hi @Ahmed-M-Naguib  Alignment in the RealSense SDK can map depth coordinates onto color coordinates (depth to color alignment) or map color coordinates onto depth coordinates (color to depth alignment).  

The SDK uses an **Align Processing Block** to resize the field of view (FOV) size to the FOV size of the target stream, and also make automatic adjustments for differences in resolution between the two stream types that are being aligned.  For example, if depth to color alignment is used then the depth FOV size is resized to the color FOV size.

When using deprojection, alignment should be performed **first** and then deprojection performed **second** afterwards.

You should not need to think about the mechanics of alignment because the align processing block takes care of the calculations for you.

Further information about deprojection can be found in Intel's **Projection, Texture-Mapping and Occlusion** white-paper guide at the link below.

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#23-de-projection-to-world-coordinates

Ahmed-M-Naguib, 2023-03-07:
hi @MartyG-RealSense , thanks for the quick answer. i was actually wondering about the correctness of the implementation, not a specific usecase/application. it looks to me like a bug

what `align_images` does is (ignoring pixel corner logic):
- deproject depth pixel to point cloud
- transform the point to `other` camera frame
- project to `other` image

[image]

let's suppose we deproject the `aligned` image back to point cloud:

[image]

as you can see, it is the 3D point in color camera coordinate. i.e. the `aligned` depth is suppose to be the z of the transformed point: `other_point[2]`

i hope it is clear now

MartyG-RealSense, 2023-03-07:
A few months ago there was a similar question about the correctness of the align logic.  I discussed it with the RealSense development team and their conclusion was that although there may appear to be a discrepancy in the logic, the actual implementation of alignment was likely to be fine.

In the majority of cases, the SDK's alignment has worked relatively correctly for the past 5 years since the SDK was launched.

It is true that when alignment is used with deprojection, a small amount of inaccuracy can be introduced into the deprojection by the alignment.

Ahmed-M-Naguib, 2023-03-07:
@MartyG-RealSense i see. thanks for the clarification and quick response

closed 2023-03-07 by Ahmed-M-Naguib