realsenseai/librealsense #11539: Device properties using linux backend

voxellated, 2023-03-07:
Hi, I think I found what I need - If I am not wrong, this https://github.com/IntelRealSense/librealsense/blob/b874e42685aed1269bc57a2fe5bf14946deb6ede/src/linux/backend-v4l2.cpp#L1564 is basically equivalent to https://github.com/opencv/opencv/blob/0052d46b8e33c7bfe0e1450e4bff28b88f455570/modules/videoio/src/cap_obsensor/obsensor_stream_channel_v4l2.cpp#L330, but for realsense. And I assume that one could get the camera parameters through the v4l2 extension units (Xu) like the linked OpenCV orbbec sensor for a video capture device. 

Is there a documentation that would tell us which control code gives us which parameter? I was not able to find it. In a nutshell, I am trying to do something similar to the OpenCV implementation, so I would be grateful if there is any documentation for the v4l2 backend implementation.

Thank you!

MartyG-RealSense, 2023-03-07:
Hi @voxellated  The camera can be queried with **v4l2-ctl** with a list of commands at the link below.

https://manpages.ubuntu.com/manpages/bionic/man1/v4l2-ctl.1.html

At https://github.com/IntelRealSense/librealsense/issues/3416#issuecomment-471720605 a RealSense team member provides guidance about querying RealSense devices with v4l2-ctl.

************

By default in the RealSense SDK, hardware commands are dispatched through the extension-unit (XU) transfer protocol because the CMake build flag **HWM_OVER_XU** is True by default.  HWM stands for Hardware Monitor and is the protocol for controlling all currently supported RealSense cameras.

voxellated, 2023-03-08:
Thank you for the quick response. I haven't been able to test this yet since I'm not in the lab until the 10th and I'm worried that the question will be closed soon - could you maybe confirm shortly if one would be able to get the values of rs2::intrinsic and/or rs2::extrinsic through the extension-unit?

MartyG-RealSense, 2023-03-08:
It's no problem to keep this issue open past the 10th now that you have made that request.

In regard to intrinsics, for most RealSense cameras the 5 distortion coefficients will all have a value of '0', as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/1430

fx and fy represent pixel width and height, so it may be worth checking whether **v4l2-ctl --all** provides that information in terms of pixel width and pixel height instead of fx and fy.

voxellated, 2023-03-10:
Hi, this has helped out a lot, thanks!

closed 2023-03-10 by voxellated