realsenseai/librealsense #11540: HDR(or other method) with alternating IR projector

echoGee, 2023-03-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 

### HDR(or other method) with alternating IR Projector
We are trying to use d435 for an application where we are trying to use the IR image with one set of setting(Set 1) and the depth image with another set of setting (Set 2)
Set 1
- Exposure 1
- IR light off
Set 2 
- Exposure 2
- IR light on

Is there a way to make it work such that we are able to set it such that it alternates between these two settings ?

MartyG-RealSense, 2023-03-08:
Hi @echoGee  You can certainly control the on-off status of the emitter with scripting, so that Set 1's instructions could disable the emitter and Set 2's instructions could enable it.  Setting the emitter to '0' disables it and '1' enables it.

**Python**

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()
pipeline_profile = pipeline.start(config)
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
if depth_sensor.supports(rs.option.emitter_enabled):
depth_sensor.set_option(rs.option.emitter_enabled, 0)
```

**C++*

```
rs2::pipeline pipe; 
rs2::pipeline_profile selection = pipe.start();
rs2::device selected_device = selection.get_device();
auto depth_sensor = selected_device.first<rs2::depth_sensor>();

if (depth_sensor.supports(RS2_OPTION_EMITTER_ENABLED))
{
    depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 1.f); // Enable emitter
    depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 0.f); // Disable emitter
}
if (depth_sensor.supports(RS2_OPTION_LASER_POWER))
{
    // Query min and max values:
    auto range = depth_sensor.get_option_range(RS2_OPTION_LASER_POWER);
    depth_sensor.set_option(RS2_OPTION_LASER_POWER, range.max); // Set max power
    depth_sensor.set_option(RS2_OPTION_LASER_POWER, 0.f); // Disable laser
}
```

echoGee, 2023-03-08:
I see this code can set the emitter on or off. But how could you reliably turn it on or off alternatively between every frame ?

MartyG-RealSense, 2023-03-08:
The SDK has an automated function for turning it on-off on a per-frame basis. 
 https://github.com/IntelRealSense/librealsense/pull/3066 has information about this mode.  The page has an animated image with a fast-strobing light, so be aware if you have epileptic sensitivity.

echoGee, 2023-03-08:
I understand there is an emitter on off functionality. But, what I want is the exposure to also change along with the on.off.

See the configs in https://github.com/IntelRealSense/librealsense/issues/11540#issue-1614437218

MartyG-RealSense, 2023-03-08:
The emitter pulses in line with exposure on the D435i camera model.  So the emitter's timing follows the exposure rather than exposure being changed by the emitter.

echoGee, 2023-03-08:
:) . I think I didn't explain properly. I want the first set to have an exposure value of let's say 100 microsecond and second set to have an exposure of thousand microsecond. I want the first set to have IR lights on and the second set to have IR lights off.

MartyG-RealSense, 2023-03-08:
A change in exposure value can take up to several frames to complete from start to finish, so it would not be practical to alternate the value on a per-frame basis.  Only the on / off status can alternate per-frame.

echoGee, 2023-03-08:
But hdr is able to set two different exposures alternatively, right?

closed 2023-03-21 by MartyG-RealSense