realsenseai/librealsense #11547: 3d reconstruction using realsense

TheNemo05, 2023-03-09:
i need to create 3D reconstruction using the Depth data captured by D455, i guess it will require co-ordinate data, pixel information etc. Do realsense have a standard procedure to create such 3D reconstruction using realsense.

TheNemo05, 2023-03-09:
do you have any examples for how to use 3D reconstruction with Open 3d realsense

MartyG-RealSense, 2023-03-09:
There are not many references about reconstruction with RealSense in Open3D other than the tutorial that I linked to above, and also https://github.com/isl-org/Open3D/issues/2467 and https://github.com/isl-org/Open3D/issues/5082 and the link below.

https://blog.georgmill.de/2020/09/01/3d-reconstruction-with-open3d-on-a-jetson-nano/

In addition, Open3D's official documentation has a non-RealSense section about its **reconstruction system**.

http://www.open3d.org/docs/release/tutorial/reconstruction_system/index.html

TheNemo05, 2023-03-13:
I am still not able to find sufficient document of open3D with realsense for depth 3D reconstruction.

MartyG-RealSense, 2023-03-13:
You could also look at https://github.com/isl-org/Open3D/issues/473

TheNemo05, 2023-03-14:
according to this comment https://github.com/isl-org/Open3D/issues/473#issuecomment-508139946 will the camera capture the entire 3D plot wherever the camera is being placed ?

MartyG-RealSense, 2023-03-14:
The camera will capture whatever is within its **field of view** (imagine looking through the viewfinder of a photographic camera).  Like a photographic camera, a RealSense camera has a **field of view size**.  Detail in the observed scene that is outside of the field of view box will not be captured unless the camera is turned to bring that outer detail into the field of view.

TheNemo05, 2023-03-15:
ok this is not working, do we have any procedure to export pointclouds files from .bag file

MartyG-RealSense, 2023-03-15:
The RealSense SDK's **rs-convert** tool can convert bag files to pointcloud **.ply** files.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

If you are using Windows and have installed the full RealSense SDK, instructions for launching the rs-convert tool can be found here:

https://support.intelrealsense.com/hc/en-us/community/posts/360051132573/comments/360014080493

If you are using Linux and have installed librealsense with the examples and tools included, rs-convert should be in the **/usr/local/bin** location and launchable with the command **rs-convert**

closed 2023-03-16 by TheNemo05