realsenseai/librealsense #11548: realsense viewer sometimes unable receive imu data

Karin-Sugi, 2023-03-09:
Hi, I'm using realsense d435i with Ubuntu18.04 on RaspberryPi.
I installed librealsense with following commands because using `sudo apt install` command didn't work as [document](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) says.
```
$ git clone https://github.com/IntelRealSense/librealsense.git
$ cd librealsense/
$ sudo apt install -y git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
$ sudo apt install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
$ sudo apt install -y cmake
$ ./scripts/setup_udev_rules.sh
$ ./scripts/patch-realsense-ubuntu-lts.sh
$ mkdir build && cd build
$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
$ sudo make clean && make && sudo make install
```
I can see depth&RGB image correctly. But about imu data, viewer says "Nothing is streaming! toggle to start" even when the toggle is ON. Also, when i motion module turns on, everything (include depth&RGB image) dissapear from the viewer and says "Nothing is streaming!"

error messages are below
```
WARNING (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
ERROR(synthetic-stream.cpp:48) Exception was thrown during callback: out of frame resource!
ERROR(synthetic-stream.cpp:48) Exception was thrown during callback: Error occured during execution of processing block! See the log for more info

INFO

```

MartyG-RealSense, 2023-03-09:
Hi @Karin-Sugi  The problem is not related to the installation method.  It is a known issue that enabling three stream types in the RealSense Viewer including the Motion Module (such as depth, RGB and Motion Module) can cause one of the streams to become blank and display the No Frames Received message for some RealSense users.  

Changing the order that the streams are enabled in makes no difference, because when enabling three streams it is the enabling of the Motion Module that is the trigger for the problem.  There is not a fix for this in the RealSense Viewer unfortunately except to only enable two streams (such as Depth and Motion Module, or RGB and Motion Module).

There are workarounds for the issue though when creating your own script.  In C++ you can use **callbacks**, and in Python you can create two separate pipelines and place IMU on its own on one pipeline and depth + RGB on the other pipeline.  The best example of this Python method is at https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238

Karin-Sugi, 2023-03-10:
Thanks @MartyG-RealSense ! Now I understood it is unable to view three stream types at once. However, even when I stream only imu data, sometimes says "no data recieved!". How can I keep streaming? or is it because of ability of RaspberryPi?

There is error messages below. Thanks in advance!

```
ubuntu@ubuntu:~$ realsense-viewer
 10/03 12:06:39,897 INFO [281473789898768] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=0
 10/03 12:06:39,897 INFO [281473789898768] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 10/03 12:06:40,117 INFO [281473789898768] (context.cpp:382) Found 1 RealSense devices (mask 0xff)
 10/03 12:06:40,163 INFO [281473789898768] (rs.cpp:2701) Framebuffer size changed to 1344 x 896
 10/03 12:06:40,166 INFO [281473789898768] (rs.cpp:2701) Scale Factor is now 1
 10/03 12:06:41,311 INFO [281473789898768] (context.cpp:382) Found 1 RealSense devices (mask 0xfe)
 10/03 12:06:48,175 WARNING [281473789898768] (ds5-motion.cpp:573) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 10/03 12:06:48,175 INFO [281473789898768] (ds5-motion.h:188) IMU extrinsic table not found; using CAD values
 10/03 12:06:48,175 INFO [281473789898768] (ds5-motion.h:219) Depth Module V2 IMU Accelno valid intrinsic available, use default values.
 10/03 12:06:48,175 INFO [281473789898768] (ds5-motion.h:231) Depth Module V2 IMU Gyrointrinsic not valid, use default values.
 10/03 12:06:48,175 INFO [281473789898768] (sensor.cpp:1586) Request: MOTION_XYZ32F Accel, 
Resolved to: MOTION_XYZ32F Accel, 
 10/03 12:06:48,177 INFO [281473789898768] (ds5-motion.h:219) Depth Module V2 IMU Accelno valid intrinsic available, use default values.
 10/03 12:06:48,177 INFO [281473789898768] (ds5-motion.h:231) Depth Module V2 IMU Gyrointrinsic not valid, use default values.
 10/03 12:06:48,177 INFO [281473789898768] (sensor.cpp:1586) Request: MOTION_XYZ32F Gyro, 
Resolved to: MOTION_XYZ32F Gyro, 
 10/03 12:06:54,803 INFO [281472669827520] (sensor.cpp:40) Frame Callback Gyro #840 overdue. (FPS: 200, max duration: 4.975124 ms)
 10/03 12:07:02,608 INFO [281472669827520] (sensor.cpp:40) Frame Callback Accel #1184 overdue. (FPS: 100, max duration: 9.900990 ms)
 10/03 12:07:05,166 INFO [281472669827520] (sensor.cpp:40) Frame Callback Gyro #2862 overdue. (FPS: 200, max duration: 4.975124 ms)
 10/03 12:07:08,470 ERROR [281472669827520] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/03 12:07:10,674 ERROR [281472669827520] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/03 12:07:10,675 ERROR [281472669827520] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/03 12:07:10,691 ERROR [281472669827520] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/03 12:07:10,693 ERROR [281472669827520] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/03 12:07:10,694 ERROR [281472669827520] (synthetic-stream.cpp:48) Exception was thrown during callback: Out of frame resources!
 10/03 12:07:31,438 INFO [281472669827520] (sensor.cpp:40) Frame Callback Gyro #7983 overdue. (FPS: 200, max duration: 4.975124 ms)
 10/03 12:07:33,364 INFO [281472669827520] (sensor.cpp:40) Frame Callback Accel #4196 overdue. (FPS: 100, max duration: 9.900990 ms)
 10/03 12:07:43,245 INFO [281472669827520] (sensor.cpp:40) Frame Callback Gyro #10317 overdue. (FPS: 200, max duration: 4.975124 ms)
 10/03 12:07:43,406 INFO [281472669827520] (sensor.cpp:40) Frame Callback Accel #5187 overdue. (FPS: 100, max duration: 9.900990 ms)

```

MartyG-RealSense, 2023-03-10:
RealSense users with Raspberry Pi do not often report using the IMU with it as depth + color are more often used, but on the rare occasions when they have, there have been reports of problems that do not occur when the IMU is used with other computers / computing devices.

It has previously been theorized at https://github.com/IntelRealSense/librealsense/issues/4979#issuecomment-540456439 that it is a kernel related problem, in which case building the SDK on Pi from source code in RSUSB backend mode with the **-DFORCE_RSUSB_BACKEND=true** build flag may help if you have not tried this already, as an RSUSB build of the SDK bypasses the kernel.

Karin-Sugi, 2023-03-11:
I see. I've already built my SDK from source code in RSUSB backend mode. It looks difficult to get imu data by RaspberryPi...
OK. I'll use another imu sensor on my robot instead. Thanks @MartyG-RealSense

closed 2023-03-11 by Karin-Sugi

MartyG-RealSense, 2023-03-11:
You are very welcome!