realsenseai/librealsense #11549: I'm trying to install pyrealsense2 on macbook m1 2020 and I get an error

Micbut, 2023-03-09:
When I do "pip3 install pyrealsense2" or "pip install pyrealsense2" I get the following error

ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2

O also tried to install using the wheel, but also doesn't work
pip install pyrealsense2-2.53.1.4623-cp36-cp36m-manylinux1_x86_64.whl
ERROR: pyrealsense2-2.53.1.4623-cp36-cp36m-manylinux1_x86_64.whl is not a supported wheel on this platform.

I tried with many combinations of python version (3.5,3.6,3.7,3.8,3.9)

Is that true that this has to do with the fact that I am using and M1 processor? Is there a solution or I have to give up?

MartyG-RealSense, 2023-03-09:
Hi @Micbut  The **pip install pyrealsense2** method is designed for use with PC computers with an x32 / x64 architecture, so will not be compatible with the architecture of Apple's M1 processor.

Using an M1 Mac with the RealSense SDK can involve specialized installation procedures.  The company **Lightbuzz** created an excellent installation guide at the link below.

https://lightbuzz.com/realsense-macos/

Within that guide, they suggest using a **shell script** at the link below to install pyrealsense2 by automating the build process.

https://github.com/cansik/pyrealsense2-macosx/blob/apple-silicon/librealsense-python-mac.ps1

[image]

Alternatively, a RealSense user at https://github.com/IntelRealSense/librealsense/issues/9687 created a pre-built Mac version of pyrealsense2 with an automated installer, though it was only tested with the Catalina and Big Sur MacOS versions.

Micbut, 2023-03-16:
Hi, no I managed, thank you a lot!

MartyG-RealSense, 2023-03-16:
You are very welcome.  Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-03-16 by MartyG-RealSense