realsenseai/librealsense #11553: Partial and inconsistent device paths listed under "Physical Port" on Raspberry Pi

zackwoodruff, 2023-03-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435/D455| 
| Firmware Version                   | 05.14.00.00/05.14.00.00 | 
| Operating System & Version |   Linux, RPI OS (Bullseye and Buster)  | 
| Kernel Version (Linux Only)    |  (5.15.84 and 5.10.103)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |     v2.53.1                | 
| Language                            |  C++                        | 
| Segment			|  Robot                        | 

### Issue Description
- I have multiple cameras connected to a raspberry pi and I would like to differentiate between them based on the physical USB port they are plugged into. I do not want to differentiate by serial number because cameras may be swapped between ports. 
- When I run `rs-enumerate-devices -S` on both Bullseye and Buster I get partial paths under "Physical Port" and these paths change when I unplug and replug a camera into the same port:
````
Device Name                   Serial Number       Firmware Version    
Intel RealSense D435          831612072717        05.14.00.00         
Intel RealSense D455          151422250645        05.14.00.00         
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	831612072717
    Firmware Version              : 	05.14.00.00
    Recommended Firmware Version  : 	05.14.00.00
    Physical Port                 : 	2-1-10
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	830513022942
    Firmware Update Id            : 	830513022942

Device info: 
    Name                          : 	Intel RealSense D455
    Serial Number                 : 	151422250645
    Firmware Version              : 	05.14.00.00
    Recommended Firmware Version  : 	05.14.00.00
    Physical Port                 : 	2-2-9
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B5C
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	151223063319
    Firmware Update Id            : 	151223063319

````

- On a Linux computer running Ubuntu 20.04 I get the the same output except the Physical Port includes the full path and is consistent when the device is unplugged and plugged back in. 
```
Device Name                   Serial Number       Firmware Version    
Intel RealSense D435          831612072717        05.14.00.00         
Intel RealSense D455          151422250645        05.14.00.00         
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	831612072717
    Firmware Version              : 	05.14.00.00
    Recommended Firmware Version  : 	05.14.00.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	830513022942
    Firmware Update Id            : 	830513022942

Device info: 
    Name                          : 	Intel RealSense D455
    Serial Number                 : 	151422250645
    Firmware Version              : 	05.14.00.00
    Recommended Firmware Version  : 	05.14.00.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video8
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B5C
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	151223063319
    Firmware Update Id            : 	151223063319
```

Do you know what could be causing the partial paths instead of the full paths under the "Physical Port"? Thanks.

MartyG-RealSense, 2023-03-10:
You may be able to set fixed ports for cameras by defining **symbolic links** via udev rules.

https://unix.stackexchange.com/questions/479302/udev-rules-for-seemingly-indistinguishable-devices

MartyG-RealSense, 2023-03-11:
As you said, you can enable a device by its serial number so that it does not matter which port the camera is plugged into so long as the camera is detectable.  But as you also stated, that approach is not suitable for your particular project because the cameras may get swapped between ports and you want to address a specific port.

Other than your own udevadm method, the best approach may be to create a new udev rule to enforce port naming, like in the guide at the link below.

https://inegm.medium.com/persistent-names-for-usb-serial-devices-in-linux-dev-ttyusbx-dev-custom-name-fd49b5db9af1

zackwoodruff, 2023-03-13:
@MartyG-RealSense
Is there a way to enable a camera by port in the realsense SDK? I have looked through the multicam example and multiple related issues (e.g., https://github.com/IntelRealSense/librealsense/issues/2219) but I have yet to see an example of enabling specific streams from specific cameras without using the serial number.

MartyG-RealSense, 2023-03-13:
The RealSense ROS wrapper enables cameras to be accessed by a specific USB port ID instead of camera serial number with the **USB_PORT_ID** instruction.  The question is whether the port address would still be changing though.

If you need to use the SDK instead of ROS, the SDK has a set of **Multi-Streaming Model** rules that govern access to multiple cameras and multiple streams.  The available code documentation for these rules is limited though.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

zackwoodruff, 2023-03-13:
Thanks @MartyG-RealSense. I am not using ROS so I will look into the Multi-Streaming Model. Worst case I should be able to parse the serial number from the part of the port that is consistent on plug/unplug and device swap.

MartyG-RealSense, 2023-03-23:
Hi @zackwoodruff  Do you require further assistance with this case, please?  Thanks!

zackwoodruff, 2023-03-27:
@MartyG-RealSense I am good for now thanks.

closed 2023-03-27 by zackwoodruff