realsenseai/librealsense #11562: Exception caught trying to detect presets: could not open directory: /home/franka/Documents/librealsense2/presets/

Shillin123, 2023-03-14:
I use ubuntu 20.04, and I have installed librealsense, but when running the command realsense-viewer, I got this error:
Exception caught trying to detect presets: could not open directory: /home/franka/Documents/librealsense2/presets/

[image]

And I can't get the RGB scenario under the cam.
[image]

Can someone please help to solve it? It's very urgent for me.

MartyG-RealSense, 2023-03-14:
Hi @Shillin123  As the 'e' series of industrial RealSense-compatible cameras are not an Intel product but are instead manufactured and sold by **FRAMOS**, Intel RealSense technical support is unable to assist with this camera range unfortunately.  FRAMOS support should be contacted instead by visiting their website at the link below and clicking on the 'Let's Talk' button on the page.

https://www.framos.com/en/resources/support

However, I take into account the urgency of your need for a solution and can at least report that recently a D435e user experienced the same error on this support forum at https://github.com/IntelRealSense/librealsense/issues/10361 and found in their particular case that the error disappeared after opening and closing realsense-viewer a few times.

For further assistance if your problem continues, please contact FRAMOS support.

Shillin123, 2023-03-14:
@MartyG-RealSense  Thanks for your quick reply. I will try it and also ask from FRAMOS.

MartyG-RealSense, 2023-03-22:
Case closed due to FRAMOS being the support channel for 'e' cameras.

closed 2023-03-22 by MartyG-RealSense