realsenseai/librealsense #11564: What is the fundamental logic for D455's driver to do with differnet configuration of resolutions?

Hypothesis-Z, 2023-03-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu20  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

For example, my supposal is, if 640x480 is set, the RealSense driver will first crop the image from 1280x720 to 960x720, then resize it to 640x480. 

I am unifying images from D455s in different resolutions. After transform, they should be in 640x480 with the same camera info. 

Thanks!

MartyG-RealSense, 2023-03-15:
Hi @Hypothesis-Z  In a discussion from 2028 at the link below, a RealSense team member states "When the camera returns depth, it uses rectified images.  During the rectification process, the image gets downsized.  In order to get the 1280x720 resolution after rectification, the extra pixels are extrapolated.  This is why the whitepapers say that the optimal depth resolution for the D435 is 848x480.  Using a resolution higher than 848x480 should give you similar or slightly worse depth quality".

848x480 is also the optimal depth accuracy resolution for D455.

https://community.intel.com/t5/Items-with-no-label/D435-Resolution-VS-Depth-Image-Quality/m-p/548110#M8401

MartyG-RealSense, 2023-03-15:
Aside from the quote above, there is not information available about the logic of the camera's rectification process and so I am not able to confirm whether your process is right compared to the camera's actual logic, unfortunately.

Would it be possible for you to set the RGB resolution of your snapshot to 640x480 instead of 1280x720 so that the image does not need to be resized after capture, since D455 supports that RGB resolution?

An alternative approach for downsampling the resolution is to use a post-processing **Decimation filter**, which is supported on D455 for the RGB stream as well as depth and has a default setting of '2', meaning that the resolution will be divided by 2 when the filter is enabled (so 1280x720 becomes 640x360).

Hypothesis-Z, 2023-03-16:
@MartyG-RealSense For some reason I have to set RGB resolution to 1280x720. Anyway, thank you! You can close the issue now.

MartyG-RealSense, 2023-03-16:
You are very welcome.  As you suggested, I will close the issue.  Thanks again!

closed 2023-03-16 by MartyG-RealSense