realsenseai/librealsense #11570: Get depth intensity bar in Python

TheNemo05, 2023-03-16:
[image]
Hello as displayed in Realsense SDK this depth intensity bar, is it possible to get in python code.

MartyG-RealSense, 2023-03-16:
Hi @TheNemo05  The RealSense Viewer's scale side-bar is not part of the depth image but is instead drawn over the top of the image as a GUI (graphical user interface) overlay.  Typically, RealSense Python programs do not make use of GUI elements like C++ programs do.  

The Viewer's user interface is a combination of raw OpenGL calls, GLFW for cross-platform window and event management, and IMGUI for the interface elements.

It is possible to create a GUI in Python - as described at https://github.com/IntelRealSense/librealsense/issues/10337#issuecomment-1078750365 - but it would not use the same graphical interface method as the Viewer and so it is unlikely that the Viewer's scale side-bar could be replicated in Python.

TheNemo05, 2023-03-16:
So basically with the use of Tkinter library a new GUI can be developed ? instead of using the original SDK. Is that what you are trying to say.

MartyG-RealSense, 2023-03-16:
Yes, it is possible to use TKinter with a Python librealsense application but it will not be the same as the GUI in the RealSense SDK.

closed 2023-03-23 by TheNemo05