realsenseai/librealsense #11571: Calculate surface area of object using 3D Depth reconstruaction.

TheNemo05, 2023-03-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |  NA  | 
| Platform                                 | PC/Raspberry Pi  |
| SDK Version                            |  2.53.1  | 
| Language                            |  Python  | 
| Segment			|  Seabot    | 

### Issue Description
Hello again, 
I need some help regarding a project i m trying to accomplish. I need to calculate Surface area of a undefined object using its depth parameters & 3D reconstruction. I ve recorded .bag file for the object.
My main objective is to calculate surface area of the object. Previously i ve tried Open3D which has predefined realsense packages but due to unavailability of information was not able to construct any reconstruction model. 
As mentioned earlier I have recorded sequence of bag file. What can be the method I can follow for accomplishment of this project.

MartyG-RealSense, 2023-03-16:
Hi @TheNemo05  Would it be practical for you to instead export a **ply** pointcloud file instead of a bag and then import it into the free open source RealSense-compatible **MeshLab** tool to perform 3D reconstruction and surface area estimation on it, as described at https://github.com/IntelRealSense/librealsense/issues/9823#issuecomment-1000413348

TheNemo05, 2023-03-17:
Also if I use meshlab i ll have to join multiple ply files together. in that case inaccuracy will be more in surface area calculation. surely i ll be using meshlab if i was able to somehow capture the 3D view of object.

I need a solution that has depth 3D reconstruction. with all the depth parameters & hopefully that can convert .bag file into 3D reconstruction.

TheNemo05, 2023-03-17:
I am not good at C++ but can you provide a algorithm using kinfu.

TheNemo05, 2023-03-17:
Also i ve finally decided to give mesh lab a try, but to build a accurate 3D model i ll have to eliminate the same co-ordinates or single (x,y,z) should be repeated only once else there might be overlapping of pointclouds. Do you have any method to resolve it.

TheNemo05, 2023-03-17:
https://www.youtube.com/watch?v=pLCVCH7ypI4 
This video from Open3D uses some webview with realsense. does D455 have access to this webview.

MartyG-RealSense, 2023-03-17:
You can manually clean up overlapping points in MeshLab.

https://en.wikiversity.org/wiki/MakerBot/Cleaning_up_cloud_point_mesh_in_mesh_lab_for_3D_printing

*********

In regard to the Open3D video, most RealSense 400 Series cameras with an RGB sensor (except D405) can also be used as an ordinary video webcam and so a D455 should work with programs / applications that use RGB video for their analysis.

*********

The requirement to use a single bag file instead of multiple ply files is likely to rule out most possible methods for achieving your reconstruction, unfortunately.

However, if you are able to use C++ instead of Python, you could try using the RealSense SDK's **rs-kinfu** example program, which can use a single camera that is moved around the object to create a 360 degree pointcloud by "fusing" frames together.  The resulting large pointcloud can then be exported as a .ply file.

Whilst rs-kinfu uses a live camera by default, you could conceivably edit the script to use a recorded bag file as the camera data-source instead.  So you could record a bag where you move the camera all around the object, and then load that file into rs-kinfu and have it turn the bag and its 360 degree recording of the object into a .ply file.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

TheNemo05, 2023-03-20:
Alright, considering the fact i ve recorded .bag file, is it possible to convert every frame in a bag file frame into .ply file using some python code or sdk method.

If yes, I ll have 1000's of pointcloud files how do I frame it in a single file using the co-ordinate information.

TheNemo05, 2023-03-20:
can i get any tutorial for RS-kinfu

TheNemo05, 2023-03-20:
Not good at understanding C++, but seen some tutorials with regards to realsense. I hope to get help with some predefined C++ codes if possible.

MartyG-RealSense, 2023-03-20:
You could convert every frame to a ply file using the RealSense SDK's pre-made **rs-convert** C++ tool.  If the SDK tools and examples were installed when the SDK was installed then an executable of rs-convert should be in the **usr/local/bin** folder of your computer.

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

You could merge all of the ply files together using the command-line interface of the free open source tool **CloudCompare**, as described at https://github.com/IntelRealSense/librealsense/issues/10640#issuecomment-1172460776 and the link below.

http://www.danielgm.net/cc/forum/viewtopic.php?t=2543

TheNemo05, 2023-03-20:
[image]
getting this fault while converting to .ply file

TheNemo05, 2023-03-20:
Hello there, i converted the .bag into ply file using the rs-convert tool. But i did not received any .ply file. I either received .raw or .txt files; is there an issue in conversion.

MartyG-RealSense, 2023-03-20:
.ply file conversion with rs-convert unfortunately does not work for some RealSense users whilst the other modes do work.  An example of this problem is discussed at https://github.com/IntelRealSense/librealsense/issues/3214

Can you confirm that you used the command **-l** please (it is a small-case L and not an i).

TheNemo05, 2023-03-21:
[image]
 for your reference i am using this command.

TheNemo05, 2023-03-21:
Also can you please share c++ code for rs-kinfu

MartyG-RealSense, 2023-03-21:
The rs-kinfu C++ code is here:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp

It's excellent to hear that you were successful with the .ply conversion in rs-convert.

A person at the link below installed **Open3D** to convert .ply to a .pcd that was readable in CloudCompare.

https://stackoverflow.com/questions/51350493/convertion-of-ply-format-to-pcd-format/62488893#62488893

There is also an online converter.

https://imagetostl.com/convert/file/ply/to/pcd

TheNemo05, 2023-03-22:
[image]
 i ve been getting this error even though i ve checked imgproc.hpp is present in opencv2 folder. Can you please help.

MartyG-RealSense, 2023-03-22:
Are you using OpenCV 4, as the rs-kinfu documentation states that this OpenCV version is required.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu#installation

TheNemo05, 2023-03-23:
surely i am using the version 4 of opencv
[image]

MartyG-RealSense, 2023-03-23:
What method did you use to compile rs-kinfu please?  There are not instructions for installing it but I assume that it would involve building it with CMake Lists.txt in order to provide the linkages to librealsense.

[image]

TheNemo05, 2023-03-27:
I used cmake CMakeList.txt from terminal.

TheNemo05, 2023-03-28:
I ve build using the following CMake, but the program still has huge number of errors.

TheNemo05, 2023-03-28:
[image]
Also while i created make for the above build option u mentioned in the build folder of librealsense i got this error regarding rs-kinfu.

MartyG-RealSense, 2023-03-28:
I do not have anything else I can suggest for correcting your rs-kinfu errors, unfortunately.  I do apologize.

MartyG-RealSense, 2023-04-02:
Hi @TheNemo05  Bearing in mind the comment above, do you require further assistance with this case please?  Thanks!

closed 2023-04-07 by MartyG-RealSense