realsenseai/librealsense #11575: Help: D435i only recognized as USB 2, including realsense-viewer, failed to update FW

iDonghq, 2023-03-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.00.50| 
| Operating System & Version | Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
 D435i only recognized as USB 2, including realsense-viewer, failed to update FW

iDonghq, 2023-03-17:
[image]

iDonghq, 2023-03-17:
[image]

MartyG-RealSense, 2023-03-17:
Hi @iDonghq  The rs-fw-update tool uses **bin** files downloaded from the RealSense firmware releases page at the link below.  The use of a zip file is likely the cause of the unsupported binary error.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

Also, can you confirm that the camera is attached to a USB 3 port on the computer, please?

You should be able to update the firmware using the firmware bin file on a USB 2 connection in realsense-viewer by going to the **More** option near the top of the options side-panel and selecting **Update Firmware** from the drop-down menu, then navigating to the location of the bin file on your computer..

iDonghq, 2023-03-17:
Thanks for replying, Marty. Yes, I have finished the update using realsense-viewer with different version, but it doesn't work, I do further confirm that D435i is attched to the USB 3 port and cable on PC, but the usb type is still 2.1.

MartyG-RealSense, 2023-03-17:
Thanks very much for the confirmation.

Are you using the official 1 meter USB cable supplied with the camera or a longer USB cable of your own choice?  The best chance of firmware update success is to use the official cable if possible, as longer cables may increase the chance of a firmware update failing if the cable is not a high-quaity one.

iDonghq, 2023-03-17:
> 

ok, I will confirm later the cable  comes from the official whether or not.

iDonghq, 2023-03-17:
[image]
The cable is here, I am not sure it belongs to the official, please help me confirm it? Thank you very much

MartyG-RealSense, 2023-03-17:
It is difficult to tell as USB cables look very similar.  It appears to be a 1 meter cable.  RealSense cables usually have the SuperSpeed (SS) logo printed on the side of the full-size connector that has the grey squares (not the blue squares on the other side of the connector) whilst yours does not seem to have the SuperSpeed logo printed on that side.

[image]

[image]

iDonghq, 2023-03-20:
> RealSense cables usu

OK，I will confirm that later

iDonghq, 2023-03-20:
> It is difficult to tell as USB cables look very similar. It appears to be a 1 meter cable. RealSense cables usually have the SuperSpeed (SS) logo printed on the side of the full-size connector that [...]

Thank you very very very much Marty, Issued from usb cable type is fixed~ U can closed it.

iDonghq, 2023-03-20:
> It is difficult to tell as USB cables look very similar. It appears to be a 1 meter cable. RealSense cables usually have the SuperSpeed (SS) logo printed on the side of the full-size connector that [...]

Thank you very very very much again, Marty~ Issue from usb cable typed is fixed, u can closed it.

MartyG-RealSense, 2023-03-20:
That's excellent news, @iDonghq - thanks very much for the update.  As you suggested, I will close the case.  Thanks again!

closed 2023-03-20 by MartyG-RealSense