realsenseai/librealsense #11576: terminate called after throwing an instance of 'rs2::error' with what(): out of range value for argument "def"

emailweixu, 2023-03-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |  Ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  5.19.0-35-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.35.1                          | 
| Language                            |  C++                          | 
| Segment			|  Robot                          | 

### Issue Description

I am trying to use librealsense2-net following the instruction at https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras.

I installed librealsense2-net using:
```bash
sudo apt-get install librealsense2-net-dev
sudo apt-get install librealsense2-net-dbg
```
After starting `rs-server`, I ran the binary compiled from the attached C++ code. I got the following error message:
```text
terminate called after throwing an instance of 'rs2::error'
  what():  out of range value for argument "def"
Aborted (core dumped)
```
I also tried compiling the SDK and ran the newly compiled `rs-server` but still got the same error. Debugging with the compiled SDK showed that the error is caused by the following line:

https://github.com/IntelRealSense/librealsense/blob/b874e42685aed1269bc57a2fe5bf14946deb6ede/src/ethernet/ip_device.cpp#L146

`control.range` at this time is  `{min=0, max=2, def=3}`, which VALIDATE_RANGE fails at the following line:

https://github.com/IntelRealSense/librealsense/blob/b874e42685aed1269bc57a2fe5bf14946deb6ede/src/rs.cpp#L2672

At this point, I had no idea what I can do.

```C++
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <librealsense2-net/rs_net.hpp>
#include <iostream>             // for cout

int main()
{
    rs2::net_device dev("127.0.0.1");
    rs2::context ctx;
    dev.add_to(ctx);
    rs2::pipeline p(ctx);

    std::cout << "hello from librealsense - " << RS2_API_VERSION_STR << std::endl;
    std::cout << "You have " << ctx.query_devices().size() << " RealSense devices connected" << std::endl;

    return 0;
}

```

emailweixu, 2023-03-17:
I was able to make it work by changing the following lines to ignoring the out of range def : https://github.com/IntelRealSense/librealsense/blob/b874e42685aed1269bc57a2fe5bf14946deb6ede/src/ethernet/ip_device.cpp#L144-L147

```C++
    else if(control.range.min <= control.range.def && control.range.def <= control.range.max)
    {
        remote_sensors[control.sensorId]->sw_sensor->add_option(control.option, {control.range.min, control.range.max, control.range.def, control.range.step});
    } else {
        continue;
    }
 ```

But I think it shouldn't get the out of range def in the first place. It smees there's something wrong with rs-server.

MartyG-RealSense, 2023-03-17:
Hi @emailweixu  There are previous cases of this ethernet error at https://github.com/IntelRealSense/librealsense/issues/7459 and https://github.com/IntelRealSense/librealsense/issues/10747

It is good to hear that you achieved a workaround solution.  The rs-server ethernet networking system is removed in the next librealsense SDK release. This is because it was an experimental proof of concept rather than an official, validated Intel RealSense feature. This networking tool will therefore not receive further fixes / updates in subsequent SDK releases.

MartyG-RealSense, 2023-03-21:
Hi @breakds  There will continue to be support for ethernet for RealSense cameras in the future after the removal of the current networking system, though information about what is coming next is not available at the time of writing this.

The **EtherSense** networking tool for Python is a currently available alternative networking solution and should remain available after the rs-server tool is withdrawn.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

breakds, 2023-03-21:
> Hi @breakds There will continue to be support for ethernet for RealSense cameras in the future after the removal of the current networking system, though information about what is coming next is not [...]

Thanks for the reply @MartyG-RealSense ! Really helpful!

closed 2023-04-02 by MartyG-RealSense