realsenseai/librealsense #11581: Whether high-speed motion of the camera during sampling will affect the accuracy of 3D data ？

FeiSong123, 2023-03-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |  Ubuntu 18  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC|
| SDK Version                            |  2.53.1                        | 
| Language                            |  C++                         | 
| Segment			|  Robot                      | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I used the L515 to assist in positioning on a mobile robot. At that time, there was a phenomenon that when the robot moved slowly, the positioning effect was significantly better than when the robot moved faster (of course, it could also be that the recognition effect was poor when the robot moved faster). Later, when I communicated with our colleagues working on two-dimensional radar (who also perform auxiliary positioning on high-speed mobile robots with a data update period of 25ms), I discovered that they have implemented a dynamic compensation for the data, which can play a significant role in their auxiliary positioning.

Then I thought that high-speed mobile robots have such a significant impact on the radar for 25ms of data updates, and that would also have an impact on the camera. So I wanted to follow their approach to dynamic compensation, but due to different data update principles, I haven't figured out how to specifically implement it.

I just want to ask if the camera is moving at high speed during camera sampling, which may have some adverse effects on the depth or 3D data of the camera? If there is an impact, can we use dynamic compensation methods to offset this impact, as well as reference dynamic compensation methods?

MartyG-RealSense, 2023-03-18:
The L515 is designed to be able to scan fast moving objects with no motion blur, as described in its official YouTube launch video at the link below.

https://www.youtube.com/watch?v=Rx-75ZEAMeg

If you are using the RGB color stream then it may be worth having depth set at 30 FPS (the only depth speed available on L515) and RGB at **60** FPS.  The RGB sensor on the L515 has a **rolling** type shutter, which can be susceptible to lag / blurring when capturing fast motion.  Increasing RGB FPS to 60 can help to negate this.

FeiSong123, 2023-03-19:
OK, I will have a try.

FeiSong123, 2023-03-29:
No, thanks. You can turn it off at will.

closed 2023-03-29 by MartyG-RealSense