realsenseai/librealsense #11590: The accuracy of depth information with L515

HappyOnion6, 2023-03-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.53.1                          | 
| Segment			|  VR/AR                      | 

<html>
<body>

Required Info |  
-- | --
Camera Model | L515
Firmware Version | 01.05.08.01
Operating System & Version | Win 10
Kernel Version (Linux Only) | NA
Platform | PC
SDK Version | 2.53.1
Segment | VR\AR

</body>
</html>

### Issue Description
<Describe your issue / question / feature request / etc..>
I used Intel.RealSense.Viewer of L515 to dump the depth information of different scenes. There was a question in .raw format file of depth which  was saved from Intel.RealSense.Viewer. The depth value of a pixel in preview mode of Intel.RealSense.Viewer was correct，but after we saved snapshot of depth and opend the .raw fomat file of depth by the image tool, we found that the depth value of a pixel was different from the value of preview mode. 
The example of preview mode is shown in Figure 1 , press the button of save snapshot to save the depth file as a .raw format file. The example of saving file is shown in Figure 2.
How can i get the correct value of depth in saved file, and Whether there is a conversion relationship between the correct depth value in preview mode and the depth value in saved file?

[image]
Figure1

[image]
Figure2

MartyG-RealSense, 2023-03-21:
Hi @HappyOnion6  When data is exported from the RealSense Viewer and then imported into another tool (such as 3D modelling software), there may be a difference between the original scale and the scale that is set in the tool that the data was imported into.  If your software allows scale to be changed then it should be set to the same scale that the Viewer uses.  On the L515 camera model the default scale is **0.000250**

HappyOnion6, 2023-03-24:
@MartyG-RealSense Thanks for your reply and it is really helpful to solve our problem

closed 2023-04-04 by MartyG-RealSense