realsenseai/librealsense #11593: Understanding OEM Calibration

kumar-akshay324, 2023-03-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.14 | 
| Operating System & Version |   Linux (Ubuntu 20)| 
| Kernel Version (Linux Only)    |  5.10.104-tegra                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX |
| SDK Version                            | 2.53.1| 
| Language                            |  C++                          | 

### Issue Description

To confirm, when I run the `rs-oem-server` for calibration and the client gives a red thumbs UP or a green thumbs DOWN; it does also flash the new calibration data onto the device and that is used thereafter, right?

Is there a way to derive when the last calibration was done on the camera? (Can't see anything on the `rs-calibration-writer` or `rs-calibration-converter` binaries for it). We are trying to calibrate the camera would want to know when it is done such that I can kill the server and shut down the device from a remote machine. Also, knowing the calibration history would be useful for posterity.

MartyG-RealSense, 2023-03-21:
Hi @kumar-akshay324  Thanks very much for your question about OEM calibration.  I will discuss it with my Intel RealSense colleagues.

kumar-akshay324, 2023-03-21:
thanks @MartyG-RealSense, looking forward to the answers. Appreciate the prompt response

kumar-akshay324, 2023-03-22:
@MartyG-RealSense bumping this one up. Any leads would be helpful

MartyG-RealSense, 2023-03-22:
I will update as soon as I have information to feed back to you.  Thanks very much for your patience!

kumar-akshay324, 2023-03-24:
Hey @MartyG-RealSense  - any updates on this?

MartyG-RealSense, 2023-03-24:
I am continuing to communicate with my RealSense colleagues about your question.  Thanks again.

MartyG-RealSense, 2023-03-24:
My colleagues have provided advice about your issue.

The calibration pass / failed result will be shown in the client window of the OEM calibration tool and the result will be saved for each calibration session (chapter 5.3 of the tool's user guide – Result folder and debugging information).  This information is based on OEM calibration tool release 1.3.0.0.

MartyG-RealSense, 2023-03-31:
Hi @kumar-akshay324  Was the advice in the comment above helpful, please?  Thanks!

closed 2023-04-13 by MartyG-RealSense