realsenseai/librealsense #11595: How to calculate the coordinates in the depth image from the coordinates in the color image, not to get the depth value in the color image directly, but to get the corresponding coordinate value in the depth image

jokeryun, 2023-03-22:
How to calculate the coordinates in the depth image from the coordinates in the color image, not to get the depth value in the color image directly, but to get the corresponding coordinate value in the depth image

MartyG-RealSense, 2023-03-22:
Hi @jokeryun  The RealSense SDK instruction **rs2_project_color_pixel_to_depth_pixel** can be used to convert a single specific color pixel to a depth pixel in order to obtain a 3D XYZ coordinate.  Examples of this instruction's use in C++ and Python can be found at the links below.

**C++**
https://github.com/IntelRealSense/librealsense/issues/2982

**Python**
https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

MartyG-RealSense, 2023-03-27:
Hi @jokeryun  Do you require further assistance with this case, please? Thanks!

jokeryun, 2023-03-27:
Thank you very much，I have solved this issue with your help。

closed 2023-03-27 by jokeryun

MartyG-RealSense, 2023-03-27:
You are very welcome.  It is excellent news that you have achieved a solution.  Thanks very much for the update!