realsenseai/librealsense #11597: unity开发过程中无法调用sdk中的工具类。

mxuGitHub, 2023-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):
在调用命名空间的时候，无法调用using Intel.RealSense.Hand;
[image]

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-03-22:
Hi @mxuGitHub  It appears that you are using the RealSense Unity compatibility wrapper from the old 2016 RealSense SDK for Windows.  This SDK was compatible with the original generation F200, SR300 and R200 models of RealSense camera but not modern models such as the 400 Series and L515.

If you are using the 2016 SDK, the error may mean that you are using the **R3** version of it.  Whilst R2 installed everything in a single 1.8 gigabyte installation, R3 divided the SDK into components, a core **Essentials** mandatory module and a series of optional modules.  One of these optional modules was **Hand**, so to have hand tracking with the R3 SDK you would first need to install the Essentials module first and then the Hand module secondly.

If your camera model is an SR300 then it is also compatible with the modern RealSense SDK 2.0 which supports the 400 Series and L515 cameras.

https://github.com/IntelRealSense/librealsense/releases

*********

您好 @mxuGitHub 看来您正在使用适用于 Windows 的旧 2016 RealSense SDK 中的 RealSense Unity 兼容性包装器。 此 SDK 与第一代 RealSense 相机的 F200、SR300 和 R200 型号兼容，但不兼容 400 系列和 L515 等现代型号。

如果您使用的是 2016 SDK，则该错误可能意味着您使用的是 **R3** 版本。 虽然 R2 在一个 1.8 GB 的安装中安装了所有东西，但 R3 将 SDK 分成组件、核心 **Essentials** 强制模块和一系列可选模块。 这些可选模块之一是 **Hand**，因此要使用 R3 SDK 进行手部跟踪，您首先需要先安装 Essentials 模块，然后再安装 Hand 模块。

如果您的相机型号是 SR300，那么它也与支持 400 系列和 L515 相机的现代 RealSense SDK 2.0 兼容。

https://github.com/IntelRealSense/librealsense/releases

MartyG-RealSense, 2023-04-04:
Case closed due to no further comments received.

由于未收到进一步的评论，案例已结案。

closed 2023-04-04 by MartyG-RealSense