realsenseai/librealsense #11599: 如何调用Intel.RealSense下的所有命名空间？

mxuGitHub, 2023-03-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
你好，我的电脑使是Windows10 版本号为21H2的操作系统。使用的是Intel realsense D415设备，安装的sdk是Intel.RealSense.SDK-WIN10-2.53.1.4623。我的开发软件是unity2018，代码编辑软件是vs code。unity插件是官网最新的Intel.RealSense.unitypackage。安装好sdk和unity插件，新建脚本，在脚本中调用命名空间时发现在Intel.RealSense下，只有math和base这两个命名空间。请问，我该如何配置，才能正常使用sdk中的接口？
[image]

MartyG-RealSense, 2023-03-22:
Hi @mxuGitHub  Intel.RealSense.Face and Intel.RealSense.Hand are from the old RealSense SDK for Windows from the year 2016.  The hand and face modules from that SDK are not compatible with the modern RealSense SDK 2.0 as the architecture and programming instructions of SDK 2.0 are completely different, unfortunately. 

*********

您好 @mxuGitHub Intel.RealSense.Face 和 Intel.RealSense.Hand 来自 2016 年的旧 RealSense SDK for Windows。该 SDK 中的手部和面部模块与现代 RealSense SDK 2.0 不兼容，因为体系结构和编程说明 不幸的是，SDK 2.0 完全不同。

mxuGitHub, 2023-03-22:
请问，RealSense SDK 2.0 有实现手部和面部识别的接口吗？如何实现呢？如果实现不了，旧版本的sdk去什么地方下载呢？

MartyG-RealSense, 2023-03-22:
The old 2016 SDK is not compatible with D415.  If you did want to download it though then the 1.8 gb installer program for the complete SDK with all features installed including Face and Hand modules can be obtained from the link below.

http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe

If you require face and hand tracking in Unity for the modern RealSense SDK 2.0 then the best option is likely to be to use the commercial software Nuitrack SDK.  It has a free trial version.

https://nuitrack.com/

https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Basic.md

https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Face_Tracking.md

**************

旧的 2016 SDK 与 D415 不兼容。 如果您确实想下载它，那么可以从下面的链接获得完整 SDK 的 1.8 gb 安装程序，其中安装了包括面部和手部模块在内的所有功能。

http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe

如果您需要在 Unity 中为现代 RealSense SDK 2.0 进行面部和手部跟踪，那么最好的选择可能是使用商业软件 Nuitrack SDK。 它有一个免费试用版。

https://nuitrack.com/

https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Basic.md

https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Face_Tracking.md

mxuGitHub, 2023-03-22:
请问：哪些设备支持旧的 2016 SDK，做面部追踪和手部动作识别？

 

 
http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe
 
 
https://nuitrack.com/
 
https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Basic.md
 
https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Face_Tracking.md
 
 
 
http://registrationcenter-download.intel.com/akdlm/irc_nas/vcp/9078/intel_rs_sdk_offline_package_10.0.26.0396.exe
 
 
https://nuitrack.com/
 
https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Basic.md
 
https://github.com/3DiVi/nuitrack-sdk/blob/master/doc/Unity_Face_Tracking.md

MartyG-RealSense, 2023-03-22:
R2: F200, R200, SR300

R3: F200, SR300

MartyG-RealSense, 2023-03-22:
RealSense SDK 2.0 不内置面部和手部追踪支持，因此这些功能通常需要第三方软件。

英特尔在 YouTube 上发布了使用现代实感 400 系列相机进行身体关节跟踪并在 Unity 中使用它的教程，尽管它很复杂，因为它需要训练您自己的数据集。

https://www.youtube.com/watch?v=VSHDyUXSNqY

mxuGitHub, 2023-03-22:
有没有免费的第三方软件支持

 

 
 
https://www.youtube.com/watch?v=VSHDyUXSNqY

MartyG-RealSense, 2023-03-22:
Unity has a tool called MARS that can perform face tracking.

https://docs.unity3d.com/Packages/com.unity.mars@1.0/manual/FaceTracking.html

Other than that, it is unlikely that a free solution will be able to be found, unfortunately, though Nuitrack has a free trial version.

There is paid face detection software in Unity called the dlib FaceLandmark Detector that is based on the OpenCV vision software.

https://assetstore.unity.com/packages/tools/integration/dlib-facelandmark-detector-64314

https://github.com/EnoxSoftware/DlibFaceLandmarkDetector

https://assetstore.unity.com/packages/templates/tutorials/realtime-facerecognition-example-73233

Another paid tool on the Asset Store is the Face Recognition System.

https://assetstore.unity.com/packages/add-ons/machinelearning/face-recognition-system-197342

***********

Unity 有一个名为 MARS 的工具，可以执行面部跟踪。

https://docs.unity3d.com/Packages/com.unity.mars@1.0/manual/FaceTracking.html

除此之外，不太可能找到免费解决方案，不幸的是，尽管 Nuitrack 有免费试用版。

Unity中有一款收费的人脸检测软件叫做dlib FaceLandmark Detector，它是基于OpenCV视觉软件的。

https://assetstore.unity.com/packages/tools/integration/dlib-facelandmark-detector-64314

https://github.com/EnoxSoftware/DlibFaceLandmarkDetector

https://assetstore.unity.com/packages/templates/tutorials/realtime-facerecognition-example-73233

Asset Store 上的另一个付费工具是人脸识别系统。

https://assetstore.unity.com/packages/add-ons/machinelearning/face-recognition-system-197342

MartyG-RealSense, 2023-03-27:
Hi @mxuGitHub  Do you require further assistance with this case, please? Thanks!

你好，请问你需要进一步的帮助吗？ 谢谢！

MartyG-RealSense, 2023-04-02:
Case closed due to no further comments received.

由于未收到进一步的评论，案例已结案。

closed 2023-04-02 by MartyG-RealSense