realsenseai/librealsense #11607: How can I get poses(positions and rotations) of the camera?

TwiceMao, 2023-03-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Windows 10 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
How can I get ground truth poses(positions and rotations) of the camera?

MartyG-RealSense, 2023-03-24:
Hi @TwiceMao  If you are seeking Pose information (position and rotation), the RealSense 400 Series camera models that are equipped with an IMU (D435i and D455) do not have in-built support for pose data.

The RealSense T265 Tracking Camera model does have in-built pose stream support, though it is a retired model and no longer receives updates in the RealSense SDK.

https://store.intelrealsense.com/buy-intel-realsense-tracking-camera-t265.html

Whilst it is possible to obtain the pose of observed **objects** with a 400 Series camera, the pose of the camera itself cannot be obtained (again, this is a T265 feature).  Conceivably, you could obtain a device pose for D455 by mounting a D455 and T265 together on the same bracket and point them both in the same direction, as the two cameras will move and rotate together as the bracket moves and rotates.

If you prefer to use only the D455 then you can obtain just the rotational orientation from the IMU (but not the device position).  The RealSense SDK's C++ example program **rs-motion** demonstrates this.

https://github.com/IntelRealSense/librealsense/tree/master/examples/motion

It is possible to plot the **relative position** of a 400 Series camera on a map by using the ROS1 project described at https://github.com/IntelRealSense/librealsense/issues/8094#issuecomment-753979162

TwiceMao, 2023-03-24:
Thanks for your quick and effective response. Actually,  I want to  construct world coordinates on the first frame. So I just need the relative poses (both relative positions and rotations) of the camera. And I'm on a Windows OS (no ros). So is there an easy way for me to get the relative pose of the camera?

MartyG-RealSense, 2023-03-24:
If you know the distance of an observed object from the camera and its height above / below the camera and can calculate its height - as described at https://github.com/IntelRealSense/librealsense/issues/9297 - then you may be able to work out the camera's 3D real-world position relative to the real-world position of that observed object.

Another possibility may be to set the camera's relative position at 0,0,0 and then calculate how far it has travelled in a particular direction after a certain period of time based on changes in the IMU **accel** (accelerometer) XYZ direction values.

TwiceMao, 2023-03-24:
I agree with your second solution. I just need to set the initial frame to the world coordinate system (ie its pose (camera2world matrix) is a 4*4 identity matrix). The pose of each frame obtained on this basis is enough. So, for the second solution you mentioned, is there a ready-made code tool? Because I don't know anything about IMUs.

MartyG-RealSense, 2023-03-24:
Intel has an introductory guide to retrieving IMU data.

https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

For the Python langauge, https://github.com/IntelRealSense/librealsense/issues/9617#issuecomment-897414367 may be a good place to start learning about accessing and displaying the accel values.

A C++ equivalent for accessing and displaying values is at https://github.com/IntelRealSense/librealsense/issues/10903#issuecomment-1250093337

closed 2023-04-07 by MartyG-RealSense