realsenseai/librealsense #11609: D455, the number of Color images and Depth images are different

TwiceMao, 2023-03-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Windows 10| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Camera }                          | 

### Issue Description
I found that the number of color images and depth images are different after processing.
Firstly,turning on RGB, Stereo and Motion botton, I used D455 to capture imges in a in-room scene and get a .bag file. 
Then I employed rs_convert.exe to translate the .bag file into color images, depths images and some other txts.

I originally thought that there is a one-to-one correspondence between RGB images and depth images. What I mean is that an RGB image and an image are acquired by the camera at the same time, the same position, and the same orientation.
But I sort the images by filename and found that the filename of the first RGB image is _Color_1679621670605.45190429687500.png. I would have thought the filename of the first depth image would be _Depth_1679621670605.45190429687500.png. But actually, the file name of the first depth map is _Depth_1679621670605.38891601562500.png, which is not the same name as the RGB image. And there are 2320 RGB images and 2312 depth images respectively. Their numbers are not the same. 
Why is this? What do I need to do?

Also, I need to know the intrinsic parameters of the camera that captures these RGB images and depth images. But in RealSense Viewer, I see Left Intrinsics , Right Intrinsics,World to Left Rotation, World to Rigth Rotation, Rectified Resolution. I want to know what they mean and which ones do I need?

Finally, is there any tutorial for my kind of shooting with D455 to obtain RGB and depth images and the processing related to computer vision?

TwiceMao, 2023-03-24:
And could you please tell me the meaning of the file's name ? 
Like _Color_1679621670605.45190429687500.png

MartyG-RealSense, 2023-03-24:
Hi @TwiceMao  The number of the depth and color frames exported from rs-convert is almost never identical.  You have not done anything wrong and there is nothing that can be done about the difference.

Filenames are assigned by rs-convert based on frame **timestamp** and not the frame number.  So your quoted color PNG filename will be the timestamp that this particular color frame was captured at.

The RealSense SDK **Dynamic Calibration** tool's user guide provides the following information about intrinsics and extrinsics.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

[image]

Further information about intrinsics and extrinsics and their meaning can be found in the SDK's **Projection** documentation.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20

In regard to intrinsic **distortion coefficients**, these are mostly set to zero by the SDK, except for the RGB stream on the D455 model which has non-zero coefficients.

You can list the unique intrinsic and extrinsic values for your particular camera with the 'rs-enumerate-devices' tool by launching it in calibration information mode with the command **rs-enumerate-devices -c** as described at https://github.com/IntelRealSense/librealsense/issues/8689#issuecomment-809118054

Intel have an extensive list of RealSense white-paper guides and YouTube tutorial videos at the links below.

https://dev.intelrealsense.com/docs/whitepapers
https://www.youtube.com/@IntelRealSense/videos

A good place to start is the RealSense camera tuning guide:
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

And then read Intel's post-processing guide:
https://dev.intelrealsense.com/docs/post-processing-filters

TwiceMao, 2023-03-24:
I want to synchronize the RGB camera with the Depth camera, that is, there is a one-to-one correspondence between the RGB image and the depth map. How to do this?
What's more. Could you please give me a usage of rs-enumerate-devices.exe in windows? All I have now is the .bag file. Thanks very much.

MartyG-RealSense, 2023-03-24:
A RealSense team member offers advice about missing frames in rs-convert at https://github.com/IntelRealSense/librealsense/issues/3500#issuecomment-473624881

Providing a list of rs-enumerate-devices output from a D455 camera may not be useful as each camera unit has unique calibration values even when new.  I have printed the complete calibration listing from my own D455 to a PDF document and attached it below in case it is helpful to you.

[D455 Calibration.pdf](https://github.com/IntelRealSense/librealsense/files/11061735/D455.Calibration.pdf)

TwiceMao, 2023-03-25:
I did not express clearly. I want to know how to use rs-enumerate-devices.exe to get the internal parameters of the camera, because I don't know how.

MartyG-RealSense, 2023-03-25:
You can read detailed calibration information from your camera by running the rs-enumerate-devices tool with the launch command below:

**rs-enumerate-devices -c**

If you are using Windows, instructions for doing so in the Windows Command Prompt interface can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-615365409

sam598, 2023-03-25:
The D455 is not capable of synchronizing exposures of the RGB and Depth cameras. This has been a fatal flaw of every single RealSense camera. 

The current engineering team has an incorrect definition of what sync means, and the support team either does not understand the issue, or is not allowed to admit that the devices are incapable of synchronizing the color cameras.

The Oak D Pro cameras may be able to do the feature you are looking for.

MartyG-RealSense, 2023-03-25:
@sam598  The problem described by @TwiceMao is about having different numbers of frames exported from the rs-convert tool instead of both depth and color having the same number of exported frames.  It is not a question about **temporal** sync between depth and color sensors.

sam598, 2023-03-26:
@MartyG-RealSense yes that is the topic, but his goal is to sync RGB and Depth. He should know that it is not possible.

https://github.com/IntelRealSense/librealsense/issues/11609#issuecomment-1482674831

TwiceMao, 2023-03-26:
@sam598 @MartyG-RealSense My purpose is to obtain a one-to-one correspondence between the RGB image and the Depth image. They need to be seen at the same time from the same camera position. I am now using rs-convert tool to convert the .bag file to color.png and depth.png. It was found that there is no one-to-one correspondence between them. Is it possible that the depth image acquired by the D455 uses the Stereo method, so the depth image needs to be projected onto the color image. But I don't know how to do it.

sam598, 2023-03-26:
There are tools in the api for projecting depth to color, and @MartyG-RealSense can provide guidance on that.

However the two cameras will never take a photo at the exact same moment of time as they are treated as separate devices at the hardware level. 

RealSense engineers define sync as two cameras running at the same frame rate with relative time stamps. But this goes against the common industry and literal definition of sync.

The frames will still be very close together temporally, but it is not useable for 3D scanning involving moving cameras or subjects.

TwiceMao, 2023-03-26:
Sorry, I didn't understand. Can you explain in detail and give an example about what you said about RealSense engineers define sync as two cameras running at the same frame rate with relative time stamps.
 

毛懋
***@***.***

TwiceMao, 2023-03-26:
@sam598 
Sorry, I didn't understand. Can you explain in detail and give an example about what you said about RealSense engineers define sync as two cameras running at the same frame rate with relative time stamps.

sam598, 2023-03-26:
True sync would be that both the color and depth cameras take a photo at the exact same moment in time.

instead the RealSense cameras both run at the same frame rate, but run out of sync. The photos may be taken close to the same moment in time, or they might be up to half a frame offset (for example 30fps would be off by 16ms or so). With a handheld or constantly moving camera this would make it impossible to get reliable depth values for the color pixels, since they would always be misaligned.

If the camera is completely still, and there are no moving subjects, you can project the depth map onto the color to get proper correspondence.

the azure Kinect provides one of the cleanest depth maps out there, but because it’s time of flight the depth values are precise but inaccurate. With RealSense the depth maps are very noisy/imprecise but accurate because they are calibrated stereo disparity. Azure Kinect also requires a windows PC and does not have the OS and SDK/API versatility that RealSense has (although Intel has been dropping support for operating systems as the project winds down).

the oak-d project seems to try and solve a lot of the shortcomings from the RealSense cameras. However their depth do not appear to be as good as the ones the RealSense cameras produce, and they are geared for on device AI compute as opposed to just capture.

TwiceMao, 2023-03-26:
Thanks very much！

MartyG-RealSense, 2023-03-31:
Hi @TwiceMao  Do you require further assistance with this case, please?  Thanks!

TwiceMao, 2023-03-31:
No thanks. Because I use the open3d library to process the .bag file and get a synchronized RGB image and depth image.

MartyG-RealSense, 2023-03-31:
Thanks very much @TwiceMao for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-03-31 by MartyG-RealSense