realsenseai/librealsense #11612: T256 and development branch

yosoufe, 2023-03-24:
Hello, I have a question regarding the `development` branch and T265 support.
1. What is the `development` branch? Should we use the `master` or `development` branch?
2. I see the T265 is removed from the `development` branch in #11287. and also it is removed from the ros2 wrapper on `ros2-development` branch in https://github.com/IntelRealSense/realsense-ros/pull/2611.

What should I do if I want to get the T265 working with ROS 2 now?

SamerKhshiboun, 2023-03-26:
Hi @yosoufe

master - this branch contains production code. All development code is merged into master in sometime, usually before releasing a formal release of librealsense.
development - this branch contains pre-production code. When the features are finished then they are merged into development.

I think it's better to use always the **master** branch, since all releases are tagged from this branch.
See https://github.com/IntelRealSense/librealsense/wiki/Release-Notes
You can always install the latest release form this link: https://github.com/IntelRealSense/librealsense#download-and-install

Regarding your question about T265, I think you can always use the code prior to these PRs changes, and continue using librealsense and ROS2 wrapper with your T265.

@MartyG-RealSense, any other advices here?

lorenzosciuto, 2023-05-05:
`Regarding your question about T265, I think you can always use the code prior to these PRs changes, and continue using librealsense and ROS2 wrapper with your T265.`

Does this mean that the code will not be maintained anymore neither by intel nor by the community?

yosoufe, 2023-05-15:
> Does this mean that the code will not be maintained anymore neither by intel nor by the community?

Unfortunately, It seems so

mzahana, 2023-06-12:
Hi @yosoufe 
Are you able to use T265 with ROS 2 ?
I am trying to do the same, but I am not sure how.

closed 2026-10-23 by MartyG-RealSense