realsenseai/librealsense #11613: Cameras  of D455

TwiceMao, 2023-03-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
From the appearance, D455 has 4 cameras, can you tell me what are these 4 cameras? And which one or which ones are used to obtain RGB images and Depth images respectively?

sam598, 2023-03-26:
The two cameras on either end are used for stereo depth. The “left camera” (from the perspective of the cameras) is the one that the depth map is computed from. The middle camera with the same lens as the outer cameras is the color RGB camera. The other thing you see is the infrared projector. If you see a red glow coming from it, that is the infrared pattern being projected.

MartyG-RealSense, 2023-03-26:
@sam598  Thank you for your accurate description of sensor and projector positions for @TwiceMao

MartyG-RealSense, 2023-04-02:
Hi @TwiceMao  Do you require further assistance with this case, please?  Thanks!

TwiceMao, 2023-04-02:
No,thank you very much!

MartyG-RealSense, 2023-04-02:
Thanks very much for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2023-04-02 by MartyG-RealSense