realsenseai/librealsense #11614: Exception in cuda code D435i on Jetson Nano

EnthusiasticRoboticist, 2023-03-26:
I am running **realsense-ros** with **docker** on **Jetson Nano** with **D435i**. I get the following exception which I think it comes from [this line](https://github.com/IntelRealSense/librealsense/blob/master/src/cuda/cuda-conversion.cu#LL237C8-L237C8) from **librealsense**. I put a print before and after it and only get the ones before it.

The exception is like this
```
[realsense2_camera_node-1]  26/03 02:34:35,950 ERROR [546844105120] (synthetic-stream.cpp:51) Exception was thrown during callback: loc_devIhEESt10shared_ptrIT_EiEUlPhE_
```

`rs-depth`  works fine in the same environment. 

Here is the [dockerfile](https://github.com/EnthusiasticRoboticist/DonkeyJet/blob/main/Dockerfile.realsense) that I am using.

MartyG-RealSense, 2023-03-26:
Hi @EnthusiasticRoboticist  May I ask which librealsense version and RealSense ROS wrapper version you are using, please?

EnthusiasticRoboticist, 2023-03-28:
[Debug.log](https://github.com/IntelRealSense/librealsense/files/11085034/Debug.log)
@MartyG-RealSense This file is the output of 
```
ros2 launch realsense2_camera rs_launch.py log_level:=DEBUG
```
if you think it might help. Thanks

MartyG-RealSense, 2023-03-28:
You are using librealsense 2.53.1 and ROS wrapper 4.51.1 according to the debug log file.  Thank you.

 The log also indicates that **control_transfer returned** warnings are occurring.  This can indicate a communication problem between the camera and the computer / computing device (the Jetson in this case).

And then immediately after launch, the synthetic-stream error occurs.

CUDA support in librealsense and the ROS wrapper only accelerates three specific types of operation: depth-color alignment, pointcloud generation and YUY color conversion.  If you are not using alignment or pointclouds then you could likely leave **-DBUILD_WITH_CUDA=true** out of the dockerfile and see whether that prevents the synthetic-stream error occurring.

MartyG-RealSense, 2023-04-02:
Hi @EnthusiasticRoboticist  Do you require further assistance with this case, please?  Thanks!

EnthusiasticRoboticist, 2023-04-04:
I was using a wrong docker image as the base and GPU wasn't accessible to the library inside the docker container. 
I am closing this. Thanks

closed 2023-04-04 by EnthusiasticRoboticist