realsenseai/librealsense #11617: Labview-D405

CJA-JIANG, 2023-03-27:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D405 } | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  V2.53.1                      | 
| Language                            |  {labview }                          | 
| Segment			| others                          | 

### Issue Description
Dear friends, I have encountered this problem when using D405, please help to find out what the problem is, thank you.
I can successfully get the image from the example

[image]
-
[image]
|---------------------------------|------------------------------------------- |
But I can't get the image in labview.(The labview library I use is RealSense for Labview ver.2.33）
[image]
[image]
|---------------------------------|------------------------------------------- |
This is my test vi, please take a look（I modified the example "RealSense_HelloWorld_PointCloud.vi" mainly by removing the advanced configuration）
[image]
[image]
|---------------------------------|-------------------------------------------
Hope to get your help as soon as possible, thank you.

MartyG-RealSense, 2023-03-27:
Hi @CJA-JIANG  Official support for the D405 camera model was introduced in SDK **2.51.1**.  So whilst 2.53.1 can detect and use the camera, 2.33.1 will not be able to detect it.

I tested the D405 in the 2.33.1 version of the RealSense Viewer tool and confirmed that it was not detectable.

CJA-JIANG, 2023-03-28:
> Hi @CJA-JIANG Official support for the D405 camera model was introduced in SDK **2.51.1**. So whilst 2.53.1 can detect and use the camera, 2.33.1 will not be able to detect it.
> 
> I tested the D405 in the 2.33.1 version of the RealSense Viewer tool and confirmed that it was not detectable.
|---------------------------------|------------------------------------------- |
Hi @MartyG-RealSense 
I have replaced realsense2.dll in RealSense for Labview ver. 2.33.0 with realsense2.dll in SDK 2.51.1, but it does not work.

MartyG-RealSense, 2023-03-28:
There was a similar issue at https://github.com/IntelRealSense/librealsense/issues/7920 when using D455 with the LabVIEW wrapper, as its most recent update uses SDK 2.33.0 and D455 support was not introduced until SDK 2.38.0.  The RealSense user in that case copied the more modern realsense2.dll file to the LabVIEW folder and updated their camera firmware driver and it worked for them.

If you are using SDK 2.53.1 then you could copy the realsense2.dll from 2.53.1 and make sure that your camera firmware driver version is **5.14.0.0**.

On Windows a pre-made dll can be obtained by installing version 2,53,1 of the full RealSense SDK using the **Intel.RealSense.SDK-WIN10** installer file first.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

[image]

Afterwards, go to the computer location below and copy the realsense2.dll file from this folder into the LabVIEW folder.

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > bin > x64**

CJA-JIANG, 2023-03-29:
Currently only the 2.51.1 and 2.53.1 SDKS support D405. I updated realsense2.dll to the labview folder several times and updated the firmware, and finally I could capture the image. But now there is a new problem, the color image is still not there, is there something wrong with my option?
[image]

MartyG-RealSense, 2023-03-29:
On the D405 camera model there is not an RGB sensor, as RGB is obtained from the depth sensor instead.  Programs such as video chat applications that are designed to detect the RGB sensor may therefore not work as the D405 does not have an RGB camera driver, only a depth driver.

CJA-JIANG, 2023-03-31:
A strange situation occurred. The day before yesterday, D405 was able to get images using labview normally. However, when I tested it again today, it could not get images without modifying the program, and it still prompted Frame Timeout. 
I tried to degrade the firmware, but it didn't work. I can't degrade the firmware now.
[image]

When will there be another appropriate SDK update?

MartyG-RealSense, 2023-03-31:
How many times did you attempt to downgrade the firmware driver please?  RealSense cameras have a limit of 20 firmware downgrades, after which the camera can only be upgraded to a newer firmware version than the currently installed one for the rest of the camera's lifetime.

There is not a date available for the next RealSense SDK or firmware driver release, though the in-development version of the SDK that will become the next release can be found on the SDK's **development** GitHub branch.

https://github.com/IntelRealSense/librealsense/tree/development

CJA-JIANG, 2023-04-01:
I was able to get depth images as well as color images.

I'm using the following configuration
realsense2.dll：V2.51.1
firmware driver version ： 5.14.0.0
The color diagram cannot be obtained for the following reasons：
[image]
[image]
The index of a deep stream cannot be the same as that of a color stream, which will result in a different frame rate and an error.

MartyG-RealSense, 2023-04-01:
Do you have the option to manually set depth to a frame rate of 60-90 instead of 30-60?  If you do, does doing this give left-color the same frame rate?

closed 2023-04-13 by MartyG-RealSense

MartyG-RealSense, 2023-08-14:
Hi @CJA-JIANG  A user of D405 with LabVIEW at https://github.com/IntelRealSense/librealsense/issues/12093#issuecomment-1675228924 managed to get it to work.