realsenseai/librealsense #11621: D450 + D4 board - Invalid SKU

AhmedMahmoud2023, 2023-03-27:
Just purchased RealSense Depth Module D450 + Vision Processor D4 Board V3 Bundle with HSF50 and HSF100 interposer cables.

The realsesne viewer is not detecting the device on (Windows 11-Ubuntu 20.04-Ubuntu 22.04) and it says 

The output of lsusb is: 
                             Bus 002 Device 034: ID 8086:0af5 Intel Corp. Intel(R) RealSense(TM) Invalid SKU

Kindly, any help would be appreciated.

MartyG-RealSense, 2023-03-27:
Hi @AhmedMahmoud2023  Could you first confirm whether you have the interposer cable plugged into the correct one of the two available 50-pin connectors on the Vision Processor D4 V3 Board?  The correct one (the Depth Module Receptacle)  is shown on the diagram below.

[image]

AhmedMahmoud2023, 2023-03-27:
Hi @MartyG-RealSense , thank you for your help. Yes, the cable is plugged to the right connector.

AhmedMahmoud2023, 2023-03-27:
@MartyG-RealSense, The terminal output when running the Realsense-Viewer is : 
                   WARNING [140371938793280] (enumerator-libusb.cpp:45) failed to read USB config descriptor: error = -1

The realsense-viewer displays a message: 
                 Connect a realsense camera or add a source.
The lsusb output is:
                  ID 8086:0ad6 Intel Corp. Intel(R) RealSense(TM) Depth Camera 405i Tester

I get the same output on different computers with different D450+D4 boards.

MartyG-RealSense, 2023-03-27:
If you purchased the 8-pack of D450 and D4 V3 from the RealSense Store then you will have the correctly matched board products.  

If you purchased the HSF50 and HSF100 interposer cables from the RealSense Store too then they should also be fine.

A possible problem point could be the USB cables, which would have been purchased separately as they are not supplied in the module pack.  Could you provide details of the USB cable that you are using, please?

JyotiLuxolis, 2023-03-28:
Hello, I'm working with Ahmed on this. We bought the 8-pack D450 and D4 V3 from the RealSense Store. We are working with and have used both HSF50 and HSF100 interposer cables. Both give us the same errors. 

We are using the USB-C to USB-A that is provided with D455 camera. So, this shouldn't be the issue.

Just to clarify, if we only connect the D4 V3 board (D450 not connected) we get `Intel Corp. Intel(R) RealSense(TM) Depth Camera 405i Tester` when we run lsusb

If we connect D4 V3 + D450 we get `Intel Corp. Intel(R) RealSense(TM) Invalid SKU` on lsusb

We installed the realsense SDK on Ubuntu 20 Kernel 5.15 using the instructions here: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

Can you guide us step by step what we should do to get D4 V3 + D450 working on Ubuntu 20?

MartyG-RealSense, 2023-03-28:
Just to confirm that you were shipped the correct products, the D4 V3 (below) and D450 (above) look like the image below, right?

[image]

AhmedMahmoud2023, 2023-03-28:
[image]

MartyG-RealSense, 2023-03-28:
I will discuss your case with my Intel RealSense colleagues.  Thanks very much for your patience!

MartyG-RealSense, 2023-03-28:
My colleagues advised that the D4 V3 board might be pre-loaded with older firmware which doesn’t support D450.  So to rectify this, you can just directly update the D4 V3 board with the latest firmware.  There is no need to connect the D450 module in order to perform the update.  

After the firmware update, disconnect the USB cable and connect the D450 back to the V3 D4 board.  Then check it again on your PC.

JyotiLuxolis, 2023-03-28:
Hello, So, when we connect the D4 V3 board directly without D450 lsusb gives us: `Intel Corp. Intel(R) RealSense(TM) Depth Camera 405i Tester`. 

If we try using `rs-fw-update` on D4 V3 (without D450) we still get `No device detected`.

MartyG-RealSense, 2023-03-29:
My colleagues asked if you can perform the firmware update on the Windows computer without the D450 attached, and then repeat the firmware update on the Ubuntu computer afterwards.

AhmedMahmoud2023, 2023-03-29:
Hello @MartyG-RealSense, 
Thanks for your help. 

The device is read on Windows as Intel Corp. Intel(R) RealSense(TM) Depth Camera 405i Tester.

[image]

If we try using rs-fw-update on D4 V3 (without D450) we can not list the device or detect it.

[image]

MartyG-RealSense, 2023-03-29:
Thanks very much for testing.  I will feed your latest results back to my colleagues for further advice.

MartyG-RealSense, 2023-03-30:
I had further discussions with my RealSense colleagues.  Could you email me at the address below please so that I can respond to you by email, please?

martyx.grover@intel.com

MartyG-RealSense, 2023-04-23:
Hi @AhmedMahmoud2023 and @JyotiLuxolis  Do you have an update about this case that you can provide, please?  Thanks!

MartyG-RealSense, 2023-05-02:
Hi @AhmedMahmoud2023 and @JyotiLuxolis  Do you require further assistance with this case, please?  Thanks!

zackwoodruff, 2023-05-04:
@MartyG-RealSense @AhmedMahmoud2023 @JyotiLuxolis 
I had the same issue with a batch of recently purchased RealSense Depth Module D450 + Vision Processor D4 Board V3 Bundle but I found a workaround. 

I believe the issue is with the RealSense Depth Module D450. In the image provided above (https://github.com/IntelRealSense/librealsense/issues/11621#issuecomment-1487048589) it shows the number "K83121-101" on the depth module which is the same label as the ones I had issues with. 

I was unable to program the firmware on the Intel RealSense Vision Processor D4 Board V3
1. when disconnected from the depth module
2. when connected to any of the modules labeled with "K83121-101"

I had a D450 depth module from a previous order labeled "K83121-100", and when I attached the Vision Processor D4 Board V3 to the older module it properly showed up on my Ubuntu 20.04 laptop, it showed up in realsense-viewer with the option to upgrade the firmware, and I was able to enumerate with `rs-enumerate-devices -l`:
`1) Name: Intel RealSense D455, serial number: 152522250847, update serial number: 210523063429, firmware version: 05.12.07.150, USB type: 3.2`

I upgraded all the boards using the depth module labeled "K83121-100" and then tested them with the newer depth modules labeled "K83121-101" and they seemed to work fine. 

@MartyG-RealSense Please let me know if there are any known issues with the "K83121-101" otherwise I'm assuming the cameras are functional now.

MartyG-RealSense, 2023-05-05:
Hi @zackwoodruff  Thanks so much for your detailed feedback and workaround.  I will pass the information and your question about K83121-101 to my Intel RealSense colleagues.

MartyG-RealSense, 2023-05-05:
My colleagues responded that K83121-101 is a newer D450 module, and it needs a newer firmware to work with it.  They thought that you may have got an older version of the ASIC card (Vision Processor D4) and the firmware in it does not work with K83121-101.

They added that there are two options to get it work:

1.  Update the firmware to the latest for the ASIC card, without the D450 module connected.

2.  Do what you did in your workaround and connect a K83121-100, update firmware and then change back to K83121-101

zackwoodruff, 2023-05-05:
@MartyG-RealSense thank you for your help. Option 2 did work for me but Option 1 did not work for me or other users in this thread. I tried upgrading the ASIC card using the following methods with the 2.53.1 [librealsense](https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1) and the D450 module disconnected
1. Ubuntu 20.04:
   1. rs-fw-update: could not detect the device or update firmware
   2. realsense-viewer: did not detect the device
2. Windows 10:
   1. Intel.RealSense.FW.Update.exe: could not detect the device, gave null pointer exception (same as this earlier comment: https://github.com/IntelRealSense/librealsense/issues/11621#issuecomment-1488834163)
   2. Intel.RealSense.Viewer.exe: could not detect the device

MartyG-RealSense, 2023-05-05:
Thanks very much @zackwoodruff for your confirmation of option 2 as a successful strategy for you.

MartyG-RealSense, 2023-05-12:
Case closed due to no further comments received.

closed 2023-05-12 by MartyG-RealSense

emqshaas, 2023-09-04:
Hello @MartyG-RealSense, I am having the same problem as @AhmedMahmoud2023 with a RealSense Depth Module D450 (K83121-101) + Vision Processor D4 Board V3 (K94249-110) with a HSF100 interposer cable. Windows 10 is recognizing the D4 Board as Camera Tester 405i without the D450 camera module connected. I tried to "Update the firmware to the latest for the ASIC card, without the D450 module connected" as suggested with rs-fw-update.exe or with RealSenseViewer but I always receive an error: "04/09 15:30:49,755 WARNING [3852] (rs.cpp:310) null pointer passed for argument "device".

Is there another way to update the ASIC on the D4 Board V3? I do not have another D450 Camera module with an older serial number that I can connect to the D4 Vision board.

MartyG-RealSense, 2023-09-04:
Hi @emqshaas  I will discuss your case of this particular issue with my Intel RealSense colleagues.

MartyG-RealSense, 2023-09-04:
Hi again @emqshaas  My colleagues advised that they may be able to ship you a D4 V3 card with the latest firmware on it.  How many D4 V3 cards would you need to replace your affected one(s), please?

emqshaas, 2023-09-05:
Hi @MartyG-RealSense, thanks for the quick reply. Yes it would only be one D4 V3 card.

MartyG-RealSense, 2023-09-05:
Please send me an email with your name, email address, telephone number and a postal address for shipping the D4 V3 card to you and I will pass the details to my colleagues.

martyx.grover@intel.com

arpaterson, 2023-09-07:
I am experiencing the same issue described above. Using a new D455 camera.

"Invalid SKU",
"8086:0af5"
" (enumerator-libusb.cpp:45) failed to read USB config descriptor"

fw update won't initiate under windows or ubuntu, although in ubuntu i can at least see the serial number

I do not have another to disassemble and attempt flashing with the earlier board.

MartyG-RealSense, 2023-09-08:
Hi @arpaterson  Thanks very much for your message and for the email that you sent.  I will respond via the email so that your case does not become mixed up with that of @emqshaas