realsenseai/librealsense #11623: Hi Team when I am running the  example code rs-data-collect or rs-record or rs-hello-realsense , getting the error RealSense error calling rs2_open_multiple(sensor:0xaaaad7397040, profiles:0xaaaad7482c60, count:1):     xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy. Will any one help me?

drakshayanidesai, 2023-03-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

drakshayanidesai, 2023-03-28:
Device selected:
        Intel RealSense D435. 
       OS: Linux
       Platform: NVIDIA Jetson 
[image]

MartyG-RealSense, 2023-03-28:
Hi @drakshayanidesai  When you launch these programs, can you confirm please that you have no other programs running that are actively using the camera, such as the RealSense Viewer tool.  

If a stream is enabled in one program then another program that is subsequently launched cannot access that stream until the first program stops using it, which may result in a 'busy' error.

drakshayanidesai, 2023-03-28:
Hi

How to identify which task is accessing camera and how to stop it?

Regards
Drakshayani

On Tue, 28 Mar 2023 12:16:45 +0530 MartyG-RealSense  wrote

Hi @drakshayanidesai  When you launch these programs, can you confirm please that you have no other 
programs running that are actively using the camera, such as the RealSense Viewer tool.
If a stream is enabled in one program then another program that is subsequently launched cannot access 
that stream until the first program stops using it, which may result in a 'busy' error.

closed 2023-04-10 by MartyG-RealSense