realsenseai/librealsense #11628: Hi Team I am facing another issue whenever I an running rs-measure.cpp  it is showing the message like: please check your graphic drivers or use the textual sdk tools

drakshayanidesai, 2023-03-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-03-28:
Hi @drakshayanidesai  Whilst some RealSense SDK example programs such as rs-hello-realsense output plain text, other examples require graphics support.  This support may be absent in 'headless' computer systems that are configured to be used without a display and cannot run programs that require graphics instead of plain text.

drakshayanidesai, 2023-03-28:
Hi 

I am actually accessing Realsense remotely from virtual machine. I order to run this rs-measure, what I 
need to do?

Regards
Drakshayani

On Tue, 28 Mar 2023 19:54:08 +0530 MartyG-RealSense  wrote

Hi @drakshayanidesai  Whilst some RealSense SDK example programs such as rs-hello-realsense output 
plain text, other examples require graphics support.  This support may be absent in 'headless' computer 
systems that are configured to be used without a display and cannot run programs that require graphics 
instead of plain text.

MartyG-RealSense, 2023-03-28:
What metjod did you use to install the librealsense SDK on the virtual machine, please?  Did you build it from source code with CMake or install it from packages?

drakshayanidesai, 2023-03-28:
I followed these steps
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

On Tue, 28 Mar 2023 20:03:33 +0530 MartyG-RealSense  wrote

What metjod did you use to install the librealsense SDK on the virtual machine, please?  Did you build it 
from source code with CMake or install it from packages?

drakshayanidesai, 2023-03-28:
I am using linux virtual machine where realsense camera is there

On Tue, 28 Mar 2023 20:03:33 +0530 MartyG-RealSense  wrote

What metjod did you use to install the librealsense SDK on the virtual machine, please?  Did you build it 
from source code with CMake or install it from packages?

drakshayanidesai, 2023-03-28:
Hi

While installing Intel Realsense SDK2.0 got the error as shown in the attachment. Camera not opened 
after running realsense-viewer.

Regards
Drakshayani

On Tue, 28 Mar 2023 20:03:33 +0530 MartyG-RealSense  wrote

What metjod did you use to install the librealsense SDK on the virtual machine, please?  Did you build 
it from source code with CMake or install it from packages?

drakshayanidesai, 2023-03-29:
[Document14.docx](https://github.com/IntelRealSense/librealsense/files/11096194/Document14.docx)

MartyG-RealSense, 2023-03-29:
Thank you very much for the image.

[image]

The most important message in the log is likely to be **E: Unable to locate package librealsense2-dkms**.  This package is the main package of the librealsense SDK and therefore the most important.

Are you using a virtual machine that has emulation of the USB controller, such as the **VMWare Workstation Player** tool recommended by Intel at the link below?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

MartyG-RealSense, 2023-03-30:
RealSense users have been able to successfully install librealsense on Jetson Orin, as described at https://github.com/IntelRealSense/librealsense/issues/10722#issuecomment-1230250762

Successful installations from source code were made either by not patching the kernel, or by building the SDK in RSUSB backend = true mode so that the SDK is not dependent on kernel versions or Linux versions and does not require kernel patching.

The reason for not patching the kernel when not building from source in RSUSB = true mode is that the SDK will not officially support JetPack 5 until the next SDK version (likely numbered 2.54.0).

As you are installing from packages though with the distribution_linx.md instructions, the patch that is bundled inside those packages will not be Jetson-compatible.  Jetson has its own dedicated L4T patch and Jetson-specific installation instructions.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

closed 2023-04-13 by MartyG-RealSense