realsenseai/librealsense #11630: Wrong firmware upload

MetehanGokmen, 2023-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400                                | 
| Firmware Version                   |  12_13_50                        | 
| Operating System & Version |    Linux (Ubuntu 20.04)     | 
| Kernel Version (Linux Only)    |  5.15.0-69-generic           | 
| Platform                                 | PC                                    |
| SDK Version                            | 2.45.0                             | 
| Language                            |   python                               |  
| Segment			|  {Robot}                                    | 

I accidentally uploaded the wrong firmware on my RealSense camera and now I am unable to open it. I have tried connecting it through USB, and although it is being detected, it does not work on the RealSense viewer. Are there any chances of recovering it?

##realsense-viewer

27/03 18:08:14,120 INFO [139998431339264] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
27/03 18:08:14,120 INFO [139998431339264] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
27/03 18:08:14,120 INFO [139998431339264] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
27/03 18:08:14,202 INFO [139998431339264] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
27/03 18:08:14,316 INFO [139998431339264] (context.cpp:382) Found 2 RealSense devices (mask 0xff)
27/03 18:08:14,353 INFO [139998431339264] (rs.cpp:2701) Framebuffer size changed to 1848 x 1016
27/03 18:08:14,353 INFO [139998431339264] (rs.cpp:2701) Window size changed to 1848 x 1016
27/03 18:08:14,353 INFO [139998431339264] (rs.cpp:2701) Scale Factor is now 1
27/03 18:08:14,446 WARNING [139997931894528] (rs.cpp:310) null pointer passed for argument "device"
27/03 18:08:14,446 WARNING [139997931894528] (rs.cpp:2704) Couldn't refresh devices - hwmon command 0x10( 0 0 0 0 ) failed (response -844431340= unknown)
27/03 18:08:14,697 INFO [139998431339264] (context.cpp:382) Found 1 RealSense devices (mask 0xfe)

##rs-fw-update -l 

Connected devices:
27/03 18:09:28,923 WARNING [140446884438208] (rs.cpp:310) null pointer passed for argument "device"
RealSense error calling rs2_create_device(info_list:0x561119864dd0, index:0):
hwmon command 0x10( 0 0 0 0 ) failed (response -844431340= unknown)

##rs-enumrate-devices

Could not create device - hwmon command 0x10( 0 0 0 0 ) failed (response -844431340= unknown) . Check SDK logs for details

##lsusb

Bus 001 Device 010: ID 8086:0ad6 Intel Corp. Intel(R) RealSense(TM) Invalid SKU

MartyG-RealSense, 2023-03-29:
Hi @MetehanGokmen  Do you have access to a Windows computer too, please?  In another case of Invalid SKU currently open at https://github.com/IntelRealSense/librealsense/issues/11621 my Intel RealSense colleagues advised performing a firmware update on a Windows computer first, which should hopefully reset the camera ASIC, and then afterwards performing the firmware update again on an Ubuntu computer to complete the firmware fix.

In https://github.com/IntelRealSense/librealsense/issues/11621 the reason why Invalid SKU was occurring was suspected to be because an old firmware was installed in the camera hardware that was not compatible with the camera model.

MetehanGokmen, 2023-03-29:
Yes, I can access a Windows computer. Can you explain to me exactly what I need to do?

sametst, 2023-03-29:
[image]

First of all, we work together with MetehanGokmen. I am a Microsoft Windows user. As you can see here, there is no section as you mentioned in the 'more options' part. If you want us to enter 'more options' from the interface that appears after reading the camera in the source section, let me remind you again; our camera is not detected by the viewer in any way. I have tried and checked all versions of Viewer from 2.38.0 to 2.53.1

MartyG-RealSense, 2023-03-30:
Hi @sametst  Is the camera also not detected by the **rs-fw-update** firmware update tool that is run in the Windows Command Prompt interface?

sametst, 2023-03-30:
yes

MartyG-RealSense, 2023-03-30:
@sametst   I have highlighted your problem to my Intel RealSense colleagues.  Thanks very much for your patience!

MartyG-RealSense, 2023-04-05:
Hi @sametst  Thanks very much for waiting.  The problem that you have may be fixable by removing the **Vision Processor D4 Board** circuit board inside the camera, though opening up the camera will invalidate the sales warranty.

MartyG-RealSense, 2023-04-10:
Hi @MetehanGokmen and @sametst  Do you have an update about this case that you can provide, please?  Thanks!

closed 2023-04-23 by MartyG-RealSense