realsenseai/librealsense #11631: How do I stream Point Cloud to "Unity" via Python server that connects RealSense cameras?

EunPyoLee1010, 2023-03-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win10 | 
| Platform                                 | PC |
| SDK Version                            |  2.x.x                          | 
| Language                            |  python/unity                           | 
| Segment			|  VR                           | 

### Issue Description

Realsense examples show how to *directly* connect and stream the realsense camera to each python server and unity.

However, I would like to receive point cloud data from python server connected to the realsense camera (whether streaming data or just one point cloud data) and visualize that data in unity scene.

But I don't know how to transfer data from Python server to unity because the point cloud data type of python realsense sdk (pyrealsense2) and unity realsense sdk are different.

I'm trying to communicate with python and unity through socket network.

I would appreciate it if you could let me know if you have any codes or links for reference.

MartyG-RealSense, 2023-03-29:
Hi @EunPyoLee1010  The Python-based **EtherSense** networking tool makes use of socket.  Links to the tool and a guide for it can be found at https://github.com/IntelRealSense/librealsense/issues/10700#issuecomment-1191377934

There is not currently a method of connecting EtherSense to Unity though.

 A RealSense user at https://github.com/IntelRealSense/librealsense/issues/9588 took the approach of rendering frames from a network as a pointcloud in Unity using a **software device**.  The RealSense SDK's software device interface enables non-standard data to be rendered by the RealSense SDK as though a real physical RealSense camera was providing the data.  However, software device works best with C++ and use of it in Python tends to have problems.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

EunPyoLee1010, 2023-03-30:
Thank you for your comment @MartyG-RealSense ! and here's my answer.
First, I have to render point cloud data in unity.
Also, I don't need to connect to Unity with EtherSense (as I said above), just one point cloud data is sent, so if it's rendering from Unity, that's fine.

Currently, I am also considering a way to create a point cloud by sending a 3d object file (ply, pcd, etc.) to unity through socket communication and reading the file from Unity. Is it impossible to do this?

Without streaming, any method of rendering point cloud 3d objects by sending point cloud data to unity is fine.

Please confirm this.
Thank you.

MartyG-RealSense, 2023-03-30:
https://github.com/bertaye/UnityPlyImporter has a ply importer for Unity that was published relatively recently, though it does not support color.

The Unity Asset Store also has a model import tool that lists ply format support.

https://assetstore.unity.com/packages/tools/modeling/trilib-2-model-loading-package-157548

closed 2023-04-13 by MartyG-RealSense