realsenseai/librealsense #11633: how to take image with python from a framos intel reals industrial depth sensor D455e

Stephen12ds, 2023-03-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Stephen12ds, 2023-03-30:
i am using this code:
import pyrealsense2 as rs

# Create a context object. This object owns the handles to all connected realsense devices
ctx = rs.context()

# Query the available devices and filter by product line and connection type
devices = ctx.query_devices()

if len(devices) == 0:
    print('No device connected')
    exit()
else: 
    print('Device connected')

# Create a pipeline object and configure the streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_device(devices[0].get_info(rs.camera_info.serial_number))
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

# Start streaming
pipeline.start(config)

while True:
    # Wait for new frames and retrieve the depth frame
    frames = pipeline.wait_for_frames()
    depth = frames.get_depth_frame()
    if not depth:
        continue

    # Print a simple text-based representation of the image
    coverage = [0] * 64
    for y in range(480):
        for x in range(640):
            dist = depth.get_distance(x, y)
            if 0 < dist and dist < 1:
                coverage[x // 10] += 1

        if y % 20 == 19:
            line = ""
            for c in coverage:
                line += " .:nhBXWW"[c // 25]
            coverage = [0] * 64
            print(line)

and it picks up a camera connected via usb but not one that is setup over a network

MartyG-RealSense, 2023-03-30:
Hi @Stephen12ds  Intel are unfortunately unable to assist with questions related to the FRAMOS 'e' series of industrial RealSense-compatible cameras as they are not Intel products.  Please contact FRAMOS support by visiting the support page at the link below and clicking the **Let's talk** button.

https://www.framos.com/en/resources/support

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.

closed 2026-10-23 by MartyG-RealSense