realsenseai/librealsense #11638: Exporting ply file when multiple cameras are connected

Ankush1909IIT, 2023-03-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Operating System & Version |   {Win (8.1/10)  | 
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi,

I have used this code to export_to_ply for one camera connected.

`import numpy as np                     
import pyrealsense2 as rs             
import numpy as np

print("Environment Ready")

pc = rs.pointcloud()
pipe = rs.pipeline()

#Create a config and configure the pipeline to stream

config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)

# Start streaming
profile = pipe.start(config)
align_to = rs.stream.color
align = rs.align(align_to)

for x in range(10):
	pipe.wait_for_frames()

frames = []
for x in range(2):
	frames = pipe.wait_for_frames()
	aligned_frames = align.process(frames)

	# Get aligned frames
	aligned_depth_frame = aligned_frames.get_depth_frame() 
	color_frame = aligned_frames.get_color_frame()
	
	points = pc.calculate(aligned_depth_frame)
	pc.map_to(color_frame)
	print("Saving to 1.ply...")
	points.export_to_ply("1.ply", color_frame)
print("Done")	
`

I was just wondering if we have a code to perform the same task, when multiple cameras are connected. If I have four cameras, it should return 4 ply files, one for each camera.

MartyG-RealSense, 2023-03-31:
Hi @Ankush1909IIT  It is possible to export data to file from multiple cameras by creating a separate pipeline and configuration for each camera, like the Python script at https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-390840337

Whilst that script is for exporting depth and color as jpg image files, you may be able to adapt the script for .ply export instead.

MartyG-RealSense, 2023-04-07:
Hi @Ankush1909IIT  Do you require further assistance with this case, please?  Thanks!

Ankush1909IIT, 2023-04-12:
Hi @MartyG-RealSense , I am working on it, by trying to modify the code that you shared. I will update here if i can get it working.

MartyG-RealSense, 2023-04-13:
Thanks very much @Ankush1909IIT for the update.  Good luck!

MartyG-RealSense, 2023-05-02:
Hi @Ankush1909IIT  Do you have an update about this case that you can provide, please?  Thanks!

Ankush1909IIT, 2023-05-08:
Unfortunately no. I will close this meanwhile. Thanks

closed 2023-05-08 by Ankush1909IIT