realsenseai/librealsense #11641: No Frames Detected Issue D435I

willyum00, 2023-04-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | (05.14.00.00) | 
| Operating System & Version |   Win 11  | 
| Platform                                 | PC  |
| SDK Version                            |   2.53.1.4623                        | 
| USB type | 2.1 |
| Processor | 11th Gen Intel(R) Core(TM) i5-1145G7 @ 2.60GHz   1.50 GHz | 

### Issue Description
I am currently experiencing issues with a real sense d435i. In the real sense viewer, when I turn on either stereo module or RGB camera, I get a No Frames Received error. 
[image]
The motion module works fine though. 
[image]

I have tried turning each stream on and off individually, no change. 
The resolution of both RGB and stereo stream is 640 x 480. 

Here are the current module settings: 
[image]

I have already tried running the rs-capture program. It simply gives me a black screen followed by this error: 

[image]

I installed the SDK from here: 
[image]

I do not have your official usb cable, I am currently using an Anker brand usb cable that is about 22 inches in length. 

I am fairly sure that the issue is with the camera and not with my system, as I experienced the same issue when trying it on a different computer. 

I also tried opening the default windows camera app but received this error: 

[image]

How can I fix this? I realize there are possibly multiple things that might be wrong here. Thanks.

MartyG-RealSense, 2023-04-01:
Hi @willyum00  It is not unusual to receive a message from the Windows **Camera** app about being unable to start a RealSense camera when the app first starts.  Please try left-clicking on the **Change camera** option in the top corner of the Camera app window to see whether the RealSense camera's RGB video appears.

[image]

After rebooting your computer, please try enabling depth and RGB in the RealSense Viewer but **not** enabling Motion Module.  This is because there is a known issue with the Viewer where enabling three stream types causes No Frames Received if one of the three streams is the Motion Module.  

It is always the Motion Module enabling that is the trigger for the problem, so enabling the three streams in a different order makes no difference.  Once the issue has been triggered then only a reboot of the computer will correct it as it continues if you close and re-open the Viewer.  There is no fix for this in the Viewer, unfortunately other than to only enable two streams if you need to use the Motion Module (such as Motion Module and Depth or Motion Module and RGB).

If you have the streams enabled in the Viewer and then subsequently launch another RealSense program such as rs-capture, the second program will not work because the Viewer has already placed an exclusive claim on access to the streams.  The streams in the Viewer must be disabled or the Viewer closed down before launching another RealSense program.

willyum00, 2023-04-01:
Hi @MartyG-RealSense 

I knew that there would be an issue with enabling the motion module first. I rebooted my computer and only enabled the stereo module. I still got the no frames detected error. 

Also, I was not clear about the windows app issue. I knew to click the change camera symbol. When I do that, I simply get a grayed out screen. 
[image]

Also, I did not have the stream enabled when using rs-capture. I have read through many of the past issues related to mine, and have tried the solutions you suggested to other people. I think my issue might have a different cause.

MartyG-RealSense, 2023-04-01:
A 22 inch cable is about 0.5 meters and so the length of your own choice of cable should not be causing a problem.  It should be a cable designed for transmitting data though and not an inexpensive device charging cable.

When you expand open the Stereo Module section of the Viewer, are the Infrared streams listed there or is there only the Depth stream listed?

willyum00, 2023-04-05:
This is what I see: 
[image]

Also, I am using an Apple usb-c to usb-c cable, so I feel like it should be able to transmit data no problem.

willyum00, 2023-04-05:
Hi Marty, I set the frame size to the lowest setting and I finally got a non-white screen, but it's definitely not fully functional: 
[image]

At this point, the issue is either with my system or my cable. I just switched to an Anker brand cable as well. I will try with a more powerful system first, but what brand of cable can you recommend me to use since I no longer have the cable that the camera came with?

willyum00, 2023-04-05:
On a different machine, setting the RGB camera resolution to 1920x1080 enabled it, though at a very slow framerate (about 1 frame every 3 seconds). It works briefly, but too much movement will freeze the video and it will say "no frames detected" until I stop moving it for a bit. 

The RGB camera only works at 1920 x 1080. 
The stereo module only works at 640 x 480.

willyum00, 2023-04-05:
Should I attempt a reset or find a different cable? What cable brand do you recommend that would work with the realsense?

willyum00, 2023-04-05:
Realsense viewer is detecting a 2.1 connection but in device managers I clearly have 3.0 ports. Is this indicating that the cable I am using is 2.1, and I need to switch to a 3.0 or higher connector?

MartyG-RealSense, 2023-04-05:
USB C to C cables typically do not work well with RealSense cameras unfortunately.  Problems have especially been reported with C to C on Apple Macs, where this type of connection is most likely to be used.

Do you have any USB-A ports on your computer so that a Type-C cable (A to C) like the one supplied with the camera could be used instead of C to C, please?

willyum00, 2023-04-05:
I am now testing the Realsense with a Newnex brand industrial grade cable.

willyum00, 2023-04-05:
Thanks for your help, the cable was indeed the issue.

closed 2023-04-05 by willyum00

MartyG-RealSense, 2023-04-05:
You are very welcome.  It's great to hear that you found a solution.  Thanks very much for the update!