realsenseai/librealsense #11642: Perform Hardware Reset With `rs-terminal`

stephanrotolante, 2023-04-01:
- Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

  - Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
  - Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
  - Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
  - Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

- All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

---

| Required Info              |                                        |
| -------------------------- | -------------------------------------- |
| Camera Model               | L515                                   |
| Firmware Version           | (Open RealSense Viewer --> Click info) |
| Operating System & Version | Win (8.1/10)                           |
| Platform                   | PC                                     |
| SDK Version                | 2.53.1                                 |
| Language                   | unity                                  |
| Segment                    | others                                 |

### Issue Description

<Describe your issue / question / feature request / etc..>

I am trying to execute a hardware reset via the `rs-terminal` tool for a L515 sensor however nothing is happening. I've dug into the source code an saw that the hex value for `hwreset` is `0x28`. This is the command I am running:

`rs-terminal.exe -a -s "14 00 ab cd 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00"`

I saw another [question](https://github.com/IntelRealSense/librealsense/issues/9622) opened about the `rs-terminal`. I tried my best to follow it

### Why am I doing this?

A third party company we are working with is using implemented LiDar cameras for a project. The sensors are locking up after some use and the only way to resolve the issue is to cycle the power.
The issue were experiencing is were unable to cycle the power without extreme intervention (i will explain why we can't cycle power). We have 8 LiDar cameras connected to two USB hubs (four cameras each hub).

The USB hubs have their own power supply so no matter if we restart or turn off the computer connected, the camera will stay powered on because the their power never gets cycled.

My best solution is to perform this hardware reset on the cameras via the `rs-terminal` in hopes of unlocking the devices.

As to why these senors are locking up I am not sure. If there are any suggested to why they are locking up in the first place I am all ears for it :)

Below you will find a screen shot of the devices we are seeing in the real sense viewer and the devices we are seeing in via `rs-enumerate-devices`. I don't know why its not populating :/

[image]

stephanrotolante, 2023-04-01:
Label this as `question` and `windows`

MartyG-RealSense, 2023-04-02:
Hi @stephanrotolante   As you have divided the 8 cameras across 2 USB hubs, you are likely within in the 5000 Mbps / 5 Gbps data bandwidth limit of a typical USB hub and not running out of hub bandwidth unless you are using a high resolution and multiple streams on each camera.  

Because you can have 6 hubs on one computer in the USB specification, it may be worth considering dividing the cameras again onto a third hub if that is possible for your project if you are using high resolution and multiple stream types on each camera.

Each attached camera consumes computer resources so the more cameras that are attached to the same computer, the more of the computer's resources are consumed.  If all attached cameras are active simultaneously then a computer CPU specification of at least Intel Core i7 is recommendable if 4 or more cameras are being used.

*********

If the cameras are using the RealSense Viewer tool when out in the field and they do not disappear from the Viewer when they lock up, a camera can have a hardware reset performed by going to the **More** option of a particular camera in the Viewer's options side-panel and selecting **Hardware Reset** from its drop-down menu.

[image]

**********

If multiple L515 cameras are enabled simultaneously and their fields of view are overlapping then they can interfere with each other.  The cameras can though be configured with **hardware synchronization** so that they take turns to capture, cycling through each of the attached cameras.  

The L515 hardware sync system is different from the sync system on the RealSense 400 Series cameras, in that it is designed to minimize inter-camera interference and allow many cameras to be attached to the computer by also keeping hub bandwidth use below its limit.  Intel have a guide to L515 sync at the link below.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

********

If you would prefer to reset the camera with scripting instead of the Viewer's 'More' option then there are a couple of possibilities for a Windows computer.

A C++ script at https://github.com/IntelRealSense/librealsense/issues/9287#issuecomment-867826974 will cycle through all attached cameras when performing hardware reset.

A Python script at https://github.com/IntelRealSense/librealsense/issues/5428 enables the reset of a specific camera using its serial number.

stephanrotolante, 2023-04-02:
@MartyG-RealSense

Yes, everything works perfect on site. Once were in the field we experience these issues 🙃

stephanrotolante, 2023-04-02:
@MartyG-RealSense 

Thanks for the quick response. We were able to able to use the pythin script share to reset; however were encountering an issue on one of the cameras

`set_xu(id=2) failed! Last Error: No error`

We were successfully able to reset 7 of the 8 camera. Do you know if we can try that `rs-terminal` because something is happening on the hand shake of `ctx.query_devices()`

stephanrotolante, 2023-04-02:
@MartyG-RealSense 
[image]

stephanrotolante, 2023-04-02:
@MartyG-RealSense 

I don't know if this is an issue with our code and how the sensor was managed, or if the sensor is actually bad??

stephanrotolante, 2023-04-02:
Narrowing down in the source code

[image]

stephanrotolante, 2023-04-05:
@MartyG-RealSense yes the cameras are connect the same computer. 

We noticed our USB hub only have a max power output of 10 watts. While the a combined four L515 sensor (3.5 Watts each) have the potential for have an output of 14 Watts 🤔

MartyG-RealSense, 2023-04-05:
The USB standard allows up to six hubs to be connected to the same computer, so you may be able to split the cameras across a second hub that is plugged into the first hub (the USB standard permits that 5 hubs can be 'daisy chained').

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a long time period.